trabucayre/openFPGALoader #9: ofl alias

Martoni, 2019-12-09:
I'm too lazy to type openFPGALoader, is it possible to have a binary alias like ofl ?
```
$ ofl -?
Usage: ofl [OPTION...] BIT_FILE
openFPGALoader -- a program to flash FPGA

  -b, --board=BOARD          board name, may be used instead of cable
  -c, --cable=CABLE          jtag interface
  -d, --device=DEVICE        device to use (/dev/ttyUSBx)
  -o, --offset=OFFSET        start offset in EEPROM
  -r, --reset                reset FPGA after operations
  -v, --verbose              Produce verbose output
  -?, --help                 Give this help list
      --usage                Give a short usage message
  -V, --version              Print program version

Mandatory or optional arguments to long options are also mandatory or optional
for any corresponding short options.

Report bugs to <gwenhael.goavec-merou@trabucayre.com>.

```

trabucayre, 2020-03-21:
It's a bit unusual to have more than one name for an application. Maybe it's better to add an alias in your .bashrc like
```
alias ofl=openFPGALoader
```
or to create a link manually.
What do you think about these solutions?

Martoni, 2020-03-21:
ok

closed 2020-03-21 by Martoni