trabucayre/openFPGALoader #14: LittleBee flash programming works but reports failure

pepijndevos, 2020-01-18:
I'm using you tool (414e6ee) to program my Trenz Gowin board, and it actually works, but the tool reports failure. Not sure what is going on actually.

```
Parse pack.fs: Done
erase SRAM Done
Flash SRAM: [==================================================] 100.000000%
Done
SRAM Flash: FAIL
```

trabucayre, 2020-01-18:
This message is displayed when the checksum in the .fs file (line //CheckSum: 0x) did not match userCode read in the FPGA.
I'm unable to find where I've read this this and maybe I'm wrong to try this test. Are you sure about this entry in your .fs? In all my test with official gowin tool this two values always match.

trabucayre, 2020-01-18:
By reading the .fs generated by apicula I've seen the header is absent. This is the cause of the "SRAM Flash: FAIL".
ASAP I will try to add this and doing a PR.

pepijndevos, 2020-01-18:
Apicula does not generate the header comments. It doesn't even generate a checksum in the bitstream. This is ignored by the actual hardware, so I don't bother.

trabucayre, 2020-03-21:
I worked on computing the configuration data checksum. This piece of code (not already pushed) is working the proprietary tool. But with Apicula, where the checksum (first line) is hardcoded, this continue to fail. I know, this information is not mandatory when FPGA is configured to don't check checksum, but I think to add this feature to Apicula to improve support. What is your opinion?

pepijndevos, 2020-03-21:
I think it'd be a great addition, and would gladly accept a PR.

For myself I'm focused on timing and CI at the moment.

trabucayre, 2020-03-22:
Now checksum computation is supported by both openFPGALoader and Apicula, could you confirm me if it's always working and if you're agree I close this issue.
Thanks

pepijndevos, 2020-03-22:
Yea, I'll test it some time. Might take a few days before I find the time.

pepijndevos, 2020-09-29:
Works great now.

closed 2020-09-29 by pepijndevos