trabucayre/openFPGALoader #21: Posibility of adding FT231X

goran-mahovlic, 2020-02-23:
On ULX3S we are using FT231X chip so I do not know if there is possibility to add that one....

[www.ftdichip.com/Documents/DataSheets/ICs/DS_FT231X.pdf](www.ftdichip.com/Documents/DataSheets/ICs/DS_FT231X.pdf)

All tools we are using are mostly described here:

[https://github.com/emard/ulx3s-bin](https://github.com/emard/ulx3s-bin)

trabucayre, 2020-02-23:
Hello,
Technically yes but it's mandatory to write class equivalent to ftdipp_mpsse to handle bitbanging JTAG.
I have no ft231x nor ULX3S (waiting for crowdsupply ;-)) but I seems reasonable to suppose an implementation with ft232 will be compatible.

goran-mahovlic, 2020-02-24:
Yes, without board that is impossible, I will see what could I do ... And we are almost ready for CrowdSupply - should be lunched this month...

trabucayre, 2020-02-24:
Maybe not impossible. JTAG with this converter is in bitbang mode, so an FT232RL seems enough (I think to ordering an ft231x eval) and I have schematics.
The first task is to rethink jtag layer to add adapter for MPSSE or bitbang depending to the converter (and, why not gpio)

trabucayre, 2020-03-05:
I'm currently cleanup a first final draft to split jtag between logic and hardware. I'm able to use openFPGALoader with an ft232R (without breaking ft2232 support).

trabucayre, 2020-03-07:
Hi,
I've finally creates a branch for bitbanging with ftdi (see ftdi_bitbang branch), with ulx3s board support.
Could you try to test this branch with:
openFPGALoader -b ulx3s --detect
and/or
openFPGALoader -b ulx3s /somewhere/file.bit

Thanks

goran-mahovlic, 2020-03-07:
Tnx! 

I fixed small typo in cable.hpp to compile

bud board was not detected...

[image]

goran-mahovlic, 2020-03-07:
typo is just missing comma in this line...

https://github.com/trabucayre/openFPGALoader/blob/ftdi_bitbang/src/cable.hpp#L27

trabucayre, 2020-03-07:
Thank for the typo, it's updated.
It's really strange for the detection... VID and PID in cable.hpp matches your lsusb. It's like libusb don't see the peripheral... I have not already received my ft231x eval board... I need to search if one board I have has this converter (just to verify detection).
It's finally a bit hard to test without corresponding hardware :)

goran-mahovlic, 2020-03-07:
Yes! 
Actually main problem is that I am at 0 boards / all versions :(
will ask emard about libusb problem maybe he knows why we do not get detection ...

trabucayre, 2020-03-07:
the dev board of http://www.opossom.com/english/products-processor_boards-opos6ul_sp.html
has the desired device.
My first try has same result as you,
So I've updated udev with this rule
ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6015", MODE="664", GROUP="plugdev"
and now I'm able to use this converter.

goran-mahovlic, 2020-03-07:
Still no luck on my side :(

[image]

goran-mahovlic, 2020-03-07:
I even rebooted ...

trabucayre, 2020-03-07:
This first try with sudo is good -> ft231x is detected.
But FPGA is not found... Maybe I'm wrong in pins used for ULX3S or in a small detail.

goran-mahovlic, 2020-03-07:
cool! 

I have 12F board if that makes difference?

trabucayre, 2020-03-07:
No.
I have tried bitbanging with spartan7 board and colorlight V7.0 (LFE5U-25F-6BG256C) and this is okay.
According to ujprog all ULX3S have the same pinout, UL2x are differents. So theorically this must work...

goran-mahovlic, 2020-03-07:
We can try with ULX2S?

trabucayre, 2020-03-07:
try to add:
```
{"ulx2s",      {"ft231X",   {FT232RL_RI, FT232RL_DSR, FT232RL_CTS, FT232RL_DCD}}},
```
just before ulx3s in board.hpp
All ECP5 are not supported so It may possible to have something like:
```
found 1 devices
Error: device XXXXXXX not supported
```

goran-mahovlic, 2020-03-07:
got it working on ulx2s, but I have ft232RL there...
will check if I have external ftdi so I can try on other targets...

{"ulx2s",      {"ft232RL",   {FT232RL_RI, FT232RL_DSR, FT232RL_CTS, FT232RL_DCD}}},

sudo ./openFPGALoader -b ulx2s --detect -v
found 1 devices
Error: device 129a043 not supported

trabucayre, 2020-03-07:
Ok. This work with ULX2S and able to detect an LFXP2-8E
So, seems to have a problem with ft231x... Theorically I receive my converter monday, will re-read again datasheet...

I have update board.hpp and part.hpp to add ulx2s and lattice XP2, thank

trabucayre, 2020-03-11:
I have, finally, received my ft231x eval board and found a solution. Just adding a FIFO buffer purge when changing mode (async/sync). Work with EPC5 and spartan7 FPGA.
Some improvment around buffer size needs to be added to increase transfert.
Could you confirm with ulx3s? If it's okay for you I will merge this branch to master
Thank

goran-mahovlic, 2020-03-11:
Amazing!

After adding 12F ID it works!

[image]

Many thanks!

trabucayre, 2020-03-12:
Yes!! :-)
the 12F idcode has been added in branch master.
If you are okay I merge this branch to master and close this issue.

goran-mahovlic, 2020-03-12:
Yes, merge and close sounds great!

trabucayre, 2020-03-12:
Thanks.
Now, I wait crowdsupply to buy the ulx3s :)

closed 2020-03-12 by trabucayre