trabucayre/openFPGALoader #23: how to build static executable

emard, 2020-03-13:
Feature with - some short note in readme on how to build
static executable, independent from shared libraries.

It could increases portability of compiled executable
accross various linux distro's.

trabucayre, 2020-03-14:
Good idea.
Currently, this feature is not supported. I've just tried after some modifications in CMakeLists.txt but in my distro udev as no .a and the build fails. I need to find a distribution where all required .a are present to validate modifications before pushing.

kost, 2020-03-14:
Here's the changes I have done together with process which builds openFPGAloader without problems:

https://github.com/alpin3/ulx3s-core/commit/810193f0615069328f219c9d5766471dd13cd4c2

I'm using Docker on Alpine with Eudev library:
https://wiki.gentoo.org/wiki/Eudev

Also, have to use argp-standalone apk package since you're using argp.h which is glibc specific.

trabucayre, 2020-03-14:
Thank for this link.
I've just push a test (and note) to use argp-standalone for musl or uClibc.
I'll just doing some verifications before applying your proposition for static libs but look good.

trabucayre, 2020-03-14:
Based on kost's patch I've added STATIC_BUILD option and updated README.
If it's good for you, and if you're agree, I close this issue.

emard, 2020-03-14:
I agree :)

trabucayre, 2020-03-15:
Great

closed 2020-03-15 by trabucayre