trabucayre/openFPGALoader #24: Multiplatform support

kost, 2020-03-14:
openFPGA loader looks promising, but I see it is targeted to Linux with glibc only. Do you have any plans to port it to something else? 

Current problems I see are:
- dependency on udev without ifdefs and cmake option
- dependency on argp which is glibc specific (musl libc does not have it by default, also other OSes than Linux)

trabucayre, 2020-03-14:
For argp I have answered in [#23](https://github.com/trabucayre/openFPGALoader/issues/23).
It's currently linux only because I don't have windows or mac for trying (if someone could test/fix I'm open) but yes it's possible to have udev (and consequently -d /dev/ttyXXX) optional to avoid dependency to this library.

trabucayre, 2020-03-14:
I have pushed option to disable udev support and corresponding cli option.
Does that seem correct to you?
Thanks for your remarks

kost, 2020-03-15:
Pull request which implements Mac OS X support: https://github.com/trabucayre/openFPGALoader/pull/25

I guess you have access to Windows or Mingw, so that is not needed.

trabucayre, 2020-03-15:
I've just applied your PR. Thank you again!
In fact, I have no windows at all and no one around me either.

trabucayre, 2020-03-21:
If you think this issue is resolved, can I close it?
Thanks

kost, 2020-04-14:
Thanks! I think you can. Regarding Windows support, you can check how I managed to cross compile fujprog for Windows:
https://github.com/kost/fujpro

You don't need to have windows since you can cross compile from linux.

closed 2020-04-14 by kost