trabucayre/openFPGALoader #26: Mac OS X detach error

kost, 2020-03-16:
Loading bitstream works on Mac OS X, but loader is reporting `detach error 12`:

```
./openFPGALoader -b ulx3s ~/fpga/saxon-45f.bit
Open file /Users/vkosturj/fpga/saxon-45f.bit DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.000000%
Done
Disable configuration: DONE
detach error -12
```

Same happens with administrator/root privileges:
```
sudo ./openFPGALoader -b ulx3s ~/fpga/saxon-45f.bit
Open file /Users/vkosturj/fpga/saxon-45f.bit DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.000000%
Done
Disable configuration: DONE
detach error -12
```

I don't see any side effects except it is annoying to see error above.

trabucayre, 2020-03-17:
Thanks to point this issue.
Error code -12 mean LIBUSB_ERROR_NOT_SUPPORTED
According to
http://libusb.sourceforge.net/api-1.0/group__libusb__dev.html#gadeba36e900db663c0b7cf1b164a20d02
libusb_attach_kernel_driver is only available on Linux.

Fixed by d210d2c67e0aceb38979c96ba888981a0a426d43
Could you confirm please?

Thank again

kost, 2020-03-17:
Can confirm it is working. No error.

```
~> openFPGALoader -b ulx3s ~/fpga/saxon-45f.bit
Open file /Users/user/fpga/saxon-45f.bit DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.000000%
Done
Disable configuration: DONE
```

closed 2020-03-17 by kost