trabucayre/openFPGALoader #27: Gowin GW2A-55 support

tinylabs, 2020-04-01:
Looking to add support for the gowin gw2a-55 part.

I've added the following line to part.hpp:
{0x0000281B, {"Gowin", "GW2A", "GW2A-55"}},

I've tested on the DK-START-GW2A55-PG484 dev board and SRAM configuration programming works fine with this addition but the calculated checksum doesn't match the userCode checksum which causes it to report it failed. Any advice on how to reverse the checksum from the .fs file?

Martoni, 2020-04-01:
Hi @tinylabs, your error seems to be same with other gowin as reported in this [issue](https://github.com/trabucayre/openFPGALoader/issues/14).

tinylabs, 2020-04-01:
I think it may be a different issue. I'm using the official gowin tools to generate the .fs file. I looked through the code and TN653 and everything looks correct (addr_length, padding, etc). I'm wondering if it's not a simple checksum used for this part... Maybe some other CRC16 poly?

trabucayre, 2020-04-02:
I think you're true about [issue](#14).
Now openFPGALoader compute checksum instead of using comment information, so I'm waiting for confirmation before closing them.
Maybe the computation is a bit too naive.
For your device I suspect there have no need to pad line, since address length is multiple of 8bits.
I've done some reverse based on Apicula documentation and an address line look like:
P D*n CCCC 6x0xff
where
P is a padding (for GW2A-55, address length is multiple of 8, so Im believe their a no padding);
Dxn are Configuration Data Bits
CCCC is the address checksum. But this area is only present if set CRC_check = true. Currently, openFPGALoader consider always true.
and finally the line end with 6x 0xff
Could you confirm me this structure?
I think I must generate something for this device, to have an fs to analyze, to improve checksum.

tinylabs, 2020-04-02:
I can confirm that structure is correct. Address lines are 5600 bits long.
0 bytes padding
5536 data bits
16 bit checksum
48 bit trailing 1's

The only discrepancy I found is the number of address lines is specified as 2038 in TN653 p.18 but the gowin tools produce 3318 lines. For the simple checksum algo this doesn't change anything as the last 1280 are zero (in my design). Maybe some of the configuration data is part of checksum as well? I'd be happy to share a couple .fs files if that helps? Appreciate your help with this!

trabucayre, 2020-04-02:
I have created design's for this FPGA version.
My analyze match yours. But the strange thing is all .fs fit number of lines.
Header and footer seems to be same as GW1N but checksum computed is always wrong...
I need to recheck all applications notes, maybe I'm lost something.
Yes If it's possible to share some .vm and .fs files maybe it's help. Thanks

tinylabs, 2020-04-02:
Here are some sample .fs configuration images:
https://github.com/tinylabs/cm3-oracle/tree/master/gowin/impl/pnr
They have the security bit set which adds a line to the header but that shouldn't matter. I removed it and it made no difference since checksum is only calculated over configuration data. Also verified checksum in official tool didn't change which leads me to believe header isn't included in checksum.

I picked through TN653 for quite a while yesterday but couldn't find anything else helpful. Do you see 2038 lines for your generated images? Maybe they are using a non-zero starting value or some other poly? When I have more cycles I'll normalize the configuration data and check against common 16bit CRC polys.

trabucayre, 2020-04-05:
Thanks for fs files. I've tried to analyze too but checksum are always wrong.
Yes all my files contains 2038 lines for configuration data. Maybe a different configuration somewhere between .vm to .fs ?
I think to add a dump method to read back memory content to check if something change...

I'll read gw2n datasheet to see if all line must be taken into account.

trabucayre, 2022-09-04:
With @Icenowy we have disabled checksum verification for GW2A until correct computation will be found.
Would you add GW2A-55 idcode?
Thanks

trabucayre, 2024-11-26:
I have added idcode for this model.
Please open an issue (or re-open this issue if something wrong).
Thanks

closed 2024-11-26 by trabucayre