trabucayre/openFPGALoader #28: unable to open ftdi device: -3 (device not found)

paulossant, 2020-04-10:
Hi, When trying to upload the code to the device (tang Nano) with:
openFPGALoader -m -b tangnano led_prj.fs
(the .fs file was generated with the vendor tools and must be good, since its the default example)

With this, I get in return:
unable to open ftdi device: -3 (device not found)

Furthermore, I connected the device via USB and the cable is good, since I tested it with other devices and works in data transfer mode.

When I do lsusb, also the device does not appear. Any idea of whats going wrong?

Best regards, and thank you.

Paulo

trabucayre, 2020-04-13:
Hi, and sorry for the late.
My first idea is a problem with access right (I've just pushed a udev rule for that and updated README.md), but if lsusb didn't displays anything about your tangNano, the problem seems not around openFPGALoader but at more low level (maybe the tangNano board itself).
Have you some messages when you plug your device in dmesg?
Try to change your cable or to test with an other computer

trabucayre, 2020-04-28:
Any news ?
Can I close this issue ?

closed 2020-10-22 by trabucayre