trabucayre/openFPGALoader #31: Posibility of adding  MachXO2

borovecek, 2020-04-23:
Hi
Would it be possible to add "MachXO2" to the supported devices?
I would like to program "lcmxo2-7000hc-4ftg256c"
[http://www.latticesemi.com/view_document?document_id=38834](url)
is it possible? Thank you

trabucayre, 2020-04-23:
As I can see, by comparing both programming guide, instructions are same.
So I suppose it's just mandatory to add the corresponding line in the fpga_list in part.hpp
with something like
```
{0x012BD043, {"lattice", "MachXO2-ES", "LCMX02-7000HC"}},
```
Please tell me if it's enough to use this device (or sent a PR).

borovecek, 2020-04-23:
I have problem, when i try programming, give me this error:

> pi@raspberrypi:/ $ sudo openFPGALoader --detect
idcode 0x12bd043
manufacturer lattice
model  LCMX02-7000HC
family MachXO2-ES
pi@raspberrypi:/ $ sudo openFPGALoader -b machXO3SK /home/pi/Desktop/LCMX.jed
Open file /home/pi/Desktop/LCMX.jed DONE
Unauthorized Memory Access (SIGSEGV)
pi@raspberrypi:/ $ 

where am I making mistake?

trabucayre, 2020-04-23:
Seems to have a problem with jed parsing. 
I've tried to parse jed file dedicated for a machxo2 (I've not board with this device) quite succesfully (a problem with checksum but out of context here).
You use an raspberrypi ? Wich model, distribution and version. Maybe a problem not visible on a computer?

trabucayre, 2020-04-23:
I've installed an raspbian buster on my raspberry3, compiled openFPGALoader:
- parsing (not flash) with a jed dedicated to a LCMXO2-256HC-4SG32C => success
- parsing and flash with a jed dedicated for a  LCMXO3LF-6900C-5BG256C => success

Could you try on a computer to check if problem is around raspberry distribution or is with openFPGALoader.

I think I must find a machXO2 to test.

borovecek, 2020-04-24:
> Seems to have a problem with jed parsing.
> I've tried to parse jed file dedicated for a machxo2 (I've not board with this device) quite succesfully (a problem with checksum but out of context here). [...]

Yes, i´am using RASPBERRY PI 3B
and here is info about SW:
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch

trabucayre, 2020-04-24:
I have tried with raspbian stretch (but 9.11 instead of 9.4).
Same tests => success...
I've pushed some small fix in jedParser... Could you try to update openFPGALoader and try again.
Maybe an update to your raspbian is mandatory.
Could you post dmesg ?
Could you share your projet? Maybe I've forget something in jedParser.
If all of them didn't fix your issue, I suppose I need to bought an machXO2 eval board to test by myself.

borovecek, 2020-04-24:
Updatoval jsem openFPGALoader a nastala změna a trošku se to pohlo: 

> pi@raspberrypi:~/Desktop $ sudo openFPGALoader -b machXO3SK /home/pi/Desktop/LCMX.jed
Enable configuration: DONE
SRAM erase: DONE
Open file /home/pi/Desktop/LCMX.jed DONE
Parse file DONE
Enable configuration: DONE
Flash erase: DONE
Writing: [] nan%
Done
Unauthorized Memory Access (SIGSEGV)

send a report from dmesg is not a problem, should it have a parameter?

progress is same for the older older "stretch" and on newer "buster"

borovecek, 2020-04-24:
I updated openFPGALoader and there was a change and it moved a bit:

> pi@raspberrypi:~/Desktop $ sudo openFPGALoader -b machXO3SK /home/pi/Desktop/LCMX.jed
Enable configuration: DONE
SRAM erase: DONE
Open file /home/pi/Desktop/LCMX.jed DONE
Parse file DONE
Enable configuration: DONE
Flash erase: DONE
Writing: [] nan%
Done
Unauthorized Memory Access (SIGSEGV)

send a report from dmesg is not a problem, should it have a parameter?

progress is same for the older older "stretch" and on newer "buster"

trabucayre, 2020-04-24:
Last lines in dmesg may displays some informations.
An other approach is to use gdb
```
cmake -DCMAKE_BUILD_TYPE=Debug .. # compile in debug mode
make
ulimit -c unlimited # coredump when crash
sudo ./openFPGALoader -b machXO3SK /home/pi/Desktop/LCMX.jed
gdb ./openFPGALoader core # post mortem analyze
```
to see all call use bt. It must display where the problem is, and list of function called between main and the line where the problem is

Thank

borovecek, 2020-04-27:
after many tries, i found out, i'm a dumbass, i wrote command: sudo ulimit -c unlimited. and it didn't work, command not found. After many attempts I tried it without sudo and it worked. Here is result:

> pi@raspberrypi:~ $ sudo openFPGALoader -b machXO3SK /home/pi/Desktop/LCMX.jed
Enable configuration: DONE
SRAM erase: DONE
Open file /home/pi/Desktop/LCMX.jed DONE
Parse file DONE
Enable configuration: DONE
Flash erase: DONE
Writing: [] nan%
Done
Unauthorized Memory Access (SIGSEGV)

pi@raspberrypi:~ $ sudo gdb openFPGALoader core
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from openFPGALoader...done.
[New LWP 2666]
[New LWP 2667]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `openFPGALoader -b machXO3SK /home/pi/Desktop/LCMX.jed'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0003ffb0 in std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::size (this=0x4) at /usr/include/c++/8/bits/stl_vector.h:806
806	      { return size_type(this->_M_impl._M_finish - this->_M_impl._M_start); }
[Current thread is 1 (Thread 0x76f0ae00 (LWP 2666))]
(gdb)

trabucayre, 2020-04-27:
This line shows where the crash occur but not the backtrack.
To have more informations use
```
(gdb) bt
```
Anyway, I've found a machXO2 based board, but I need to wait some days before reception.

borovecek, 2020-04-27:
new result is: 

> For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from openFPGALoader...done.
[New LWP 2666]
[New LWP 2667]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `openFPGALoader -b machXO3SK /home/pi/Desktop/LCMX.jed'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0003ffb0 in std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::size (this=0x4) at /usr/include/c++/8/bits/stl_vector.h:806
806	      { return size_type(this->_M_impl._M_finish - this->_M_impl._M_start); }
[Current thread is 1 (Thread 0x76f0ae00 (LWP 2666))]
(gdb) bt
#0  0x0003ffb0 in std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::size (this=0x4) at /usr/include/c++/8/bits/stl_vector.h:806
#1  0x00047d04 in std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::vector (this=0x7eb3ad7c, __x=<error reading variable: Cannot access memory at address 0x8>) at /usr/include/c++/8/bits/stl_vector.h:460
#2  0x0006a17c in JedParser::data_for_section[abi:cxx11](int) (this=0x7eb3ae10, id=0) at /home/pi/Desktop/openFPGALoader-master/src/jedParser.hpp:46
#3  0x000686e4 in Lattice::Verify (this=0x14ec040, _jed=..., unlock=false) at /home/pi/Desktop/openFPGALoader-master/src/lattice.cpp:778
#4  0x000663a4 in Lattice::program_intFlash (this=0x14ec040) at /home/pi/Desktop/openFPGALoader-master/src/lattice.cpp:323
#5  0x00067308 in Lattice::program_flash (this=0x14ec040, offset=0) at /home/pi/Desktop/openFPGALoader-master/src/lattice.cpp:449
#6  0x000675d0 in Lattice::program (this=0x14ec040, offset=0) at /home/pi/Desktop/openFPGALoader-master/src/lattice.cpp:475
#7  0x000593b0 in main (argc=4, argv=0x7eb3b714) at /home/pi/Desktop/openFPGALoader-master/src/main.cpp:190
(gdb)

trabucayre, 2020-04-27:
Thank,
These dump is really helpful. I have updated some part of Verify.
Could you test again, after a git pull and make ?
Thanks

borovecek, 2020-04-28:
today is a step forward, I created a new build, and when it threw an error, I repeated yesterday's procedure so you could see the report

> pi@raspberrypi:~ $ sudo openFPGALoader -b machXO3SK /home/pi/Desktop/LCMX.jed
Enable configuration: DONE
SRAM erase: DONE
Open file /home/pi/Desktop/LCMX.jed DONE
Parse file DONE
Enable configuration: DONE
Flash erase: DONE
Writing: [] nan%
Done
Verifying: [] nan%
Done
Write program Done: DONE
Disable configuration: DONE
Refresh: FAIL

> pi@raspberrypi:~ $ sudo gdb openFPGALoader core
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

> For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from openFPGALoader...done.

> warning: exec file is newer than core file.
[New LWP 2666]
[New LWP 2667]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `openFPGALoader -b machXO3SK /home/pi/Desktop/LCMX.jed'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0003ffb0 in std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::operator[] (this=0x4, 
    __n=509) at /usr/include/c++/8/bits/stl_vector.h:951
951		return *(this->_M_impl._M_start + __n);
[Current thread is 1 (Thread 0x76f0ae00 (LWP 2666))]
(gdb) bt

> #0  0x0003ffb0 in std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::operator[] (this=0x4, 
    __n=509) at /usr/include/c++/8/bits/stl_vector.h:951
#1  0x00047d04 in std::vector<JedParser::jed_data, std::allocator<JedParser::jed_data> >::operator[] (this=0x7eb3b584, __n=21938240)
    at /usr/include/c++/8/bits/stl_vector.h:933
#2  0x00047d04 in std::vector<JedParser::jed_data, std::allocator<JedParser::jed_data> >::operator[] (this=0xe1a01003, __n=3800248324)
    at /usr/include/c++/8/bits/stl_vector.h:933
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb)

trabucayre, 2020-04-28:
Thank.
Initial problem with segfault is fixed. It's great!
So It's no more mandatory to use gdb.

Now the problem is around Refresh. I pushed a modification to display status register when loadConfiguration fails. Could you update your repo and test again. Maybe this register will displays some interresting informations.

borovecek, 2020-04-28:
Now is result:

> pi@raspberrypi:~/Desktop $ sudo openFPGALoader -b machXO3SK /home/pi/Desktop/LCMX.jed
Enable configuration: DONE
SRAM erase: DONE
Open file /home/pi/Desktop/LCMX.jed DONE
Parse file DONE
Enable configuration: DONE
Flash erase: DONE
Writing: [] nan%
Done
Verifying: [] nan%
Done
Write program Done: DONE
Disable configuration: DONE
Refresh: FAIL
displayReadReg
	Config Target Selection : 0
	SDM Enable
	Preamble ERR
	EXEC Error

trabucayre, 2020-04-29:
Thanks.
I suspect a problem with bitstream transmission. Could you share your jed, Maybe by reading this file, and code behavior I will be able to focus this issue.

trabucayre, 2020-05-02:
Hi.
I have, finally, received a machXO2 based board.

Tested and adding some fix:
jed: comment before STX, potential **\r** char before **\n** (DOS file format)
lattice support: missing EBR init data use.

Could you update your repo and try again?

borovecek, 2020-05-04:
Sorry for late answer, i tried communicate with my superior due to send jed file, because this project unfortunately is not open-source. But with last change in your program, is programming and verification functional.  In file part.hpp ,is only one change, added this line what you send in second answer:

> {0x012BD043, {"lattice", "MachXO2-ES", "LCMX02-7000HC"}},

trabucayre, 2020-05-04:
Great!
I've added LCMX02-7000HC in part.hpp
If you're agreee I close this issue ?
Thank again!

borovecek, 2020-05-04:
Yes, i agree.
Thank you

closed 2020-05-04 by borovecek