trabucayre/openFPGALoader #35: Lichee Tang the SiPeed version

japina, 2020-05-20:
I know that it's currently not supported, but is there any chance for it to be? The device is Bus 020 Device 010: ID 0547:1002 Anchor Chips Inc. USB-JTAG-Cable, but I guess the FPGA is the same (or at least similar) as on Lichee Tang nano?

trabucayre, 2020-05-20:
After a quick check in internet, this board is based on anlogic FPGA.
I have an tang Primer board with same caracteristic (maybe it's the same it's not clear).
My problem is:
- USB<->JTAG adapter is not documented. Only some binaries are available;
- I'm unable to find anything about JTAG protocol for this FPGA. English part is just dead for a long time, with translate I'm able to move in the website but no way to create a account to have access to documentation....
Without more information it's really an hard task because it's mandatory to revert USB protocol+Jtag command...

But Yes I wish to add this support.

trabucayre, 2020-08-20:
Sorry the long time.
I have updated openFPGALoader to work with lichee tang board.
see https://github.com/trabucayre/openFPGALoader/blob/master/README.md#sipeed-lichee-tang
For instructions

closed 2021-01-06 by trabucayre