trabucayre/openFPGALoader #38: Setting up a Windows build

edbordin, 2020-07-21:
Hi there!
I've been maintaining nightly builds of a selection of open fpga tools here https://github.com/open-tool-forge/fpga-toolchain Someone suggested adding openFPGALoader and it looks like it would be a great addition to the fpga-toolchain package.

I read #24 and if you are interested I'd like to help add Windows support so that I can provide nightly builds for Linux, OS X and Windows.

We're already building other tools using libusb/libftdi so those are no problem, but libargp is trickier. It looks like libargp has a standalone version for OS X but I couldn't find one for MinGW on Windows. I spent a little bit of time seeing what it would take to port to Windows and to be honest, it looks like it would be easier to use a different library. I was wondering if you would be open to a pull request using a different library to parse the command line arguments? Given that you are already set up to compile C++11 I think this single-header library would be a good fit https://github.com/jarro2783/cxxopts

trabucayre, 2020-07-21:
Adding openFPGALoader to this is a great idea!
I've read # 24 (and corresponding code) but since I have no skills with this OS and no windows to validate I've preferred delayed this addition. I'm happy by help for this topic.
for argp,I have found this https://packages.msys2.org/base/libargp but I don't know if it's compatible with the argp used in  openFPGALoader.
Anyway, I know, my choise is maybe not the best and I'm open to move to other option.
for cxxopts I'm a bit mitigate, I need to take into account buildroot and distribution's package manager. Consequently, I prefer a solution widely available, to avoid a required library to install manually.
Maybe an other solution is like for [fujprog](https://github.com/kost/fujprog) with getopt only used with windows (maybe more complex compared to libargp but more OS independant).

Thanks and let me know if libargp is compatible (or not) to evaluate an alternate solution.

edbordin, 2020-07-22:
> I have found this https://packages.msys2.org/base/libargp but I don't know if it's compatible with the argp used in openFPGALoader.

Unfortunately this is not compatible. MSYS2/MinGW can get a little confusing. This package has a dependency on the MSYS runtime and won't work for a MinGW build target. A package for a MinGW64 target would be called something like `mingw-w64-x86_64-libargp`. It does look like gnulib can be compiled for MinGW but we would have to build it from source, which is what I was having trouble with.

If you prefer a C-style library then using getopt like in fujprog looks like it would also work.

The easiest way to use cxxopts would be to copy the single header file into your repo. It only uses C++11 standard library so it should just work on most compilers. Or you could use git submodule to include the repo. You can see what it looks like here https://github.com/jarro2783/cxxopts/blob/v2.2.0/include/cxxopts.hpp

trabucayre, 2020-07-22:
Okay.
So It's time to move to other args parser. I've tried cxxopts, it work well, with same functionalities. I need to fix one little thing (nitpicking): these syntax are allowed
```bash
--foo=bar
--foo bar
-f bar
```
but this one no
```bash
-fbar
```
Since I wish to keep compatibility I need to modify a bit the code to have same behavior between argp and cxxopts. This modification must be PR to official repository and waiting to integration I'll copiying the header

As soon as I've pushed (in a new branch to test) this update I warn you to validate if it's working for all OS

Thanks.

trabucayre, 2020-07-23:
I have pushed in cxxopts_migration branch everything to use cxxopts instead of argp.
Could you test/validate if it's good with windows (and with OSX if it's possible) (and maybe check if there are no regression I've not seen).

Thanks

edbordin, 2020-07-26:
OK, I got it building and it seems to be working!

I had to make some minor adjustments which I have pushed here https://github.com/edbordin/openFPGALoader/tree/cxxopts_migration
The build script I have been using is here:
https://github.com/open-tool-forge/fpga-toolchain/blob/feature/13-prog/scripts/compile_openfpgaloader.sh

In summary:

1. MinGW doesn't seem to provide `bzero()` so I replaced it with `memset()`. Hopefully you agree this is reasonable (apparently `bzero` is deprecated in recent POSIX standards anyway).
1. I replaced one occurrence of `error_t` with `int` as it also didn't seem to be defined in MinGW. According to this stackoverflow answer this should work fine: https://stackoverflow.com/a/24207339
1. On windows I had to link with the winsock library for the `ntohs` function.
1. On linux I have been using custom built libusb/libftdi without udev support. This enables static linking without a dependency on systemd. I needed to add an option to disable finding the libraries with pkg_config to make this easier to build. Due to the way cmake automatically reorders the `LDFLAGS` the easiest way I found to get it to link correctly without pkg_config was to add the workaround under `LINK_CMAKE_THREADS`.

I tested it on all three platforms using a ft232 cable and a colorlight ECP5 board. All three are able to run `openFPGALoader -c ft232 --detect` successfully as well as programming flash and sram on the board. I didn't notice any differences in behaviour in my testing.

Let me know if anything is unclear, of if there is anything else you'd like me to do!

You can download the builds from my branch here if you would like to test anything yourself:
https://dev.azure.com/open-tool-forge/fpga-toolchain/_build/results?buildId=321&view=artifacts&type=publishedArtifacts

trabucayre, 2020-07-26:
Hi,
1. Looks good for me. It's true, bzero is deprecated (it's explicitly said in the manpage. This page recommend to use memset)
2.error_t seems to be provided by argp.h -> since parse_eng return an int is okay for me too
3. ok. Maybe I need to provide something better than ntohs. Using inet just to swap two byte is maybe overkill
4. this modification doesn't break anything so seems good. Anyway, adding CMAKE_xxxx_FLAGS is good (mandatory?) in many use case (cross-building F.I.)

It's awesome to have openFPGALoader working on all system! Thanks for your work!

The only thing that surprises me is the need of threads.

I will testing your build and waiting for your PR!

Thanks again!

trabucayre, 2020-07-26:
I've finish to pass a serie of test around arguments parsing. Now cxxopts_migration branch is merged, with your contribution, into master.

Can I close this issue ?

Thanks

edbordin, 2020-07-27:
Great! Nothing has changed since my PR so I'm happy to close this.

Hopefully by tomorrow I will start shipping openFPGALoader nightly builds!

closed 2020-07-27 by edbordin