trabucayre/openFPGALoader #39: Colorview:  Device failed to verify

hvegh, 2020-07-24:
I tried to flash the blinky demo on the colorview 5a-75b v7
any suggestions would be greatly appreciated.

First detect:
```
$ openFPGALoader -cft232 --detect
idcode 0x41111043
manufacturer lattice
model  LFE5U-25
family ECP5

```
then write bitstream to sram:
```
$ openFPGALoader -cft232 blink.bit 
Open file blink.bit DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.000000%
Fail
displayReadReg
	Config Target Selection : 0
	JTAG Active
	ISC Enable
	Write Enable
	Read Enable
	Std PreAmble
	SPIm Fail1
	SDM EOF
	EXEC Error
	Device failed to verify
	Invalid Command
	SED Error
	Bypass Mode
	FT Mode
```

trabucayre, 2020-07-24:
The detect part show a correct communication between openFPGALoader and your FPGA.
The flash part seems to show an incompatibilty somewhere. Just to test I've tried to load a .bit 
- targeted for this board (and an other targeted for orange crabe (same FPGA)) -> okay
- targeted for ecp5_evn (not the same model/size) -> fail (with more or less same register status content)

Are you sure your bit has been produce for the colorlight's FPGA model. Have you a link to your blinky demo so that I can reproduce?
Thanks

hvegh, 2020-07-24:
Hi Gwenhael,

The demo is from: [https://github.com/Martoni/blp](https://github.com/Martoni/blp) in blp/platforms/colorlight
Unedited, just run make that should do it.

I am trying different cables now.
Yes! 
The lesson is check the wires...

```
blp/platforms/colorlight $ openFPGALoader -v -c ft232 blink.bit 
found 1 devices
idcode 0x41111043
manufacturer lattice
model  LFE5U-25
family ECP5
File type : bit
Open file blink.bit DONE
Parse file DONE
Lattice bitstream header infos
Part: LFE5U-25F-6CABGA256
IDCode : 41111043
displayReadReg
	Config Target Selection : 0
	Done Flag
	Std PreAmble
	SPIm Fail1
	No err
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.000000%
Done
userCode: 00000000
Disable configuration: DONE
displayReadReg
	Config Target Selection : 0
	Done Flag
	Std PreAmble
	SPIm Fail1
	No err
```

openFPGAloader works great, thanks!

closed 2020-07-24 by hvegh

trabucayre, 2020-07-25:
Great if it's work! Thank for feedback!
But what do you mean by "check the wires" ?
Your first try has shown FPGA's idcode. This seems to prove your probe seems correctly connected to the colorlight.

hvegh, 2020-07-28:
I resoldered the header and changed the wires on the JTAG connector. Downloading bitstreams seems much more stable now...