trabucayre/openFPGALoader #41: Can't compile on centos7

Fatsie, 2020-08-02:
I can't compile openFPGALoader on CentOS7, I get the following error:
```console
[verhaegs@localhost build]$ LANG=en_US.utf8 make
[  4%] Building CXX object CMakeFiles/openFPGALoader.dir/src/ftdipp_mpsse.cpp.o
/home/verhaegs/eda/code/openFPGALoader/src/ftdipp_mpsse.cpp: In constructor ‘FTDIpp_MPSSE::FTDIpp_MPSSE(const FTDIpp_MPSSE::mpsse_bit_config&, const string&, uint32_t, bool)’:
/home/verhaegs/eda/code/openFPGALoader/src/ftdipp_mpsse.cpp:26:49: error: incompatible types in assignment of ‘const char [1]’ to ‘char [64]’
     _clkHZ(clkHZ), _buffer_size(2*32768), _num(0)
                                                 ^
/home/verhaegs/eda/code/openFPGALoader/src/ftdipp_mpsse.cpp: In constructor ‘FTDIpp_MPSSE::FTDIpp_MPSSE(const FTDIpp_MPSSE::mpsse_bit_config&, uint32_t, bool)’:
/home/verhaegs/eda/code/openFPGALoader/src/ftdipp_mpsse.cpp:52:51: error: incompatible types in assignment of ‘const char [1]’ to ‘char [64]’
       _clkHZ(clkHZ), _buffer_size(2*32768), _num(0)
                                                   ^
make[2]: *** [CMakeFiles/openFPGALoader.dir/src/ftdipp_mpsse.cpp.o] Error 1
make[1]: *** [CMakeFiles/openFPGALoader.dir/all] Error 2
make: *** [all] Error 2
```
I have following versions of libftdi and g++ installed:
```console
[verhaegs@localhost build]$ rpm -q libftdi-devel gcc-c++
libftdi-devel-1.1-4.el7.x86_64
gcc-c++-4.8.5-39.el7.x86_64
```

trabucayre, 2020-08-04:
Seems to be an issue with gcc 4.8. Anyway it's now fixed.
Could you validate and if it's okay close this issue?

Thanks

Fatsie, 2020-08-04:
OK, got it compiled now but get a when trying to run, backtrace:
```console
[verhaegs@localhost build]$ gdb ./openFPGALoader 
GNU gdb (GDB) Red Hat Enterprise Linux 7.6.1-119.el7
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /home/verhaegs/eda/code/openFPGALoader/build/openFPGALoader...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/verhaegs/eda/code/openFPGALoader/build/./openFPGALoader 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
terminate called after throwing an instance of 'std::regex_error'
  what():  regex_error

Program received signal SIGABRT, Aborted.
0x00007ffff6d7e387 in raise () from /lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install bzip2-libs-1.0.6-13.el7.x86_64 elfutils-libelf-0.176-4.el7.x86_64 elfutils-libs-0.176-4.el7.x86_64 glibc-2.17-307.el7.1.x86_64 libattr-2.4.46-13.el7.x86_64 libcap-2.22-11.el7.x86_64 libftdi-1.1-4.el7.x86_64 libusbx-1.0.21-1.el7.x86_64 systemd-libs-219-73.el7_8.8.x86_64 xz-libs-5.2.2-1.el7.x86_64 zlib-1.2.7-18.el7.x86_64
(gdb) bt
#0  0x00007ffff6d7e387 in raise () from /lib64/libc.so.6
#1  0x00007ffff6d7fa78 in abort () from /lib64/libc.so.6
#2  0x00007ffff76bbcb3 in __gnu_cxx::__verbose_terminate_handler ()
    at ../../../../libstdc++-v3/libsupc++/vterminate.cc:95
#3  0x00007ffff76c1e26 in __cxxabiv1::__terminate(void (*)()) ()
    at ../../../../libstdc++-v3/libsupc++/eh_terminate.cc:47
#4  0x00007ffff76c1e61 in std::terminate () at ../../../../libstdc++-v3/libsupc++/eh_terminate.cc:57
#5  0x00007ffff76c2094 in __cxxabiv1::__cxa_throw (obj=obj@entry=0x73a550, 
    tinfo=tinfo@entry=0x7ffff79a7a50 <typeinfo for std::regex_error>, 
    dest=dest@entry=0x7ffff76ea7f0 <std::regex_error::~regex_error()>)
    at ../../../../libstdc++-v3/libsupc++/eh_throw.cc:95
#6  0x00007ffff76bdd2b in std::__throw_regex_error (__ecode=std::regex_constants::_S_error_brack)
   from /home/verhaegs/software/no-stow/gcc-8.3.0/lib64/libstdc++.so.6
#7  0x00000000004dacb0 in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_bracket_expression() ()
#8  0x00000000004d8bca in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_atom() ()
#9  0x00000000004d61b9 in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_term() ()
#10 0x00000000004d2514 in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_alternative() ()
#11 0x00000000004ce86a in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_disjunction() ()
#12 0x00000000004d8ab0 in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_atom() ()
#13 0x00000000004d61b9 in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_term() ()
#14 0x00000000004d2514 in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_alternative() ()
#15 0x00000000004d255e in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_alternative() ()
#16 0x00000000004d255e in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_alternative() ()
#17 0x00000000004d255e in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_alternative() ()
#18 0x00000000004d255e in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_alternative() ()
#19 0x00000000004ce86a in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_disjunction() ()
#20 0x00000000004ca36e in std::__detail::_Compiler<char const*, std::regex_traits<char> >::_Compiler(char const* const&, char const* const&, std::regex_traits<char>&, unsigned int) ()
#21 0x00000000004c5ed3 in std::shared_ptr<std::__detail::_Automaton> std::__detail::__compile<char const*, std::regex_traits<char> >(char const* const&, char const* const&, std::regex_traits<char>&, unsigned int) ()
#22 0x00000000004c277f in std::basic_regex<char, std::regex_traits<char> >::basic_regex(char const*, unsigned int) ()
#23 0x00000000004b7ed2 in __static_initialization_and_destruction_0 ()
#24 0x00000000004bb9b5 in _GLOBAL__sub_I_main ()
#25 0x00000000004f41dd in __libc_csu_init ()
#26 0x00007ffff6d6a4e5 in __libc_start_main () from /lib64/libc.so.6
#27 0x0000000000495509 in _start ()
```

Compiled with own compiled gcc-8.3.0 and then the program runs fine on CentOS7.
Maybe better to not try support such an old gcc version ?

Fatsie, 2020-08-05:
Tried now the different devtoolset versions of [software collections](https://www.softwarecollections.org/) and they all seem to generate proper executables. This is g++ v6.3.1 in devtoolset-6, g++ v7.3.1 in devtoolset-7, g++ v8.3.1 in devtoolset-8 and g++ v9.3.1 in devtoolset-9.
It's OK for me to close this issue.

trabucayre, 2020-08-06:
Thank for these tests. Really interesting! I must update the README to add this information.
It's true, I'm not sure It's a good idea to try support for this old gcc
Thanks again

closed 2020-08-06 by trabucayre