trabucayre/openFPGALoader #42: Arty S7-50 support

Fatsie, 2020-08-02:
I have an Arty S7-50 but the FPGA is not recognized by openFPGALoader from open-tool-forge. I do think the following diff should add the FPGA:
```diff
diff --git a/src/part.hpp b/src/part.hpp
index e81095a..c686ac9 100644
--- a/src/part.hpp
+++ b/src/part.hpp
@@ -16,6 +16,7 @@ static std::map <int, fpga_model> fpga_list = {
 
        {0x44008093, {"xilinx", "spartan6", "xc6slx45"}},
        {0x03620093, {"xilinx", "spartan7", "xc7s15ftgb196-1"}},
+       {0x0362f093, {"xilinx", "spartan7", "xc7s50csga324-1"}},
 
        {0x020f30dd, {"altera", "cyclone 10 LP", "10CL025"}},
 
```
But due to #41 I can't test ATM.

trabucayre, 2020-08-04:
This addition looks good for me. Do you prefer create a PR with this, or I add this line by myself?

Thanks

Fatsie, 2020-08-04:
Got it compiled now with gcc 8.3.0 and will do some testing with some bitstream and then do a PR.

closed 2020-08-04 by Fatsie

trabucayre, 2020-08-06:
Thanks.
Testing bitstream loading in memory is easy. But to store bitstream in the SPI flash it's a bit less easy since you need a bitstream to do bridge between jtag and SPI.
spiOverJtag directory provides bases but not tested with spartan7 (my seedstudio spartan edge has no flash...)

Fatsie, 2020-08-07:
Indeed, bitstream loading in memory worked with the patch as is. Am now working on SPI flash support. I got spiOverJtag bitstream now generated for the xc7s50, need to test it.
What I don't understand is that the spiOverJtag bitstream file name is not package specific but for Vivado compilation a part with a certain package is specified and the SPI pins are mapped to certain pins in the package. Shouldn't the spiOverJtag bitstream file be package specific ? Or will the SPI IO in the bitstream file be valid for all FPGAs of the same type but with different package ?
If it needs to be package specific, then the next question is how to find the package of the FPGA, the JTAG IDCODE does not contain this information AFAIK.