trabucayre/openFPGALoader #44: Flashing ProjectTrellis ECP5_evn multiboot example failes

feldim2425, 2020-08-11:
I recently tried to flash the [ECP5 multiboot example](https://github.com/YosysHQ/prjtrellis/tree/master/examples/ecp5_evn_multiboot) on in the ProjectTrellis Repo for the ECP5 evaluation board (LFE5UM5G-85F-EVN) but it failed during the "Refresh" step. 
I used the following command:
``openFPGALoader -c ft2232 -b ecp5_evn -f -v multiboot.mcs`` 
```
Refresh: FAIL
displayReadReg
	Config Target Selection : 0
	SPIm Fail1
	Preamble ERR
	EXEC Error
```

I also tried flashing the ``blinky1.bit`` and ``blinky2.bit`` Bitstreams using the same command (also to the onboard SPI Flash) which worked without problem. 
I was also able to flash the ``multiboot.mcs`` file from ProjectTrellis with the official Lattice Diamond programmer util also without any problems.

feldim2425, 2020-08-13:
Ok. I figured it out. It seems like Lattice Diamond programmer ignores the preamble in the mcs file and adds it's own or does some error correction. The problem is that the ProjectTrellis Multiboot example removes all 0xFF bytes at the beginning and that messes up the preamble. Simply removing the ``-unfill 0xFF`` parameter in the Makefile fixes it.

closed 2020-08-13 by feldim2425

trabucayre, 2020-08-17:
Thanks for this information.
Maybe I need to check if this preample is present (or not) and adapt content accordingly.
I suppose it's the better approach to avoid this issue