trabucayre/openFPGALoader #45: raw FLASH option needed

emard, 2020-09-01:
HI

Is there some some option to flash given file unmodified directly to flash chip,
starting from some offset. It's OK for me if offset is multiple of 4K
or 64K what I think is erase block size.

trabucayre, 2020-09-02:
Hi,
all prerequisites are present (-o / --offset to provides base address and I've fixed a bug in spiFlash the last week) but this option was only tested with cyc1000 board.
So, I need to create a design to check behaviour before unlocking this feature (for ECP5 in a first time).
Currently since erase is done by sector it's a 64k offset. But I must improving erase and flash algorithm to handle smallest size as possible.
In fact this feature is in my ASAP todolist.
Thank to motivate me to hurry about that ;)

emard, 2020-09-02:
There should not be any hurry. Of course my primary interest
should be ecp5-ft231x :) but I can also test and report what
happens on ft4232 based boards with ecp5-85, artix7-100, and cyclone5-49

trabucayre, 2020-09-02:
In fact the interface is not a problem. If it's work with one of them, it will work with all.
I need to enable this feature FPGA per FPGA since flash access is a bit different for each vendor:
- flash access is tranparent for anlogic eagle_s20 and ECP5 (it's only mandatory to write a specific instruction before)
- one bitstream must be loaded in RAM to provide a bridge for xilinx and cyclone devices. An in this situation it's mandatory to have as much as bitstream as FPGA model (artix, spartan6/7, cyclone 10/5) and size. Bitstreams are available for artix A35t and A50t).
- for the specific case of cyc1000 I use the second interface of FT2232 (interface B) to emulate SPI (bitstream is just connexion between pins on the Ftdi side and pins on the Flash side). I must updating this to have jtag<->SPI interface like for xilinx FPGA.

In a first time, I focus to ecp5 (colorlight and ecp5_evn).

emard, 2020-09-02:
I kinde understand everything mentioned.
I have a board with ecp5 and 256-mbit flash, the one with 4-byte
address and openFPGALoader works while I know it uses only
3-byte legacy addresses. That's great

I know for every artix anc cyclone, a different jtag-spi passthru
bitstream must be prepared. Not only this but there's more, cyclone5
or MAX10 i htink have some internal flash that is used to initialize
BRAM content so the programmer needs to handle this thing also.

trabucayre, 2020-09-03:
I know I need to improve spiFlash support to handle both 4 and 3-bytes address.

For artix and spartan7 se same implementation may be used (@Fatsie has added spartan7 support). For altera I need to first reworks the passthru to use jtag <-> spi instead.

With FPGAs with internal flash it's a bit more easy because in most case the flash is only dedicated for bitstream and it's not possible to add something else.

I have locally unlocked the ability to flash a raw data, a read after write show correct data but I'have a problem with my test design. I'm waiting to fix this before updating lattice support.

emard, 2020-09-03:
OK!

I have some jtagoverspi for artix7 that works for xc3sprog,
do you think it can work for openfpgaloader? I don't know how
can I actually try it? If I pass -f option, it still sends bitstream
to SRAM instead of FLASHing it

Raw option is great, sometimes there's need to write cpu
bootloader of vfat image on some higher addresses than bitstream

trabucayre, 2020-09-04:
Yesterday I'have fixed my design for my ECP5_EVN:
- an riscV is written at the beginning of the FLASH
- the CPU firmware is written at 0x200000
After fixing some issue it's works -> openFPGALoader is able for ECP5 to write something everywhere in flash \o/
Now it's time to cleanup openFPGALoader and the design before pushing it.

for [ECP5](https://github.com/trabucayre/openFPGALoader#lattice-ecp5-colorlight-5a-75b-lattice-ecp5-5g-evaluation-board-ulx3s) you have to known file format: bit and mcs.
- mcs is written in flash
- bit is loaded in RAM if -m (or nothing) and written in flash if -f

For the jtagofspi, xc3sprog bitstreams are not compatible. I use a piece of [HDL](https://github.com/trabucayre/openFPGALoader/blob/master/spiOverJtag/xilinx_spiOverJtag.vhd) I've written. But to support new xilinx FPGA you have just to update the [Makefile](https://github.com/trabucayre/openFPGALoader/blob/master/spiOverJtag/Makefile#L2) and add corresponding xdc file.

emard, 2020-09-04:
HI

I always like something simple, raw binary file and offset - a binary file
just written there without any conversion. Of course mcs parser is great
for tools that output such formats.

Also great would be an option to read flash content back to file,
very useful at troubleshooting.

When the new stuff is ready for ECP5 I will test

Thanx for info - xc3sprog is not compatible, a fresh spioverjtag
should be compiled so I started in this direction...
compiler needs significant disk free,
38GB free to download, 38GB free to unpack download, 100GB free
to install = total 1-2 days of setup is over.
After small vacation, I can proceed to compile spi-jtag bitstream
from source so no hurry there :)
When compressed with gzip -9 I expect to get some small file
of 3.3KB long :)

trabucayre, 2020-09-04:
It's true for read back mainly because it's possible to check write task. In fact I've already this (to read back content just after write) to verify the read everywhere.
If I have artix's details I'm able to update the Makefile to produce required bitstream if you want.
For gzip, as mentionned [here](47), It's not clear for me if it's for storage or to increase write speed task.

emard, 2020-09-04:
for artix7 board, this is the source that works for xc3sprog:

https://github.com/f32c/f32c/tree/master/rtl/proj/xilinx/ffm-a7100/ffm_a7100_jtag_spi_bridge

I'm a bit puzzled on why is it incompatible and how to convert
to spioverjtag

about spioverjtag*.bit.gz - yes it is just to significantly reduce storage size,
imagine you support say 10-20 different artix chips, each bitstream will be
uncompressed 2-4MB size and compressed 3-4KB size.
Speed difference at decompressing on stardard PC is so small,
few ms, nobody can notice.

emard, 2020-09-05:
I was experimenting with esp32 micropython programmer
using xc3sprog bscan and got it working, it uploads bscan7.bit.gz
first and then writes the flash. It first reads a flash block, compares
with to-be-written block content and decides if it must erase or write or
nothing, thus reducing flash wear and working faster.

Sometimes (about 1 in 20), after uploading bscan7,
flash write fails on first block (erase, write, verify
retried 3 times and give up). 

If you need some inspiration or reference, here it is:

https://github.com/emard/esp32ecp5/tree/master/rbp

trabucayre, 2020-09-05:
Okay I will seeing this. Thanks
In fact I have written my spiOverJtag in early time of this project. Maybe I must move to xc3sprog version. I will compare both.
For altera I need to find where I have stored this (but It has no real interest since it use interfaceB of the ft2232, and consequently limited to cyc1000). My idea is to rewrite this (or adapt the xc3sprog passthru) to use the JTAG interface.

I've updated the repository with support for writing raw data at an arbitrary offset (readback is currently missing).
My test design is an adaptation to [picosoc](https://github.com/trabucayre/picorv32)

emard, 2020-09-05:
Great, I will pull your code, compile and test.

About artix, I was looking at C source and I found
difference between spioverjtag for openFPGALoader
and xc3sprog:

xc3sprog, after SIR USER1 sends SDR 4 "magic" bytes
0x59,0xA6,0x59,0xA6
and then the bit-length and SPI data

While openFPGALoader code I didn't see that it sends magic bytes,
but maybe it actually sends them somehow but I have just overlooked C code

trabucayre, 2020-09-05:
My bridge has nothing to do with that of xc3sprog. This why both are incompatible.
If you read [xilinx_spiOverJtag](https://github.com/trabucayre/openFPGALoader/blob/master/spiOverJtag/xilinx_spiOverJtag.vhd) this piece of code just adapts JTAG signals to SPI.
I need to compare both speed to move to xc3sprog bridge.

emard, 2020-09-05:
I see!
xc3sprog is then probably not better than yours just different.
It has 2KB buffer, to issue USER1 first as magic bytes and
request and clock dummy bytes for future response,
and then issue USER1 without magic bytes and take the response
repeating that for each 2KB of data. I think xc3sprog "knows"
how to pipeline that and take response with 1 step delayed

trabucayre, 2020-09-22:
Sorry for delay.
I've updated xilinx support to write raw data in arbitrary address (tested again with picosoc) and added bitstream for xc7a100 device. If you can try it.
ASAP I must compare for bridge to see if I need to move to xc3sprog version. But usually SPI communication is quite small (for tx there are no overhead, for rx with a delay of one bit I need to read one byte more and reconstruct byte).

emard, 2020-09-22:
I'm currently away from artix7-100 board but in few days I could test
this I'm very curious to try how it works :)

This board has spansion 32MB flash and I think only 64K erase block
works. I have tried 4K erase block, datasheet says its supported but for
me it doesn't work reliable. Few 4K blocks OK but then I got some corruption

For 64K erase block, spansion works always ok

trabucayre, 2020-09-23:
OK. No problem. It's not hurry at all.
Currently, openFPGALoader use only 64K erase block. (I know it's sometime overkill and the algorithm needs to be improved).

emard, 2020-09-24:
I have artix-7 100 with ft4232 interface and compiled latest openFPGALoader
but what options do activate flashing? It prints Flash SRAM but it is
not writing
to flash but to SRAM

./openFPGALoader --cable ft4232 -f artix7.bit
load program
Flash SRAM: [===================================================] 100.000008%
Done

trabucayre, 2020-09-24:
To write a bitstream in flash you need to [convert](https://github.com/trabucayre/openFPGALoader/blob/master/README.md#spi-flash-only-for-arty) the bit to mcs.
In fact I've not tried to write a bit to flash (If it so, i'll update openFPGALoader to allows this)

emard, 2020-09-24:
but *.bit is simple - just write it raw to flash similar as lattice bit does

emard, 2020-09-24:
does -o option imply raw image writing?
(so I can write bit to flash with options:)

openFPGALoader --cable ft4232 -o 0x000000 -f artix7.bit

trabucayre, 2020-09-25:
I've tried with a .bit: It didn't work in flash. 
lattice .bit and xilinx .bit are not same...
Lattice .bit contains [instructions](https://prjtrellis.readthedocs.io/en/latest/architecture/bitstream_format.html) to drive the FPGA in configuration phase.
I suspect this is not true for xilinx .bit file this is why it's mandatory to use mcs (maybe .bin)
In openFPGALoader,  xilinx class don't use -f and -m. 
if file is .mcs -> FLASH
if file is .bit    -> RAM
if file is xxx    -> FLASH with a sanity checking about offset
It's true this mean a file with .bit can't be written at an offset > 0x00

emard, 2020-09-25:
Hmmm as for quick test it's not easy for me to setup complete
build env and compile all (45 minutes on normal PC just compilation with vivado)
I have recorded some *.bit files (maybe they are *.bin inside who knows).

I know something from my esp32ecp5 project. ESP32 would be very slow if it
had to parse bitstream, so I found best option is just to upload raw
bit content directly
to SRAM or FLASH addr 0 using SPI to JTAG in MSB first order.
It works great for xilinx and lattice. Same binary can be uploaded to SRAM or
flashed from address 0 and it also works.

For altera it works for SRAM if "bit" is extracted from *.svf
FLASH I don't know JTAG sequence how to write on altera yet, so I don't know
will such bit work.

Anyway I will try to convert my artix7 bit to mcs on some linux
lighter tool than
vivado and try again

emard, 2020-09-25:
Can I somehow use objcopy to convert?
I tried to convert .bin to .mcs using this:

objcopy -I artix7.bit -O ihex artix7.mcs

but openfpgaloader complains
rror: unknown type
Parse file FAIL

the bit converted to mcs file with objcopy starts like this

:1000000000090FF00FF00FF00FF000000161003158
:10001000616D6967615F66666D5F61373130303B86
:100020005573657249443D305846464646464646F5

emard, 2020-09-25:
So I renamed bit file to xxx and this seems to work and pass verify
only I'd like to be able to flash raw binary exactly at offset 0,
currently it accepts other offsets than 0

guest@brix4:~/src/fpga/ulx3s/openFPGALoader/build$ ./openFPGALoader --cable ft4232 -o 0x400000 /tmp/artix7.xxx
load program
Flash SRAM: [===================================================] 100.000008%
Done
parsing 3825900

file length 30607200
Parse file DONE
Detail: 
Jedec ID          : 01
memory type       : 60
memory capacity   : 19
EDID + CFD length : ff
EDID              : ffff
CFD               : 
Erasing: [==================================================] 100.000000%
Done
Writing: [===================================================] 100.000008%
Done

emard, 2020-09-25:
I can test something similar (same cable ft4232 and 32MB spansion flash)
with cyclone5, part 5CEBA4F23C7N

trabucayre, 2020-09-25:
After comparing .bit file from xilinx and mcs file from xilinx too, it's true the .bit file should work with the flash. I need to redo my tests, I'have missed something, but not now I'm not at home and hav'nt my arty with me.
For .mcs produces by using objcopy, I suppose my implementation don't cover all possible type... I've added this in my (long) TODO list.
For rawData at offset 0x0, I've (it's maybe an error) forbid this behavior considering only a valid bitstream could be write in this area.
I have never tested with CycloneV, I need to redo bridge (currently cyc1000 only). For SRAM you need to use a svf file. .sof file is just not documented, quartus do a magic thing before sending bitstream data...

emard, 2020-09-25:
What could be with bitstreams is 8-bit reverse of each byte.
For ecp5 and artix7 *.bit is file can should be shifted over JTAG as
MSB first, opposite as e.g. SVF commands which are shifted LSB first.

Bitstream has a header and footer with CRC so for lattice (and I guess artix)
it tolerates few bits but not too much like 1K of junk before
its beginning and any amount of junk after its end.

We can initially assume similar raw *.bit to flashing will work with altera.
Currently I have a python script that extracts *.bit from *.svf and at least
such thing works for cyclone5 SRAM. Original formats like *.sof and similar
are currently too difficult to understand.

trabucayre, 2020-09-25:
I've redo test and now it's possible to write a .bit in flash (In fact my previous tests have failed due to an inconscient in parser...)
Yes sof are too difficult... Tried to analyse this file but fails...

emard, 2020-09-25:
Great!

Besides *.sof, there is option to create *.rbf file what
means like "raw binary flash file"
but I tried and it's not raw enough to be directly flashed.

rbf file looks similar to extracted data from svf but it doesn't work
while svf works. Most obvious difference is that bits of *.rbf in each
byte are 8-bit reversed compared to that what is be extracted from *.svf
files begin the same but are not equal.

obviously, there's still something more that we don't understand on
those formats...

trabucayre, 2020-09-26:
I've planned to study .rbf. This format is used by Linux, for cycloneV Soc, to program the FPGA. I suppose this filetype is more or less documented (at least enough to be usable).
If it's possible to use them, it's maybe better (or simplier) than using .svf

emard, 2021-10-21:
Is altera (cyclone-5 + epc flash = 5CEBA4F23C7 + EPCS64) supported?
I see there's some code to load bridge and read/write flash.

there's possible misspelling at sram write, it prints message
"Flash SRAM" maybe it should print message "Write SRAM".

trabucayre, 2021-10-21:
altera cycloneV are supported to write SPI flash.
This model must be added in spiOverJtag. Have you a ref board? I could update accordingly.

emard, 2021-10-29:
> I have de10nano, de0nanoSo and de1Soc.
> I have produce a jic file from sof -> it's quite similar to rbf with an
> header. Need to check why some small diff.
> Having to load something into RAM before using spiOverJtag is really
> strange, currently I don't see why. Need to investigate.
> Using an rbf into flash it not always possible (You can't use one in
> compress mode -> real time decompression isn't supported by the FPGA in JTAG
> mode).

Strange, I have enable flash compression with option
(example only with .svf, other formats created similar)

        $(quartus_env); quartus_cpf -c \
        -q $(OPENOCD_SVF_CLOCK) -g 3.3 -n p \
        -o bitstream_compression=on \
        $(PROJECT).sof $(PROJECT).svf

and the compressed bitstream is shorter (2.6MB->0.7MB)
and works when programmed from .svf and .rbf to SRAM
but also as flashed from .jic and my method
(between 0xFF delimiters), here's python script that
converts .jic to .rbf
https://github.com/emard/fpga-flashrom/blob/main/proj/altera/ffm-c5a4-sd-lcdif/jic2rbf.py

trabucayre, 2021-10-29:
for svf the compression is, as far I know, not used.
I've started a parser for jic
But it's clear I can't reproduce your issue (and it's a situation I don't like). I need to check everything... I, definitively, have to find a board with same hardware.

emard, 2021-10-29:
I have quick and dirty workaround to this by updating spioverjtag.rbf
bridge two times in altera.cpp, then I don't need preparation with blink.svf

        /* first: load spi over jtag */
        try {
                RawParser bridge(bitname, false);
                bridge.parse();
                programMem(bridge);
                programMem(bridge);
        } catch (std::exception &e) {
                printError(e.what());
                throw std::runtime_error(e.what());
        }

emard, 2021-10-29:
Next fix is that I made overloaded programMem(); without
parameter and it has deleted code between /* write */ and /* reboot */

https://github.com/trabucayre/openFPGALoader/blob/master/src/altera.cpp#L85
https://github.com/trabucayre/openFPGALoader/blob/master/src/altera.cpp#L105

this is called like this just before bridge
programMem();
programMem(bridge);

You shourly have better idea how to code this cleanly,
but it's just my research what is needed for flashing
to work standalone.

trabucayre, 2021-10-30:
Tried with my de10nano -> success...
flash id is read successfully...
Ok I'm unable to program chip for an unknown reason -> maybe with the clip I will be able to dump transaction with a logic analyzer.

emard, 2021-10-30:
check also MSEL switch position, described on page 12
or other text. It has to be set to passive serial position
I guess should be 10010 or 10011, but as you got flash
ID correct then probably MSEL setting is OK.
Flash ID should be same as of W25Q128

http://www.terasic.com.tw/cgi-bin/page/archive_download.pl?Language=Taiwan&No=1046&FID=1c19d1d50e0ee9b21678e881004f6d81

On some boards and chips (ulx3s with ISSI) when we
want to reliably program in standard SPI mode, then
quad lines HOLDn and WPn must be driven 1. It is
not enough if lines are pulled up with resistors,
because hi freq SPI traffic capacitive crosstalk will apply
noise to HOLDn and WPn and in practice ISSI flash
doesn't work if not driven 1. Winbond doesn't need
drive 1 and in practice works with only resistors on ulx3s

emard, 2021-10-30:
quartus_cpf conversion of .sof to .rbf works only for SRAM but not for FLASH

I have improved script to convert .jic to .rbf
https://github.com/emard/fpga-flashrom/blob/main/proj/altera/ffm-c5a4-sd-lcdif/jic2rbf.py

resulting .rbf file is 487 bytes shorter than .rbf from quartus_cpf
and works for both SRAM and FLASH
bitstream compression is used.

-rw-r--r-- 1 guest guest 764612 Oct 30 12:50 project.rbf
-rw-r--r-- 1 guest guest 764125 Oct 30 12:49 project_flash.rbf

trabucayre, 2021-10-30:
it's maybe time to open an dedicated issue, this thread become very long :-)
My board is currently in AS mode (but I have tried all modes)
I can't set holdn and wpn since I use intel IP (serial flash loader).
I need to finalize my jic parser and to integrate this. But it's really weird, I have flashed many rbf this morning to verify behaviour with cylone10, cycloneIV and cycloneV (not socfpga) without any problem. Maybe something with quad/dual/single mode?

emard, 2021-10-30:
OK!

I opened new issue and there's link to my slightly improved .jic
parser. I have also done some attempts for esp32 micropython
to program altera and there are even more problems with
spioverjtag bridge not starting

Mecrisp, 2021-11-23:
I am trying to write a raw binary file, too, but despite the -o option given, openFPGALoader does not accept random data. Do I need to provide additional options to enable this?

`openFPGALoader --board=ulx3s -f -o 0x200000 LICENSE -v`

```
write to flash
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
found 1 devices
index 0:
        idcode 0x41113043
        manufacturer lattice
        model  LFE5U-85
        family ECP5
        irlength 8
File type : LICENSE
Error: Failed to claim FPGA device: incompatible file format

```

```
openFPGALoader -V
openFPGALoader v0.5.0

```

emard, 2021-11-24:
I think option "--file-type bin" is required to avoid content check
and flash raw binary data

trabucayre, 2021-11-24:
@Mecrisp Good catch! file type are deduced from extension, but without extension, and without `--file-type`, openFPGALoader is unable to decide what to do. It's  now fixed in master branch.
@emard true. It's a solution to avoid this error.

Thanks to all

Mecrisp, 2021-11-24:
Thank you for the quick solution - specifying the file type did the trick. Idea: Would be great if the help text could include the possible "file-type-specifiers". I tried with --file-type raw intuitively before, but that wasn't accepted.

This did the trick with the version already installed, will update later:
`openFPGALoader --board=ulx3s -f -o 0x200000 LICENSE -v --file-type bin`

emard, 2021-11-24:
Maybe file type "raw" should be treated equal as "bin" :)

trabucayre, 2021-11-24:
@emard exactly (and done this morning).
I have relaxed constrains about filetype:
- known format are treated using corresponding parser
- unknown format are allowed only if the file is wrtten in flash

now if you use:
```
openFPGALoader --board=ulx3s -f -o 0x200000 LICENSE -v --file-type bin
```
or
```
openFPGALoader --board=ulx3s -f -o 0x200000 LICENSE -v --file-type raw
```
or 

```
openFPGALoader --board=ulx3s -f -o 0x200000 LICENSE -v 
```

Result (and behaviour) is same (or I have introduce a bug ;-) )

@Mecrisp it's true: message must be more explicit and more informative

trabucayre, 2026-01-01:
I think this issue may be closed now