trabucayre/openFPGALoader #46: Adding LCMX02-640HC

TheFlav, 2020-09-02:
I'm using a LCMX02-640HC for a project.  My main machine is a Macbook, but I run Lattice Diamond under a Windows VM.  I'd love to be able to program the chip using the Mac without booting the VM.

I added the following line to part.hpp
	{0x012b9043, {"lattice", "MachXO2",   "LCMXO2-640HC"}},

That seems to have made it work.  I can't fully test the part at the moment, but it acts like the process is completing and the chip is probably working (very simple test).

Also, the only other main hurdle I had to overcome to get it to compile was to install libftdi1.  brew didn't have it, but MacPorts did.

sudo port install libftdi1
After that, I could configure and build it.  I could not build a static executable, but dynamic linking seems fine.

Also, FYI, there are a few more parts listed here, in case you care.
https://www.retrotronics.org/svn/jride/trunk/machxo2_internal.h

jeanthom, 2020-09-03:
Hi,

Homebrew's package for libftdi1 is called `libftdi` ([formula link](https://formulae.brew.sh/formula/libftdi))

trabucayre, 2020-09-03:
Thanks for your comment.
A soon as you have validated openFPGALoader with the LCMX02-640HC I will added this part officially.
Thanks @jeanthom for the information.
macOS compatibilty has been done by @kost, In fact I have only linux OS to test (no apple nor windows OS). If you could provide details to build openFPGALoader with macOS I will update the README to provides missing instructions (you can create a PR if you want).

The same thing must be done for windows.

TheFlav, 2020-09-03:
@jeanthom, My homebrew seemed to show that libftdi was libftdi0.  I'm not sure what was up, then.

I didn't have to modify anything to get it to build.  It was just getting libftid1 installed.  I had originally wanted to create a statically linked executable, but that didn't work (missing -lcrt0.s or something like that).  I removed the -D for static, and it was fine after that.

I will try to validate the LCMX02-640HC soon.  If you don't hear back from me in a week or so, feel free to ping me again.  I had to burn a bunch of LCMX02-640HC yesterday, but that was before I got this working.  It'll be a bit of time before I have to run through the process again, but next time I'll be happy if I don't have to boot the VM.

trabucayre, 2020-09-04:
open-tool-forge provide binaries and scripts for many tools and openFPGALoader is included into this. I will use these informations to add instructions for building openFPGALoader for macOS and windows (with a big "need to check").

No problem to wait a bit for confirmation about LCMX02-640HC.

trabucayre, 2020-09-22:
ping ;-)

TheFlav, 2020-10-28:
We have been using this under Mac to program many LCMXO2-640HC recently.  It seems to work just great!

trabucayre, 2020-10-29:
Great news !
I've updated part.hpp to add this device.
It's possible for you to update [README.md](https://github.com/trabucayre/openFPGALoader/blob/master/README.md) with instructions to install openFPGALoader in Mac?

Thanks again!

closed 2021-01-06 by trabucayre