trabucayre/openFPGALoader #48: Cannot program Colorlight with usb-blaster

ranzbak, 2020-09-05:
I'm trying to program my ColorLight-5a-75b with OpenFPGALoader, but I get the following output:

```
$ openFPGALoader -c usb-blaster -v blink.bit

found 1 devices
idcode 0x41111043
manufacturer lattice
model  LFE5U-25
family ECP5
File type : bit
Open file blink.bit DONE
Parse file DONE
Lattice bitstream header infos
Part: LFE5U-25F-6CABGA256
IDCode : 3000000
displayReadReg
	Config Target Selection : 0
	Overflow ERR
Enable configuration: FAIL
displayReadReg
	Config Target Selection : 0
	No err
```

I tried Googling around and it seems a similar bug has been fixed: [bug](https://github.com/trabucayre/openFPGALoader/commit/23b8e61bc01ca9cc6eb69efe0606cc0d1bd6f141)

I can flash the FPGA with urjtag, using svf files.

trabucayre, 2020-09-06:
It's strange in my side:
- with the de0nano board (onboard usb-blaster) everything work well
- with my external usb-blasterI and a spartan7 or an ECP5: openFPGALoader is unable to detect the FPGA.
I suspect my setup to be wrong (bunch of wire between the usb-blaster's connector and FPGA). I will doing a clean adapter before continuing to try to find an issue in the code.
Thanks to point this issue

trabucayre, 2020-09-06:
Okay.
I've fixed my hardware (it's just cleaner but it's now work).
With spartan7 I've fixed an issue -> it's work again
With ECP5 it's more strange. Seems to be unstable. Sometime idcode is okay, sometime no.
And now, after a wrong electrical connection my colorlight is just dead.
Need to order a new one before continuing...

ranzbak, 2020-09-09:
Okay, is there something I can do with my hardware to help the debugging process?

ranzbak, 2020-09-09:
I've tried it with the latest source code, I still get an error, but a different one this time.

```
openFPGALoader -c usb-blaster --bitstream blink.bit
Open file blink.bit DONE
Parse file DONE
Enable configuration: problem -1 written 64
problem -1 written 65
```

trabucayre, 2020-09-12:
Sorry for the delay.
Waiting to receive my new colorlight I've tried with an ecp5_evn. First interresting thing is no more random idcode, I suspect pullup and pulldown are mandatories.
But I'm unable to load bitstream. I need to investigate.
The most surprisingly this is: I have tested usb-blaster with 3 different ECP5 based board succesfully and now It's not working anymore.

For your second problem have you connected both GND ? Have you access right for the probe?

trabucayre, 2020-09-12:
I've fixed an issue, now openFPGALoader works with ecp5_evn, but I want to wait to have finish a serie of test (orange crab, spartanEdge and de0nano) before pushing this to avoid noise in the repository.

ranzbak, 2020-09-12:
Is there a way I can have your current changes? I would like to confirm that it works for my hardware setup too.

trabucayre, 2020-09-12:
in https://github.com/trabucayre/openFPGALoader/blob/master/src/usbBlaster.cpp#L264
change:
```
_in_buf[_nb_bit++] = mask | mode;
```
by
```
_in_buf[_nb_bit++] = mask;
```
It's works for ecp5_evn (pull resistors) but not for orangecrab (not pull resistors) and introduce an issue for spartanEdge.
I review the full implementation.

trabucayre, 2020-09-12:
I've push my full fix for usb blaster.
Tried with spartan7, 2 different ecp5, de0nano.
Could you try with colorlight please.

trabucayre, 2020-09-12:
@ranzbak

ranzbak, 2020-09-13:
Hi, 
thank you so much! 
I can confirm that programming works now.
No more errors or warning.
The issue can be closed, if that's okay with you.

trabucayre, 2020-09-13:
Great news !
Thank for your tests.
It's okay to close this issue.
Thanks again

closed 2020-09-13 by ranzbak