trabucayre/openFPGALoader #49: Tang Nano stuck at 99%

pepijndevos, 2020-09-16:
I'm trying to program the Tang Nano but it gets stuck at 99%
```
$ openFPGALoader -b tangnano ../pack.fs 
Parse ../pack.fs: 
00
Done
erase SRAM Done
Flash SRAM: [==================================================] 99.902534%^C
```

My git reflog looks as follows:
```
a220226 (HEAD -> master, origin/master, origin/HEAD) HEAD@{0}: pull: Fast-forward
7a8b80d (origin/cxxopts_migration) HEAD@{1}: pull: Fast-forward
414e6ee HEAD@{2}: clone: from https://github.com/trabucayre/openFPGALoader.git
```

The oldest commit does not support the Tang Nano, and neither of the newer ones work. Now that I think about it I'm not sure I ever used OpenFpgaLoader with Tang Nano or only with the Trenz GW1NR-9 board.

pepijndevos, 2020-09-16:
I have tested with another bitstream and precompiled binary with the same results.
I have attached a strace and bitstream in case it is helpful.

[pack.zip](https://github.com/trabucayre/openFPGALoader/files/5232431/pack.zip)
[strace.txt](https://github.com/trabucayre/openFPGALoader/files/5232402/strace.txt)

trabucayre, 2020-09-16:
could you try with -v (verbose)
I suspect you will see long list of pollFlag: xxxx
If I'm true it's not a regression but the FPGA in a wrong state.
I need to to find how to unlock the FPGA (it's done by gowin programmer) but datasheet has missing informations for this topic.

pepijndevos, 2020-09-16:
For reference, I tried it and the FPGA finished programming.
Apparently it was indeed in a wrong state.

closed 2020-09-16 by pepijndevos

TheFlav, 2021-06-22:
I have this problem.  If I use the -v, I do get a long list of pollFlag messages.  Is there a workaround?  I thought that maybe updating to a new ch552 firmware would help, but I don't think it has (or I have the wrong one).  Any ideas?

trabucayre, 2021-06-23:
I have lately tested different version of ch552 firmware with an external MCU, behaviour is more or less the same...

But according to your EDIT and apicula's issue I suspect an incompatible bitstream (built for a different device).
A check for device and bitstream IDCode must be added and load the fs only if it's targeted for connected FPGA.