trabucayre/openFPGALoader #53: compilation error

j75, 2020-09-29:
When building on Ubuntu 20.04 I encounter the following error:
```
...
[ 95%] Building CXX object CMakeFiles/openFPGALoader.dir/src/bitparser.cpp.o
[100%] Linking CXX executable openFPGALoader
/usr/bin/ld: cannot find -ludev
/usr/bin/ld: cannot find -ludev
collect2: error: ld returned 1 exit status
```

My build incantations are:
1. cd build ; cmake -DBUILD_STATIC=ON -DCMAKE_INSTALL_PREFIX=/usr ../
1. make -j$(nproc)

Any ideas how to solve this?

trabucayre, 2020-09-30:
With ubuntu (and debian) there are no libudev .a
The annoying thing is this error shows a problem with the CMakeFiles
To work around is to add `-DENABLE_UDEV=OFF` to disable udev.

j75, 2020-09-30:
Well, in my case I succeeded by leaving out the `-DBUILD_STATIC=ON` option...

trabucayre, 2020-10-16:
Sorry for the delay.
I suggested the solution to disable udev because you had explicitly requested a static build, but this is an other solution yes.
UDEV is used to open a cable by using /dev/ttyUSBx instead of using vid, pid (and serial)
I must add informations about each option in the README, it's true.

j75, 2020-10-21:
I have actually disabled the static buildin option and it worked.

Thanks for helping !

closed 2020-10-21 by j75

trabucayre, 2020-10-22:
I need to find a work around to force disabling udev, in static mode, when the system didn't provide needed .a
Thank to point this issue