trabucayre/openFPGALoader #54: Broken command line behavior

jadafi, 2020-10-06:
Since 99929f99c37590be451d092e504ae341602380bb I have had problems with using cables different from ftdi, as it throws the error "Error: FTDI serial param is for FTDI cables." Even without FTDI serial param specified.
Regards JD

jadafi, 2020-10-07:
fixed by changing 	
```struct arguments args = {false, false, false, 0, "", "", "-", "", -1, 6000000, "-",
			false, false, false, false, false, true, false, false};```
to
```struct arguments args = {false, false, false, 0, "", "", "-", "-", -1, 6000000, "-",
			false, false, false, false, false, true, false, false};```
in line 73 of main.cpp
https://github.com/trabucayre/openFPGALoader/blob/master/src/main.cpp#L73

trabucayre, 2020-10-07:
Thank to point this error!
Pushed a fix

trabucayre, 2020-10-16:
It's okay for you?
Can i close this issue?

closed 2020-12-09 by trabucayre