trabucayre/openFPGALoader #58: Lichee Tang - Fails to open device

naxola, 2020-10-21:
Hi!,

I am using an JTAG-USB with FTDI FT2232D chip inside. 

Typing:
>  _$ lsusb_

result in:

> ...
>    _Bus 003 Device 011: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC_
> ...

So it is recognized. However, when I set the command:

> _$ openFPGALoader -f -b licheeTang /home/nacho/FPGA/Tang_FPGA_Examples/2.LCD/test_lcd.bit_

results in:

>    _fails to open device
>    Error: Failed to claim cable_

Any hints on what could the problem be?

Thanks in advance.

naxola, 2020-10-21:
Sorry, I didn't say before... I am using a Lichee Tang Primer board.

Adding this command:

> openFPGALoader -c ft2232  /home/nacho/FPGA/Tang_FPGA_Examples/2.LCD/test_lcd.bit 

Results in

> Error: device 10e31913 not supported

I don't know what it could be... any help?

trabucayre, 2020-10-21:
Hi,
For the first issue, I suspect a missing access right to the cable, in fact if the board has a default cable the -c option is not used (it's now fixed and udev rules updated).
I have an lichee Tang Primer (but to be short It's called licheeTang) too. The weird thing is the idcode. It's not one for eagle device. 
The lichee Tang has an onboard jtag interface maybe there is a conflict between your ft2232d and this one.
Could you try with official cable to see if it's work... Or not ?
Thanks

naxola, 2020-10-21:
Thanks trabucayre.
If it helps, I installed Anlogic IDE in ubuntu 20, but no cigarr.... It doesn't recognize licheeTang fpga...

I am stuck at this point but I don't know what else I can do... 
what do you mean with "_I suspect a missing access right to the cable_", maybe I am not connecting well the jtag cables with lichetang? My connection is (could you review if it is ok?):

**Tang pin - Usb Jtag**

> A4 - TDO
> C5 - TCK
> B6 - TDI
> C9 - TMS
> G - GND
> H13 - TXD
> J13 - TXD

I am using the official usb-jtag/ttl connected to pins on licheetang (see pictures below), and as you can see rgb led is blinking so board seems to be working fine. 

[image]
[image]

trabucayre, 2020-10-21:
In your first message you show a call to openFPGALoader with `-b licheeTang`
This board, as shown [here](https://github.com/trabucayre/openFPGALoader/blob/master/src/board.hpp#L50) has an onboard jtag adapter (the device near the USB connector). If udev [rules](https://github.com/trabucayre/openFPGALoader/blob/master/99-openfpgaloader.rules#L22) are not updated to allows you to claim the devices it's not possible to continue.

The second point is: FPGA JTAG pins are connected to the device near USB, I'm not sure if it's possible to use an external probe.

Could you try to unplug your external probe, keep USB connected and using:
```
openFPGALoader -b licheeTang --detect
```
To see if it's works.

naxola, 2020-10-21:
**First test:**
I connected from my pc USB directly to FPGA board micro USB. The RGB starts to blink. Then I write the command but had the same result, as if nothing were connected:

```
openFPGALoader -b licheeTang --detect
fails to open device
Error: Failed to claim cable
```
If I connect and disconnect usb and check some changes with lsusb, no changes happens.

**Second test:**
Just connected USB-JTAG/TTL probe to pc (and not wired to the licheetang board) and the same result as before.
If I connect and disconnect USB-JTAG/TTL and check some changes with lsusb, I detected FTDI usb.

I checked rules file is copied to _/etc/udev/rules.d/_ and my _user_name_ belongs to the _plugdev_ group.

naxola, 2020-10-21:
## Solved!!!!
This is what I did if may help to anyone.
1. I changed to USB3.0 port licheeTang directly connected to PC
2. Check if there were changes in usb devices with:
`lsusb`
With results in:
`...
Bus 003 Device 008: ID **0547:1002** Anchor Chips, Inc. Python2 WDM Encoder
...`
3. Edit 99-jopenfpgaloader.rules
`sudo nano etc/udev/rules.d/99-jopenfpgaloader.rules`
4. Add new rule:
`Original GigaDevice
ATTRS{idVendor}=="0547", ATTRS{idProduct}=="1002", MODE="664", GROUP="plugdev"`
5. Restart udev service
`service udev restart`
6. Now I can use openFPGALoader commands

Many thanks for your support and help trabucayre

closed 2020-10-21 by naxola

trabucayre, 2020-10-22:
Happy to see it's work for you.
My fault, 99-openfpgaloader.rules, until your second post was not up to date. Now this file provide rule for the anlogic cable.
Thanks