trabucayre/openFPGALoader #66: Add a donation link to the README

jfoucher, 2020-12-03:
Hi @trabucayre ,
Thank you for this software. I suggest you add a donation link to the readme for people who would like to support your work on this project.
You work has helped me avoid buying a rather costly programmer and I would like to thank you somehow.
Thanks

trabucayre, 2020-12-08:
Hi @jfoucher 
Thank for this proposition, I need to think about those.
Okay sometime it's a bit expensive (2 times I needed to buy a board to help an user, one time I've received a gift (thank again @jeanthom for the [DirtyJTAG](https://github.com/jeanthom/DirtyJTAG))) but the most important thing it to provide a software, independant to the platform and OS, able to works with all FPGAs, allowing user to bypass vendor tools.
I need to check solution for donation but I'm not convinced by monthly donate in the specific focus of openFPGALoader.

shreekumar3d, 2021-02-03:
+1 to @jfoucher . This project has already saved me much trouble, but surprisingly it was so hard to find !  I am using it with the Lattice ECP evaluation board, btw.

I would like to do a one time donation for sure.  I will also write up about this somewhere so that this project gets more visibility.

trabucayre, 2021-02-04:
> +1 to @jfoucher . This project has already saved me much trouble, but surprisingly it was so hard to find ! I am using it with the Lattice ECP evaluation board, btw.

Thanks. It's the second [time](https://github.com/trabucayre/openFPGALoader/issues/73) this week someone said about the difficulty to find this project. I really needs to do efforts to fix this (need to find the best way). In fact I'm a bit surprise because since it's in buildroot, archlinux, freebsd, [fpga-toolchain](https://github.com/YosysHQ/fpga-toolchain), mentionned as reference tool to program gowin FPGA in @pepijndevos apicula, I've believed a more visible status. I'm wrong and need to resolve that.

> I would like to do a one time donation for sure. I will also write up about this somewhere so that this project gets more visibility.

I don't know any solution for one shot donation. If someone know I'm open.
Thanks again

shreekumar3d, 2021-02-04:
@trabucayre  Does PayPal work for you ? I don't know any other way unfortunately...

> a more visible status

Certainly.  One way involves adding a reference to the tool in relevant projects & samples.  That way folks will know that the tool exists & how to use it. E.g., see #https://github.com/YosysHQ/prjtrellis/issues/162

trabucayre, 2021-02-06:
> @trabucayre Does PayPal work for you ? I don't know any other way unfortunately...

It's works but I don't know if it's a good idea to display paypal informations. Unfortunately systems like patreon or github sponsors are monthly pay instead of one shot... I search...

> Certainly. One way involves adding a reference to the tool in relevant projects & samples. That way folks will know that the tool exists & how to use it. E.g., see #[YosysHQ/prjtrellis#162](https://github.com/YosysHQ/prjtrellis/issues/162)

Thanks for this "ad" and it's true. I'm certainly wrong again, but I'm a bit unconfortable with the idea to "brute force" all projects to adopt this tool.