trabucayre/openFPGALoader #68: Segmentation fault (core dumped)

jiv4ik, 2020-12-15:
Hi!
When i try to program TangNano, i see:
$ openFPGALoader -m -b tangnano  led_prj/impl/pnr/led_prj.fs -v
found 1 devices
idcode 0x900281b
manufacturer Gowin
model  GW1N-1
family GW1N
File type : fs
Parse led_prj/impl/pnr/led_prj.fs: 
xBackgroundProgramming: OFF
xCRCCheck: ON
xCheckSum: 0xB0ED
xCompress: ON
xCreated Time: Tue Dec 15 11:31:02 2020
xDevice-package: GW1N-1-QFN48
xEncryption: OFF
xFile Title: Bitstream file
xGOWIN Version: V1.9.7Beta
xJTAGAsRegularIO: OFF
xLoadingRate: 2.667MHz
xPart Number: GW1N-LV1QN48C6/I5
xSPIAddr: 0x00fff000
xSecureMode: OFF
xSecurityBit: OFF
xUserCode: 0x0000B0ED
Segmentation fault (core dumped)

$ dmesg 
[11730.387256] openFPGALoader[21674]: segfault at 55ab63abb000 ip 000055ab61c4edb3 sp 00007ffd46f43ad0 error 4 in openFPGALoader[55ab61ac3000+1d7000]
[11730.387262] Code: 00 00 83 bd f0 fe ff ff 0f 7f 61 8b 95 f0 fe ff ff 8b 85 ec fe ff ff 01 d0 89 c2 48 8d 45 80 48 89 d6 48 89 c7 e8 1d 8b f5 ff <0f> b6 00 3c 31 75 07 b8 01 00 00 00 eb 05 b8 00 00 00 00 66 89 85

trabucayre, 2020-12-15:
Hello,
You use the master branch?
Could you share your fs? I have tried with many bitstreams without being able to reproduce this issue.
Thanks

jiv4ik, 2020-12-16:
Yes, I use master 
$ git log
commit 6e4479767755e77585fe35803f9017773d139e00 (HEAD -> master, origin/master, origin/HEAD)
Author: Gwenhael Goavec-Merou <gwenhael.goavec-merou@trabucayre.com>
Date:   Tue Dec 15 09:47:10 2020 +0100

    add fairwaves xtrx pro board

I build loader with  -DLINK_CMAKE_THREADS=ON

[led_prj.zip](https://github.com/trabucayre/openFPGALoader/files/5700674/led_prj.zip)

trabucayre, 2020-12-16:
Thanks.
I'm able to reproduce your issue.
data part lines have not all the same length, I suspect this is related to compress option (never tested).
Need to update the parser to take this option into account.
Could you retry without compress option to check if my assumption is true? (my gowin license is no more valid and I'm waiting for a new one).

Thanks again.

jiv4ik, 2020-12-16:
Thanks!
You are right!
Loader works fine without compression!
Thanks for your work! Your loader is the only tool that helped me program TangNano!

closed 2020-12-16 by jiv4ik

trabucayre, 2020-12-16:
Great to see it's possible to have a workaround! But this issue demonstrate a partial support for fs format.
For the specific case of the tangNano, the programmer interface is a bit unstable, I have already done some workaround to load bitstream in RAM but it's not working to write flash.