trabucayre/openFPGALoader #69: openFPGALoader does not work with some bitstreams

kammoh, 2020-12-22:
```
$ openFPGALoader --board ulx3s --bitstream  ./build/ulx3s/gateware/top.bit -v
ret 0
found 1 devices
idcode 0x41113043
manufacturer lattice
model  LFE5U-85
family ECP5
File type : bit
Open file ./build/ulx3s/gateware/top.bit DONE
Parse file DONE
Lattice bitstream header infos
Part: LFE5U-45F-6CABGA381
IDCode : 41113043
displayReadReg
        Config Target Selection : 0
        Std PreAmble
        ID ERR
        EXEC Error
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.000000%
Fail
displayReadReg
        Config Target Selection : 0
        JTAG Active
        ISC Enable
        Write Enable
        Read Enable
        Std PreAmble
        ID ERR
        EXEC Error
```

[top.bit.zip](https://github.com/trabucayre/openFPGALoader/files/5731303/top.bit.zip)

Programming a small "blinky" bitstream is successful though.

trabucayre, 2020-12-22:
According to the log your ulx3s has a LFE5U-85 but the bitstream is targeted for a LFE5U-45F. I'm not sure it's allowed.
Could you check if configuration is the same for this design and for the blinky ?

Thanks

trabucayre, 2022-02-12:
It's okay for you?
Can I close this issue ?
Thanks

kammoh, 2022-02-12:
Oh sure! Sorry I’d forgot to close it.
Thank you!

closed 2022-02-12 by kammoh