trabucayre/openFPGALoader #72: Altera ID codes?

udif, 2021-01-25:
While testing the Altera USB blaster support in openFPGAloader, I decided to plug in my old Cyclone III EP3C25 based board.
To my surprise, it identified the chip as a Cyclone 10CL025.
While checking the openFPGAloader docs, I ran into this note:

> Note: cyclone IV and cyclone 10 have same idcode. A WA is mandatory to detect correct model for flash programming.

Upon checking the Altera documentation, I saw that indeed EP4CE22, EP3C25 and apparently 10CL025, all share the same IDcode.
Checking the data sheet, all 3 chips have 594k bits of embedded memory, 66 multipliers, 4 PLLs, and 20 global clock networks.
Looking at the 3 families' Logic Element diagram reveals they are identical.

Could it be that all 3 devices share the same bitstream format, and the family difference is just a die shrink?
Actually, the Ep3C24 and the 10CL025 has the same number of LEs (24,624) but the EP4CE22 has slightly less LEs (22,320).

Also, does WA above stands for "Workaround"? Do you mean that while loading a bitstream may be identical, you can't program the flash withough knowing the exact device, because each device requires different bitstream file for programming flash devices that are connected to the FPGA?

trabucayre, 2021-01-26:
It's amazing 3 different FPGA generation with same idcode ! :-/
I don't know if bitstream are compatible, maybe people working on [mistral](https://github.com/Ravenslofty/mistral) could provide an answer.A workaround must be found to determine which bridge using without using idcode. I'm assume an CLI option is the only way to do that.
For the bridge itself, used to flash eeprom, it's only supported not really for 10CL025 but for cy1000. This bitstream use the ftdi interfaceB to connect directly SPI between this interface and flash. I need to do more or less same design as xilinx to have something more generic. It's in my todolist, but not already done.

udif, 2021-01-26:
From what I see by looking at Mistral, they concentrate only on Cyclone V.
Looking at the Cyclone V arch, it seems to be different (not using the same LAB, memories, and multiplier, PLL counts as above).
Checking the Cyclone V IDCODE they are indeed different.

The Altera IDCODE is split into fields:

| Bit Position | Value | Description |
|------|---------|------|
|0|  1|  "LSB" |
| 11:1| 001 0110 1110 | Manufacturer  |
| 27:12 |  Part Number| See table below|
| 31:28| 0000 | Version - so far all devices I've seen use 0|

Part Number table:
| Part Number | Family | Notes|
|-----------|---------|------|
|0010 0000 1111 0xxx| EP4CExx / EP3Cxx / 10CLxxx|
|0010 0111 0000 00xx | EP3CLSxx
|0010 1000 00xx xxxx | EP4CGXxx
|0010 1011 00xx xxxx | 5CE/GX/GT | GT devices share IDCODE with the corresponding GX device|
|0010 1101 00xx xxxx | 5SE/SX/ST | All devices with the same size 2/4/5/6 share IDCODE|
|0011 0001 1000 xxxx | 10Mxxx - Single supply | (10M02 U324 uses 10M04 code...)|
|0011 0001 0000 xxxx | 10Mxxx - Dual supply| 

Would you find any value in identifying specific components even if we can't program them at the moment?
Maybe we could also do hierarchical generic detection, based on bit masks:
```
0x00000fff:0x000000dd:Manufacturer:Intel
  0xf0000000:0x{X}0000000:Version:{X}
  0x0ffc0000:02800000:Family:EP4CGX
0x00000fff:0x0000002F:Manufacturer:MicroSemi
0x00000fff:0x00000043:Manufacturer:LatticeSemi
0x00000fff:0x00000093:Manufacturer:Xilinx
  0xf0000000:0x{X}0000000:Version:{X}
```
The indented lines would be parsed only if the level above them matches.

For Xilinx we have:
|version | family code | part size | manufacturer ID | 1|
|-|-|-|-|-|
|31-28 | 27 - 21 | 20 - 12 | 11 - 1 | 0|

So perhaps parsing based on Hex values is not a good idea, so maybe we could use binary or some form of regex notation.

References:
Cyclone III - https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/hb/cyc3/cyclone3_handbook.pdf -  Table 12-2, page 266
Cyclone IV: https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/hb/cyclone-iv/cyclone4-handbook.pdf - Table 20-2, page 260
Cyclone V: https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/hb/cyclone-v/cv_5v2.pdf - Table 9-1, page 305
Cyclone 10 - https://www.intel.com/content/www/us/en/programmable/support/support-resources/download/board-layout-test/bsdl/cyclone-10-lp.html - ID codes taken from the BSDL files
Max 10 - https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/hb/max-10/ug_m10_jtag.pdf - Table 4, page 7
Xilinx revision code - https://www.xilinx.com/support/answers/8265.html

trabucayre, 2021-01-27:
Since the only information I have about the device, trough JTAG, is IDCODE, I can in most case determine by the hex value which device is present. For most of them there is no ambiguity about the FPGA. I think for those sharring with the same IDCODE for different devices the simpliest way is to ''blacklist'' them. Ie if this IDCODE appear writting in flash can't be done unless provide through a new CLI arg the real model.
Current approach with xilinx work well, thanks to the respect of unique IDCODE, I'm able to automatically load a bridge in RAM to have access to the SPI flash and now I need to do the same for altera (as mentionned in your table, ambiguities are for a subset of devices).
For your idea to analyze IDCODE by area instead of comparing this is a good idea to simplify part.hpp description, I keep this in my idea list.

trabucayre, 2021-07-11:
spi flash is now supported for altera devices (cyclone IV, V and 10lp). It's required  to use `--fpga-part` with device type if the board is not officially supported by openFPGALoader.
`part` to pass has the form: `model size package` like `5ce223` for an `5CEFA2F23I7`

trabucayre, 2022-02-12:
It's okay for  you?
Can I close this issue ?
Thanks

udif, 2022-02-12:
Fine by me.

closed 2022-02-13 by trabucayre