trabucayre/openFPGALoader #73: Quiet option switch

eblot, 2021-01-29:
Would it be possible to add a `--quiet` option switch so that openFPGALoader does neither emit:
 - `=` progression chars on stderr
 - completion %-age on stdout
This would be very useful to use it in unattended context (such as CI/CD), as it generates a large amount of text which makes the output harder to parse and not useful it this kind of context.

Thanks a lot for this tool, it seems to overcome many limitations encountered with `xc3sprog`, it is a pity it is much harder to find for now. I hope it will soon move up in search ranks!

trabucayre, 2021-01-29:
> Would it be possible to add a `--quiet` option switch so that openFPGALoader does neither emit:
>
>     * `=` progression chars on stderr
>
>     * completion %-age on stdout
>       This would be very [...]
Yes of course. What is the required level of quiet (just progressBar or everything?)
> 
> Thanks a lot for this tool, it seems to overcome many limitations encountered with `xc3sprog`, it is a pity it is much harder to find for now. I hope it will soon move up in search ranks!
Thanks! I know I need to improve my communication ... :)

eblot, 2021-01-29:
It think it is ok to have the info and steps, but progression (both as bars and percentage) generates a lot of lines. I'd say, everything that requires emitting a CR / `\r` char :-)

>  I know I need to improve my communication ... :)

:-) I had a hard time finding this tool w. google, which is really a pity.

BTW, unrelated question: I'm using an Arty-A7 100T, sending the bitstream to RAM (not SPI flash). I'm a newbie with FPGAs but what does the "erase" step do when using RAM?

trabucayre, 2021-01-30:
I've pushed this feature. It's okay for you ?

> :-) I had a hard time finding this tool w. google, which is really a pity.
I know except for twitter I haven't done anything to promote this tool (and to improve search rank)... @Martoni has done more in this topic (blog + french magazine) than me.

> BTW, unrelated question: I'm using an Arty-A7 100T, sending the bitstream to RAM (not SPI flash). I'm a newbie with FPGAs but what does the "erase" step do when using RAM?

RAM must be blank before loading a new bitstream. This force the FPGA in a specific state and start with the new content.

eblot, 2021-01-30:
Perfect ! Thanks a lot.
Thanks also for the explanation of the flashing steps.

closed 2021-01-30 by eblot