trabucayre/openFPGALoader #74: Raspberry pi and cyclone10 board

CrackenDesigns, 2021-01-31:
Hi, I got a cyc1000 from Trenz attached to a raspberry pi 3 and tried your soft to send .svf files. After compiling and installing I can flash the test file sucesfully but at the end of the command the error "detach error -5" appears always. The test file seems to work  but all my generated cores with quartus doesn´t work, looks like the flashing is correct but always get "end of flash" followed by the "detach error -5". Any clue of this?

Thanks.

trabucayre, 2021-01-31:
Hi,
the "detach error -5" is unfortunatly normal. the FTDI interface is configured as jtag, and, consequently, the driver didn't attach this.
At the beginning, openFPGALoader detact the driver and a the end try to reattach the device, but this fails because the reattach can't be done in this situation.
Anyway if "end of flash" is shown it's okay, the bitstream is correctly loaded.

CrackenDesigns, 2021-01-31:
Hi, thanks for your reply. It looks like the bitstream is ok but doesn´t work, the board gets like if you just powered up. Of course these .svf are tested programming with the quartus programmer. Compared your test file with mine and they have the same header and end of file.

Thanks.

trabucayre, 2021-01-31:
The design works when programmed with quartus but didn't work with openFPGALoader?
If true it's weird because with svf, the task is to do is to follows instructions contained in the file.
The only main difference is: quartus use sof to load bitstream instead of svf.

CrackenDesigns, 2021-01-31:
Yes, it doesn´t work neither in some boards like banana pi, raspi and even with my laptop with ubuntu.
Your test file takes a little bit more time to flash than my files.

trabucayre, 2021-01-31:
Really weird. I, currently, works with my cyc1000 without issue. Which quartus version you use ?
when you talk about test file it's test_svf.svf (in fact it's a bridge to access eeprom through ftdi interfaceB).
Could you copy command you use to generate svf file ?
Thanks

CrackenDesigns, 2021-01-31:
I use the quartus programmer to generate the .svf file. The strange thing is that this .svf works without problem using urjtag svf command. Attached is the one I´m using.

trabucayre, 2021-01-31:
Ok. I convert the svf using cli:
```bash
quartus_cpf -c -q 12.0MHz -g 3.3 -n p file.sof file.svf
```

I'm unable to find your attached file...

CrackenDesigns, 2021-01-31:
Good news, looks like with the quartus .svf generator doesn´t work with openFPGAloader, urjtag works, but using the command line to covert does the trick and now works perfect.

Thanks for your help, I appreciate so much.

trabucayre, 2021-01-31:
Perfect!
I need to see the svf built by the GUI. If it's not works this mean something is wrong/missing in the svf Parser....
could you share your svf?

Thanks

CrackenDesigns, 2021-01-31:
Yes, of course I can share, just only testing for an addon we are working on for the cyc1000.
If you have telegram you can reach me @nabateo

Greetings.

trabucayre, 2021-02-06:
No sorry I haven't telegram.
It's however possible to attach a file to an issue. Is it possible to have access to the svf generated by quartus (not working) and the one from CLI (working).
I have tried to do this with one of my designs but files are identical...
Thanks

CrackenDesigns, 2021-02-06:
Wells I discovered that the files generated  by quartus for windows doesn't work. If you generate with the Linux version works like a charm, weird thing, isn't it?

Regards.

trabucayre, 2021-02-06:
Interresting! And maybe less weird as you think.
Windows usually use CRLN (\r\n) instead of LN (\n).
Consequently the '\r' is maybe left in buffer.
Again, can you provide one working file and one not working to check my assumption.
Thanks

CrackenDesigns, 2021-02-06:
Hi, here we go...
As I said, each one is generated with quartus for windows and quartus for linux.

Regards.

trabucayre, 2021-02-06:
Hi,
my problem is I have only linux computers, so I have no way to generate an svf with quartus for windows :-(

trabucayre, 2021-02-08:
Hi,
With the help of `unix2dos` utility I have converted an svf file, this file has confirmed the issue with CRLN format.
Now fixed.
Could you retry to confirm it's okay for you?

Thanks

trabucayre, 2021-07-23:
It's okay for you ?

CrackenDesigns, 2021-07-26:
Hi mate, it´s ok now, Good job, thanks.

closed 2021-07-26 by trabucayre