trabucayre/openFPGALoader #82: Add support for XC7A75T

zh522130, 2021-04-15:
can you add support for the Xilinx XC7A75T

now I use  openFPGALoader -f -c digilent_hs3 ./Test.mcs have error:

Error: device 13632093 not supported

Thanks!

trabucayre, 2021-04-15:
I have added the idcode in the list of known FPGA.
Could you try to load a bitstream in RAM to validated this step.
To be able to add a bitstream to bridge JTAG<->SPI I need qspi's pinout. Could you share schematic of your board ?
Thanks

zh522130, 2021-04-16:
I use openFPGALoader -c digilent_hs3 ./Proj_Top.bit also has same error:

Error: device 13632093 not supported

the config schematic below:
[image]

the real fpga is XC7A75T-2FGG484I

trabucayre, 2021-04-16:
Hi,
Have you git pull your repo? I've added this idcode yesterday, so now you may be able to detect this part and load bitstream in SRAM.
I will build a bridge, but I need to redo the approach since for one model, qspi pinout is not same for all packages.

Thanks

zh522130, 2021-04-19:
Thank you add 75t support!

Now I can use openFPGALoader -c digilent_hs3 ./Proj_Top.bit to program fpga,terminal show:

write to ram
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE
load program
Flash SRAM: [===================================================] 100.000008%

but when I use openFPGALoader -f -c digilent_hs3 ./test.mcs to program flash,It doesn't look like the download was successful，the terminal show:

write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE

and no more

trabucayre, 2021-04-19:
Great news for SRAM.
For QSPI, I know the bridge must be added, I'm currently working on an upgrade of this part of openFPGALoader, but I think this rework will ready tonight.
I must improve error messages to displays something usefull when no bridge is available for the specified device ...

zh522130, 2021-04-19:
Yes, it would be more user friendly to show the reason why you can't download, looking forward to the update tonight

trabucayre, 2021-04-19:
I've updated to repository with a message more explicit and an update to have spi to jtag bridge.
For your specific case, since there is no board you need something like
```
openFPGALoader -f --fpga-part xc7a75tfgg484 -c digilent_hs3 ./test.mcs
```

Let me know if it's ok for you.

Thanks

zh522130, 2021-04-20:
I tested with the command you gave me and everything worked fine, I was able to download the program successfully，Thank you

trabucayre, 2021-04-20:
Again it's a great news.
One new device fully supported ;-)

Can I close this issue?

Thanks again

closed 2021-04-21 by zh522130