trabucayre/openFPGALoader #83: Add support for Alchitry Au board

paller, 2021-04-16:
The Alchitry Au board is very similar to the Digilent Arty and Basys boards and should be easy to add support for.

paller, 2021-04-16:
I already verified it works by just using the board as an arty board, so I simple added a new line with
```c
JTAG_BOARD("alchitry_au", "digilent", 0, 0),
```
However, it also works if the cable is specified as "ft2232". As those two cables are defined differently I'm not sure which is correct.

trabucayre, 2021-04-16:
I've added this board but with ft2232 cable instead of digilent.

In fact digilent cable is a ft2232 converter with original vid/pid but with a buffer or level shifter driven by another pin to enable output. This why you able to use both.

The support is currently limited to memory, for flash I must redo bridge approach since qspi pins depends on package instead of model.
Thanks

paller, 2021-04-17:
How is the flash programmed?

When using the official tool to program the flash there's an animation running on the onboard LEDs, so I'm guessing something is loaded into memory that is bridging the jtag or uart to qspi?

trabucayre, 2021-04-17:
It's exactly that.

If you look at `spiOverJtag` directory you will see a set of bitstream. These are automatically loaded in the RAM to add bare minimal logic to connect the jtag interface to the qspi flash.

But I have done a mistake, currently you have a bitstream for each model (15t, 35t, 75t,...), but in fact you must have a bitstream for eache model + package. For exemple the qspi flash for arty and alchitry are not connected to the same pins.

Modifications are quite straightforward, I have my idea to do that (and in fact it's the preliminary step to be able to add qspi support for altera devices).

paller, 2021-04-17:
Let me know if there is something I can help with or test. I have the Alchitry Au and a Basys3 board.

trabucayre, 2021-04-17:
It's interresting.
I'm working to add everything needed to be able to determine exact model (version + pin_count). In current state nothing to test... But I think to order an alchetry to have more than one A35 with different package.

maybe try your basys in SRAM mode to validate support (I don't remember exact model and this board is not already supported).
Thanks

trabucayre, 2021-04-19:
I'have update the repository.
Now with:
```
openFPGALoader -f -b alchitry_au yourBitstream
```
You will be able to write your flash

Thanks

paller, 2021-04-20:
> maybe try your basys in SRAM mode to validate support (I don't remember exact model and this board is not already supported).

I wasn't too hopeful as `--detect` gave me nothing, but once again just setting it up as an Arty board did the trick for Basys3 as well.
```
./openFPGALoader -b arty ~/basys3_test/basys3_test.runs/impl_1/loadbar.bin
write to ram
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Open file DONE
Parse file DONE
load program
Flash SRAM: [==================================================] 100.000000%
Done
```

paller, 2021-04-20:
> ```
> openFPGALoader -f -b alchitry_au yourBitstream
> ```
Not sure if the bridge is supposed to be included or I have to compile them?
```
./openFPGALoader -b alchitry_au -f ~/fpga/build/top.bit
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a35tftg256.bit
Error: fail to open /usr/local/share/openFPGALoader/spiOverJtag_xc7a35tftg256.bit
```
I can confirm that writing to SRAM is working.

paller, 2021-04-20:
> Not sure if the bridge is supposed to be included or I have to compile them?

`make install` solved that.

Now writing to flash but it does not seem to load correctly. Will try some more bit files and verify them with the official tool just to make sure.

trabucayre, 2021-04-20:
arty, alchitry and basys3 are based on the same device (but with different packages) this why it's possible to use arty board to load in RAM (Only the IDCODE is required and all have the same).
To write in flash the bridge must be available. I will add one for basys.
For alchitry if you can't write flash with openFPGALoader, but if you can with the official tool I will order one to check and fixing.

trabucayre, 2021-04-20:
I have added everything required to use the basys3 to load in SRAM and writing in the QSPI Flash.

Could you try with something like
```
openFPGALoader -b basys3 xxxxx
```
Thanks

paller, 2021-04-21:
> ```
> openFPGALoader -b basys3 xxxxx
> ```

That part works fine.
```
% ./openFPGALoader -b basys3 ~/basys3_test/basys3_test.runs/impl_1/loadbar.bit
write to ram
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Open file DONE
Parse file DONE
load program
Flash SRAM: [==================================================] 100.000000%
Done
```
Tried writing to flash
```
% ./openFPGALoader -b basys3 -f ~/basys3_test/basys3_test.runs/impl_1/loadbar.bit
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a35tcpg236.bit
load program
Flash SRAM: [==================================================] 100.000000%
Done
Detail:
Jedec ID          : c2
memory type       : 20
memory capacity   : 16
EDID + CFD length : c2
EDID              : 1620
CFD               :
Error: rx 42 21 84
42
wait: Error
write en: Error
Erasing: [==================================================] 100.000000%
Fail
```

trabucayre, 2021-04-21:
The good point is SRAM is working, bridge too -> your able to read the correct JEDEC ID.
The bad point is the flash is configured in quad mode, the spiFlash must be improved to take this configuration into account, for example by disabling this before any operations and reenable this after.
I will simulate this with a Flash in PMOD format to update the class.
Thanks for your feedback.

paller, 2021-04-21:
I was trying to understand QSPI flash better. Sounds easy enough to write some data, but after reading the [ZipCPU blog post](https://zipcpu.com/blog/2019/03/27/qflexpress.html) about making a controller I really lost faith.

If it helps reading other implementations I believe the most recent Alchitry Au bridge is this one https://github.com/alchitry/au-bridge/tree/master/jtag_prog.srcs/sources_1/new

I can't remember where I might have read it, but if I recall correctly then there are two versions of the basys3 board with different flash chips.

trabucayre, 2021-04-21:
For this current purpose it's less complex than that. The bridge is more or less just used to adapt signals between jtag and SPI. So everything is done in PC side.
According to jedec ID your board seems a rev.D with an MX25L3233F chip.

My assumption is I have to check status register to determine is the flash is in quad mode and change configuration (maybe it's more easy than that, I need to try).

trabucayre, 2021-04-22:
After some check, I've suppress a residual test definitively useless and error prone.
Could you update the repository and try again (with both boards).

Thanks

paller, 2021-04-22:
**Basys3**
```
% ./openFPGALoader -f -b basys3 ~/basys3_test/basys3_test.runs/impl_1/loadbar.bin
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a35tcpg236.bit
load program
Flash SRAM: [==================================================] 100.000000%
Done
Detail:
Jedec ID          : c2
memory type       : 20
memory capacity   : 16
EDID + CFD length : c2
EDID              : 1620
CFD               :
Erasing: [==================================================] 100.000000%
Done
Writing: [==================================================] 100.000000%
Done
```
Freaking works

**Alchitry Au**
```
% ./openFPGALoader -b alchitry_au -f ~/project_1/project_1.runs/impl_1/top.bin
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a35tftg256.bit
load program
Flash SRAM: [==================================================] 100.000000%
Done
Detail:
Jedec ID          : bf
memory type       : 26
memory capacity   : 42
EDID + CFD length : bf
EDID              : 4226
CFD               :
Erasing: [==================================================] 100.000000%
Done
Writing: [==================================================] 100.000000%
Done
```
It goes to a 100% but nothing happens. It should display a number on the 4 7-segment displays which is working if writing the .bit file to SRAM.
First writing the .bit file into SRAM and it displays a number. Then writing the .bin file to flash and it goes dark as if there's no valid data to load from flash. Tried to power-cycle it just to make sure it's not related to reset, but it still doesn't run anything.

paller, 2021-04-22:
I don't know what I expected to find but I captured the USB data for both openFPGAloader and alchitry-loader. I assume that, what ever it is that WireShark can't understand and marks as "left over data" is mostly the .bin file.
Alchitry-loader seems to encapsulate more data in a packet and they are very consistent. openFPGAloader has some of the same data which I take as a good sign, but it's as if it enters a loop of a sort where it doesn't transmit the data from the .bin file any more.

Don't know if that makes any sense.

trabucayre, 2021-04-22:
For the basys3 it's a good news.
For alchitry, I will reread schematic (and qspi flash datasheet) but definitely will order one to fix support.
wireshark is a really good tool to see behavior and sometime provides really useful informations. I'm interested to see these files.

paller, 2021-04-23:
You can find a zip with both captures at https://send.bitwarden.com/#s_i41vTPa06jsK0TAPsUTA/mCzdESti5el00bHF8drvAQ it should be freely available.

trabucayre, 2021-04-28:
Sorry for delay to answer and thanks for the dumps.
Unfortunately these give no informations since alchitry-loader has an bitstream with everything about the communication with the flash (unlike openFPGALoader).
Anyway I have received my board today. The issue is due to a default flash lock.
After adding a global unlock command I'm able to write sucessfully to the QSPI flash with a new bitstream.
I need now to see if it's a common behavior of storage devices or specific to the one embeded in this board and add the correct piece of code (or at least a workaround for this specific case).

Thanks

paller, 2021-04-30:
> I need now to see if it's a common behavior of storage devices or specific to the one embeded in this board and add the correct piece of code (or at least a workaround for this specific case).

If it's a feature of the QSPI I would assume there's no harm in unlocking an already unlocked flash and it would be possible to start out with a unlock attempt for most flashes/boards?

Appreciate your work, thank you.

trabucayre, 2021-05-01:
It's usual to have write protection by sector or area. But with this flash, this protection is more global. I'm re-reading fully the datasheet to check this behavior.
I'm re-reading other datasheet for others flashs to see if the global unlock register is common or not. I don't want to send a potential risky command for all devices...
After that, according to my conclusions I will added a work around based on jedec ID (and by the way starting to have a better detection display).

trabucayre, 2021-05-05:
I've pushed a workaround to unlock flash sectors for microchip SST26VF032B / SST26VF032BA.

Now you will be able to write bitstream in flash.

A less funny thing: To be able to use quad SPI read (fastest) the configuration register (non volatile modification) must be changed. So it's seems required to use x1 or x2 bitstreams configuration.

paller, 2021-05-07:
> A less funny thing: To be able to use quad SPI read (fastest) the configuration register (non volatile modification) must be changed. So it's seems required to use x1 or x2 bitstreams configuration.

Not sure I understand. The B and BA version have different defaults at power-on for the configuration register but they can be changed.

trabucayre, 2021-05-08:
In fact it's not a problem for openFPGALoader: it use only x1. It's a problem for the FPGA when it load the bitstream.

Yes B has IOC='0' by default and '1' for BA. It's possible to change this by writing in the configuration register but it's state is lost after a power cycle, so, I assume, for the FPGA there is no way to use x4 (at least for B model).

trabucayre, 2021-05-21:
Is it okay for you?
Can I close this issue ?

Thanks

paller, 2021-05-21:
> Is it okay for you?
> Can I close this issue ?

Absolutely, thank you.
So awesome to have a proper cli tool now, great work.

trabucayre, 2021-05-25:
Thanks for your feedback

FlorianKoegler, 2022-02-26:
Hi, first off: I am unsure if it would be better to open a new issue or continue here. 

I'm trying to use openFPGALoader to program an Alchitry Au board and cannot get it to work with the external flash. Internal SRAM works fine.

I have tried both with the official [Au-Base-Project](https://github.com/alchitry/Au-Base-Project) and my own project. 

### Results with openFPGALoader
Write to SRAM is okay. After completing the programming, the DONE LED turns on.
```
pi@raspberrypi:~ $ openFPGALoader -b alchitry_au test.bit
write to ram
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Open file DONE
Parse file DONE
load program
Flash SRAM: [==================================================] 100.00%
Done
```

But when writing to flash, the DONE LED never turns on and the FPGA is not functional (my own project implements a I2C slave; which does not respond; the Au-Base-Project should enable some LEDs which don't light up). 

Also, when including the `--verify` parameter, it fails:
```
pi@raspberrypi:~ $ openFPGALoader -b alchitry_au test.bit -f --verify
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a35tftg256.bit.gz
load program
Flash SRAM: [==================================================] 100.00%
Done
Detail:
Jedec ID          : bf
memory type       : 26
memory capacity   : 42
EDID + CFD length : bf
EDID              : 4226
CFD               :
Non Volatile
flash chip unknown: use basic protection detection
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Fail
Verification failed at 8204
```

### Results with Alchitry Loader
Programming to flash works with the Alchitry Loader on Windows. I am however unsure how the JTAG->SPI is implemented there, as it appears to be packaged in a single .exe with some Java frontend.

For my design, I had to use the following constraints, to get it to work with flash (derived from the Au-Base-Project)
```
set_property BITSTREAM.GENERAL.COMPRESS TRUE [current_design]
set_property CONFIG_VOLTAGE 3.3 [current_design]
set_property CFGBVS VCCO [current_design]
set_property BITSTREAM.CONFIG.SPI_BUSWIDTH 1 [current_design]
set_property BITSTREAM.CONFIG.SPI_FALL_EDGE YES [current_design]
set_property BITSTREAM.CONFIG.CONFIGRATE 33 [current_design]
set_property BITSTREAM.CONFIG.SPI_32BIT_ADDR NO [current_design]
```

Comparing it with the constraints in `constr_xc7a_ftg256.xdc`, the SPI_BUSWIDTH is different, but this doesn't seem to matter (I guess as the SPI->JTAG bridge never sources the bitstream from flash). Just to be sure, I have tried generating a spiOverJtag bitstream with different SPI_BUSWIDTH and using this; but to no avail. 
The [au-bridge](https://github.com/alchitry/au-bridge/blob/master/jtag_prog.srcs/constrs_1/new/pins.xdc) actually doesn't state any Bitstream-related constraints at all; though I am not sure if this is the one used by Alchitry Loader

### Possible Version Differences?

I have tried with version v0.7.0 as well as v0.3.0, which i believe should be the version where the board was initially implemented and working, after this issue.

I am using the [Sparkfun version of the board](https://www.sparkfun.com/products/16527) with added QWIIC I2C connector. But comparing the schematic and the spiOverJtag source code, I can't see any problems. The flash memory seems to be the exact same model.

So I am at a loss here and would appreciate your help. Are there any steps, that I'm missing? 😕

trabucayre, 2022-02-26:
It's true. Thanks to point this issue!
I have tested and I'm unable to update the flash.
I have to check why.
v0.3.0 is the first release with this board but seems not working too. So maybe a regression has been introduce just before or there is another (external?) problem.
I use most of the time the SRAM (and it's not the board I use the most) so I have not seen that before...

trabucayre, 2022-02-26:
Okay... I have used another bitstream and flash works in v0.6.1
spiFlash class has been rewritten after this release, I see possible reason why it's not working (same as #154)

trabucayre, 2022-02-26:
One working example is [blinky](https://github.com/fusesoc/blinky)
fusesoc will starting vivado programmer to load bitstream into SRAM, you can stop it or not
After you can flash bitstream with openFPGALoader

trabucayre, 2022-02-26:
In fact I've just discovered an issue with a compressed bitstream: in this situation I have a fail too...

trabucayre, 2022-02-27:
I have to improve some tests to validate this chip (and maybe to create a specific workaround for erase operation).
I'm not sure it's directly related to the size of the bitstream. I think for an area with the content always == 0 and left to this value instead of set to 1 during erase step.
Same I have to check for blinky example.
Thanks

FlorianKoegler, 2022-02-27:
I found the reason, why blinky did not blink: I used the source file and constraint directly in Vivado, and the clk_freq_hz  parameter was set to 0. Normally fusesoc overwrites it to 100000000.
With correct parameter, it works. 🤦 

I tried switching between my own project's and blinky's bitstream several times; so far the problems appear to be fixed.

trabucayre, 2022-02-27:
Great news! :+1: 
Thanks!
Feel free to reopen this issue (or a new one) if needed.

closed 2022-02-27 by trabucayre