trabucayre/openFPGALoader #84: dirtyJtag -> failed to claim cable

JDFischbach, 2021-05-03:
I have compiled and installed openFPGALoader (running into linker errors when compilig with `-DBUILD_STATIC=ON` or `-DENABLE_UDEV=OFF`, but just bypassing by leaving those flags out). Set up the udev rules, rebooted (multiple times) and tried 
```
openFPGALoader -cdirtyJtag --detect
```
which results in

```
write to ram
getVersion: usb bulk write failed -7
Jtag frequency : requested 6000000Hz -> real 6000000Hz
setClkFreq: usb bulk write failed -7
Fail to set frequency
Error: Failed to claim cable
```

when unplugging dirtyjtag the following result occurs:
```
write to ram
fails to open device
Error: Failed to claim cable
```
after plug cycling:
```
write to ram
Jtag frequency : requested 6000000Hz -> real 6000000Hz
writeTDI: read: usb bulk read failed -7
writeTDI: fill: usb bulk write failed -7actual length: 0
writeTDI: fill: usb bulk write failed -7actual length: 0
writeTDI: fill: usb bulk write failed -7actual length: 0
writeTDI: fill: usb bulk write failed -7actual length: 0
sendBitBang: usb bulk write failed 1
sendBitBang: usb bulk write failed 1
sendBitBang: usb bulk write failed 1
sendBitBang: usb bulk write failed 1
sendBitBang: usb bulk write failed 1
sendBitBang: usb bulk write failed 1
Error: currently only one device is supported
```
the cable is detected correctly in urjtag and i can load FPGAs no problem...

I am running on Ubuntu 20.04 LTS

these are the installed packages:
```
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libftdi1'
--   Found libftdi1, version 1.4
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.23
-- Checking for module 'libudev'
--   Found libudev, version 245
```

What am I doing wrong??
Regards

JDFischbach, 2021-05-03:
Right after I ran my tests i went back to Urjtag trying to write a bitstream -> which resulted in it not being able to read from usb. After plugcycling once more it worked flawless again (if a bit slow, but thats normal). This indicates to me that openFPGALoader leaves the device in a broken state??
Maybe this information helps piecing it all together
Regards JD

trabucayre, 2021-05-03:
Your dirtyJTAG is up to date ?
Since 9ee8e84 the clock frequency has been increased but with an  old dirtyJTAG's firmware you must use --freq 600000 to slow down clk frequency.

Maybe this is why your dirtyJTAG is in incoherent state

JDFischbach, 2021-05-04:
Well... That worked thank you...

closed 2021-05-04 by JDFischbach

trabucayre, 2021-05-04:
Thanks for that.

I wonder if it's required, or not, to slow down this cable to avoid this type of failure.
Okay, this information is in the README, but it's quite easy to miss it.

JDFischbach, 2021-05-04:
Yeah I just thought that my cable was fairly recent. But unfortunately I must have created it just before the new version...

JDFischbach, 2021-05-04:
Btw: writing to flash on a Colorlight (ver 7.0 -> LFE5U-25F-6CABGA256) seems to not work 100% reliable:
```
Erasing: [==================================================] 100.000000%
Done
Writing: [==============================================    ] 91.173119%timeout: 4 20 1000
4
wait: Error
write en: Error
Refresh: FAIL
displayReadReg
	Config Target Selection : 0
	SPIm Fail1
	Preamble ERR
	EXEC Error
```
any idea what happens there?

trabucayre, 2021-05-04:
Could you check cables between board and probe?
Often a connection problem causes this kind of error