trabucayre/openFPGALoader #85: Unable to write to flash on CrossLink-NX evaluation board

shreekumar3d, 2021-05-07:
I am unable to write to flash on my Lattice CrossLink-NX evaluation board. I have used this earlier on the Lattice ECP5 board -  it works for me there.

Here is the what I see -- any idea what might be going wrong ?

    $ ./build/openFPGALoader  -b crosslinknx_evn -f  /media/shkumar/Windows/Users/shkumar/my_designs/blinky/impl_1/blinky_impl_1.bit
    write to flash
    Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
    Enable configuration: DONE
    SRAM erase: DONE
    Open file DONE
    Parse file DONE
    Detail: 
    Jedec ID          : 00
    memory type       : 00
    memory capacity   : 00
    Detail: 
    Jedec ID          : 00
    memory type       : 00
    memory capacity   : 00
    timeout: 0 0 1000
    0
    wait: Error
    write en: Error
    Erasing: [==================================================] 100.000000%
    Fail
    Refresh: DONE

shreekumar3d, 2021-05-07:
I investigated this issue a bit. Adding a 1 second sleep before DisableISC() allows the flashing procedure to finish completely (see below)

    diff --git a/src/lattice.cpp b/src/lattice.cpp
    index 6dd6428..48e4fa6 100644
    --- a/src/lattice.cpp
    +++ b/src/lattice.cpp
    @@ -492,6 +492,7 @@ bool Lattice::program_flash(unsigned int offset)
                    printSuccess("DONE");
            }
     
    +       sleep(1);
            DisableISC();
     
            if (_file_extension == "jed")

Regarding how/why this works - I need to explain how I am managing the flashing process.  The CrossLink-NX evaluation board has a button SW5 - connected to PROGRAMN.  To flash the board, I first push this button.  Then run openFPGALoader. Then I release the button.

According to [CrossLink-NX sysCONFIG Usage Guide (FPGA-TN-02099)](http://www.latticesemi.com/view_document?document_id=52790) :

> For slave port configuration or programming, if the PROGRAMN pin is persisted in user function mode, 
> it has to be driven high before the ISC_DISABLE command is issued.

I take this to mean, "do not issue the ISC_DISABLE command when the PROGRAMN pin is high".  My 1 second delay tries to ensure this. And it works well for now, for me at-least...

Holding down the PROGRAMN pin is required for this board for writing to SRAM as well (default -m option).  If anybody else is using openFPGALoader with CrossLink-NX, it would help if you can chime in with your experiences as well.

trabucayre, 2021-05-08:
It's really interesting because I have this board (with the `ES` version) and never used PROGRAMN button (loading a bitstream in flash or write in Flash works well without manual intervention).
I fact according to the FGPA-TN-02099 p15:
```
If the device is currently being programmed via JTAG, PROGRAMN is ignored until the JTAG mode programming
sequence is complete.
```
I must reproduce your setup to check this status
Thanks

shreekumar3d, 2021-05-08:
Actually I stumbled upon the sleep workaround quite by accident.  Then I realized there is a potential explanation for it (given above). 

I have a board with the ES2 version of the chip : 8BG400CES2.

Lattice Radiant's "programmer" tool doesn't work either without SW5/PROGRAMN.

If you look at the schematic of the CrossLink-NX evaluation board, you will see that there is a DNM resistor (R173) that can potentially be used to connect PROGRAMN to the JTAG interface of the FT2232H.   This connection option does not exist in the ECP5 EVN.... Not saying this is relevant, but something I noticed in my study.

trabucayre, 2021-05-08:
It's weird. Okay your device is an ES2 but theorically the JTAG take control to the FPGA without anything (it's at least true for all FPGA including my ES).
Have you tried to use openFPGALoader without using PROGRAMN ?
I regret that there is no errata...

shreekumar3d, 2021-05-08:
I started off using openFPGALoader without PROGRAMN. It doesn't work at all for flash memory.  For RAM, the behaviour is inconsistent. Sometimes it works first time after DC power is applied to the board.  In the subsequent invocations, the tool says "pass", but the bitstream doesn't change.  That's weird too...

The only consistent behaviour I've been able to get until now is using PROGRAMN.

Note that this is not just with openFPGALoader.  The official tool from Lattice behaves exactly the same way too for RAM and flash. I don't see any difference really.

shreekumar3d, 2021-05-10:
Thanks for looking at it.

No need to buy a new board for now I think.  I tested two boards - both showed the same behaviour. I will contact the official Lattice support & see what solution they suggest for their programmer.   

We got six more boards for a project - but access to them is off limits at this time due to COVID related restrictions...

For now, this is a workaround that I need to ensure my development & workflow stays on Linux.  Once a solution is properly identified, then it can be implemented in a better way I think.

trabucayre, 2021-05-10:
True. For sure when stable device will be available I will bought a new one, ESx seems really unstable (again I regret no errata available).

It's a good idea to ask vendor for that, maybe answer will help to find the correct way to fix this.

Maybe the solution will be available when official devices will be publicly availables (with less issues)...

trabucayre, 2021-10-20:
it's not an ES but an ES2. This why PROGRAMN must be manually driven... This workaround is not required with my ES device...
ES2 is maybe included into LIFCL-40-8BG400? I assume @gatecat may provides more information to this topic.
It's required for me to have access to ES2 or stable devices.

siluks-plumerai, 2023-02-10:
Ive just tested sram and flash writing with [Crosslink-NX VVML board ](https://www.latticesemi.com/products/developmentboardsandkits/crosslink-nxvoiceandvisionmachinelearning) It works great!

```
slawek@solo:~/git/github/openFPGALoader/build$ openFPGALoader -b crosslinknx_evn ~/git/plumerai/lai/hw/syn/CROSSLINK_NX_VVML/impl_1/CROSSLINK_NX_VVML_impl_1.bit
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file: DONE
Parse file: DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.00%
Done
Disable configuration: DONE
slawek@solo:~/git/github/openFPGALoader/build$ openFPGALoader -b crosslinknx_evn -f ~/git/plumerai/lai/hw/syn/CROSSLINK_NX_VVML/impl_1/CROSSLINK_NX_VVML_impl_1.bit
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Detail:
Jedec ID          : c2
memory type       : 20
memory capacity   : 18
EDID + CFD length : c2
EDID              : 1820
CFD               :
flash chip unknown: use basic protection detection
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Refresh: DONE
```

In order to avoid problems with PROGRAMN follwing constrains was added to .pdc:
```
ldc_set_sysconfig {JTAG_PORT=ENABLE CONFIGIO_VOLTAGE_BANK0=1.8 MASTER_SPI_PORT=SERIAL BOOTMODE=SINGLE CONFIG_IOSLEW=FAST}
ldc_set_vcc -bank 0 1.8
```