trabucayre/openFPGALoader #86: DE10 Nano support

gatecat, 2021-05-11:
The DE10 nano has a built in USB Blaster, VID:PID `09fb:6810`.

Running `openfpgaloader -c usb-blaster lut.rbf`

fails with:

```
write to ram
unable to open ftdi device: -3 (device not found)
Error: Failed to claim cable
```

Is this board supposed to work (the various USB blaster versions are a bit unclear?)

trabucayre, 2021-05-12:
This is an usb-blaster II based on cypress FX2 instead of FTDI.
Thanks to point this because this version is not currently supported (I only have usb blaster I).
Need to get one to add support.

Thanks

trabucayre, 2021-05-13:
I've pushed support for usb-blasterII
```
openFPGALoader -c usb-blasterII --probe-firmware /opt/altera/quartus/linux64/blaster_6810.hex
```
Not fully tested since I use an de0nanoSoc with cycloneVSoc.
Now, next step is to allows a chain with more than one device (as mentioned in #81).

Thanks

trabucayre, 2021-05-15:
Hi,
I have pushed in branch `JTAG_chain` all required to load an `rbf` or `svf` in a cycloneV Soc.
Tested with my de0nanoSoc board.
Need to check all part before merging this feature in `master` branch.

Thanks

trabucayre, 2021-05-21:
Is it okay for you?
Can I close this issue ?

Thanks

gatecat, 2021-05-21:
Sorry, I haven't had a chance to test, let me try and remember over the weekend! I'll close the issue and reopen if I hit problems.

trabucayre, 2021-05-21:
Don't worry.
I have added idcode for your model of FPGA so theoretically your board will be immediately ready to use.

gatecat, 2021-05-23:
I'm trying this out with the `JTAG_chain` branch. It's currently appearing to program, but the LEDs never actually do anything.

Command I'm using is `openFPGALoader  -c usb-blasterII --probe-firmware $HOME/intelFPGA_lite/20.1/quartus/linux64/blaster_6810.hex --index-chain 1 test1.rbf`

trabucayre, 2021-05-23:
It's partly a good news. This mean usb blaster is working and your FPGA is recognize.
`--index-chain` is not required for socFPGA since the ARM processor is seen but not considered as an supported device.
could you try with `-v` to have more message?

I  will order one of this board to reproduce this.
Thanks

gatecat, 2021-05-23:
running with `-v`, and removing `--index-chain`, the output is:

```
write to ram
USB-Blaster II firmware version: 1.39
found 2 devices
index 0:
	idcode   0x4ba00477
	type     ARM cortex A9
	irlength 4
index 1:
	idcode 0x2d020dd
	manufacturer altera
	model  5CSEBA6
	family cyclone V Soc
	irlength 10
File type : rbf
Flash SRAM: [==================================================] 100.000000%
Done
```

trabucayre, 2021-05-23:
Display seems good. Could you try with SVF? Maybe a small subtelity (delay, pause or something like this).
I have ordered one de10nano. Will move to 20.1 (currently I have 19.1).

trabucayre, 2021-05-24:
Could you share your design, to be able to reproduce?

Thanks

gatecat, 2021-05-24:
RBF (and SOF): [test1.zip](https://github.com/trabucayre/openFPGALoader/files/6533344/test1.zip)

Will try the SVF route in a bit

trabucayre, 2021-05-24:
Thanks.
Is it possible to chair the project too :-)

gatecat, 2021-05-24:
[test1.zip](https://github.com/trabucayre/openFPGALoader/files/6533407/test1.zip)

trabucayre, 2021-05-24:
Thanks.
For test with `rbf` could you try with `ON_CHIP_BITSTREAM_DECOMPRESSION=OFF`

gatecat, 2021-05-24:
The uncompressed `rbf` works fine. `svf` works with or without compression.

trabucayre, 2021-05-24:
Good news. Problem is about compression.
I've not found how to specify compression for svf with `quartus_cpf` with and without `--option=bitstream_compression=on` produces the same file.

gatecat, 2021-05-24:
Ah, I didn't actually check whether or not the SVF was compressed. In that case, maybe compression just doesn't work with JTAG?

trabucayre, 2021-05-25:
Good question.
I will compare both rbf version, search a bit in datasheet and in linux driver to see if a specific step is required to load a compressed bitstream.

trabucayre, 2021-05-25:
Answer is in  CV-52007 p 7-37 (_Configuration Data Compression_)
```
Decompression is supported in all configuration schemes except the JTAG configuration scheme.
```
So I must added a note to explain this in the README.md.

trabucayre, 2021-05-27:
Hi,
I have received my *de10nano* yesterday and tested:

- with `svf` -> OK
- with compressed `rbf` -> KO
- with uncompressed `rbf` -> OK

This behaviour is, according to the handbook, the one expected and reproduce your experiments.

I have updated the README to add a note about real-time decompression not supported in JTG mode.

It's ok for you

Thanks

gatecat, 2021-05-27:
Thanks for looking into this!

closed 2021-05-27 by gatecat

kassane, 2022-06-29:
Would it also apply on the DE10-lite?

trabucayre, 2022-06-29:
de10nano is based on cycloneV SoC, de10-lite is based on max10.
Currently max10 could be programmed only using SVF file. I have to restart my investigations to add a more complete and efficient for this device. Currently I'm able to program my trenz board but [deca](#215) seems not working (and I can't bought one).
Could you try with your board ?
Anyway rbf isn't compatible with max10

kassane, 2022-06-30:
> de10nano is based on cycloneV SoC, de10-lite is based on max10.

Good to know! 👀

> Currently max10 could be programmed only using SVF file. I have to restart my investigations to add a more complete and efficient for this device. Currently I'm able to program my trenz board but [...]

I will soon receive this board during a complementary fpga course and plan to run some tests after the course. But depending on the result will open an issue explaining the situation.

trabucayre, 2023-01-14:
Hi,
the de10lite board is now officially supported. But it's still required to use svf format.