trabucayre/openFPGALoader #87: Under Windows + MinGW, with fs format (gowin), CRLF make file size check fail

sylefeb, 2021-05-26:
Under Windows + MinGW, gowin_pack outputs a text fs file with CRLF end of lines.

It turns out ifstream::gcount() does not count two characters but only one for these, making the following file size check fail:
https://github.com/trabucayre/openFPGALoader/blob/a242168e7fb4376c1b93e440a27ea1c33d033708/src/configBitstreamParser.cpp#L32

A simple external fix is to apply, e.g. dos2unix to convert the FS file end of lines.

Fixing this internally may require either reading in binary mode and making the parser compatible, or reading the file in a different way?

trabucayre, 2021-05-26:
Thanks and good catch!
`dos2unix` may be a work around but it's add an extra step and, consecuently I'm not happy as long term solution.
I'm hesitate between two options:

- using `ftellg()` to read current position, and eof / bad to check if the read is out of content
- moving to the most traditional (and safe) `fread()` and compare return value

In a first approach I think the second option (really near current implementation) is the most robust and don't have impact to the read duration.

sylefeb, 2021-05-26:
Agreed, sounds like a good fix!

trabucayre, 2021-05-26:
With this problem about CRLF vs LF and some test using unix2dos I seen an issue with fsparser, so now if a line (after getline()) end with CR I pop this.

trabucayre, 2021-05-26:
Done!
Could  you test by providing a file or by using `cat file.fs | openFPGALoader xxx` to check if it's working with a windows system?
Thanks

trabucayre, 2021-05-26:
I haven't precise: the correction is in `jtag_chain`  branch

sylefeb, 2021-05-26:
Works like a charm, fantastic! -- attaching the file just in case, and closing. Thanks for the quick fix.
[pack.zip](https://github.com/trabucayre/openFPGALoader/files/6548517/pack.zip)

closed 2021-05-26 by sylefeb

trabucayre, 2021-05-26:
Thanks for the feedback