trabucayre/openFPGALoader #88: Detect works but

kargeor, 2021-06-06:


trabucayre, 2021-06-06:
Which board you use ?
With these interface I assume you use "dupont" cable? Could you check wires, It's usually a source of problem/glitches.
Sometime pullup and pulldown are required to have stable signals
Thanks

kargeor, 2021-06-07:
I am using a custom built board. Yes, "dupont" cables. 😃 
Turns out adding a resistor to TDO fixed all problems. Maybe add a note in the docs?
Also added the XO2 part I'm using [here](https://github.com/trabucayre/openFPGALoader/pull/89/files)

closed 2021-06-07 by kargeor

trabucayre, 2021-06-08:
It's true I must add warning about "dupont" cables and pullup/pulldown resistors.
Maybe adding a FAQ somewhere with common issues will be a good feature too.

Thanks for you feedback

DoctorMikeReddy, 2021-08-09:
What value resistor was used on the TDO line?

kargeor, 2021-08-10:
1k or 10k.

stefan63, 2024-01-08:
TDO pullup or down?
Kind regards/Stefan