trabucayre/openFPGALoader #93: Reading flash wanted

emard, 2021-06-22:
HI

This is flash read feature request :)

Sometmes I'd like to read content of the flash,
a bytes length from some starting address.
Flash can contain other data in use not only bitstream.

So I thought the openFPGALoader utility could
write and read, maybe even verify.

I'm highly interested in ECP5 on ULX3S, I know
ft231x usb-serial read is much slower than write,
that's expected for this chip.

Have also ft2232 and other platforms to try, they
should be very fast at reading.

trabucayre, 2021-06-23:
Hi,
This is in my TODO list but now I must change priority :)
The SPI lowlevel drive has already basic method to read from Flash.
So I must
- add an optional verify arg to check after write (with same base address and bitstream size)
- add an optional read to dump flash to a file (so maybe a size args is required and knowledge about flash size too).

Thanks

emard, 2021-06-23:
Yes something like that!

When dumping flash to binary file, please add parameters
for starting address and size wanted to dump.

To inspect flash config, sometimes I need to dump 64K (one sector)
from 0x380000. Also I'd like to write modified sector 64K from 0x380000
to flash. (This things are used to set up ulx3s booting linux)

It can already be done on ULX3S using ftp esp32 wifi, I was just
thinking a USB tool would be nice to be able to do the same :)

trabucayre, 2021-06-23:
`--offset` args is already available so it's possible to write everywhere in flash.
To verify this information must be used too (It didn't make sense to read at 0 if we write at 0xnnnn :) ), size is binary file size.
I must add `--size` to dump not all the flash.

emard, 2021-06-23:
yes, this will cover all cases needed

trabucayre, 2021-06-24:
I've pushed a first draft to match your requirements.
- dump raw data (`--dump-flash`) with `-o` for base address and `--file_size` to specify amount of Bytes (currently lattice only)
- verify (`--verify`) in this mode, after write, it read starting at 0 if `-o` is not used or at offset provided by this option (currently lattice and xilinx only).

trabucayre, 2021-06-25:
I updated xilinx support to add dump too.
Must doing the same for all device with flash (ice40, efinix, anlogic)

emard, 2021-06-26:
Hey that's wonderful, I tried on ULX3S and it works!!!

trabucayre, 2021-06-26:
I've tried first with UL3XS :)
After that I've tried with quite all boards I have: works with arty, alchitry_au, xyloni, icebreaker, ...

emard, 2021-06-26:
1MB save to file from some offset takes about 1min at ULX3S which is very
good for the capability of FT231X.
Of course add the progress bar when possible.
I can try this function on other boards and cables too!

trabucayre, 2021-06-27:
adding a progressBar may a good solution to avoid waiting without information, but currently I read the full size in one operation to reduce flash configuration overhead (1B for instruction + 3B for address).
For 1MB with, for example, a read by 512B each, you need 2048 access -> 8192B for configuration must be added and consecuently this add more delays.
But if you think this is acceptable it's possible to rewrite dumpFlash, and verification step to slice read and add progress bar.

emard, 2021-06-27:
I see, this read-everything approach makes sense,
because flash is rarely read and when it is, it will be
read from 0 most of the time - that's why I was surprised
why is it so fast :). OK if it works, leave it as it is now!

I guess it's fine only to write few lines of comments note in
the code what should be done to speed up reading short
length of data from near the end of flash.

trabucayre, 2021-06-27:
Reading at the beginning or the end is exactly the same thing: you send 0x03 followed by the 3Bytes for the base address and you read file-size Bytes.
It's true a note to explain this operation may take time without display is a good idea to avoid user ask: reading or crashed?

emard, 2021-06-27:
I know for our fujprog, it was erasing flash just without any printed notice and
users were pressing ctrl-c and complaining "it just halts and doesn't works".
We added just printf("erasing flash please wait"); and users consider
it "fixed" :)

trabucayre, 2021-06-29:
It's exactly my idea! A too long wait without message or information  is a risk to let user thing something goes wrong.
I will add message to explain situation, but before, I need to merge all code for verify and dump in spiFlash. Most device has the same piece of code, I hate to have multiple time 100% same code :)

trabucayre, 2021-07-11:
I have added a note about potential long delay to read flash. dump & verify are now implemented at spiFlash level to reduce dupplicate code.
It's okay to you ?

emard, 2021-07-11:
Yes absolutely!! Awesome work
I will test a bit, if I find something

eventual bugs are expected at boundary
conditions, like write to address
that is not aligned to flash block

Proper, clean, write should first
read block and then modify it

In my esp32ecp5 I need simple code
and just report error if write address is
not block aligned

trabucayre, 2021-07-12:
i'm interested by all possible failures.
I think the idea to move quite everything in a library is interesting to create tests case with this type of experiments.

trabucayre, 2021-07-23:
It's okay for you ?

emard, 2021-07-24:
Yes

trabucayre, 2021-07-27:
Thanks!
feel free to reopen or create a new issue if you see something wrong

closed 2021-07-27 by trabucayre