trabucayre/openFPGALoader #95: Can I expect Lattice HW-USBN-2B programming cable to work?

metscore, 2021-06-24:
Hi,

First of all, thank you for this project ! 👍 

I have been forced to use Diamonds Programer to program my Lattice devices, but since this feels like having a tooth pulled out over and over again I thought I give this a try. 

Since I'm a Windows guy I have downloaded OpenFGPALoader using MSYS2 and the package manager resulting in the following files required to run the application :

```
libftdi1.dll
libgcc_s_seh-1.dll
libstdc++-6.dll
libusb-1.0.dll
libwinpthread-1.dll
openFPGALoader.exe
```

My biggest concern so far is that I am seem to be unable to use the Lattice HW-USBN-2B programming cable device (which has a FTDI 2232 device) that I have. 

I get
```
openFPGALoader.exe mgb2_lcmxo2-256hc_10.jed
write to ram
unable to open ftdi device: -4 (usb_open() failed)
[31mJTAG init failed with: std::exception[0m
```

If I unplug my HW-USBN-2B device I get

```
openFPGALoader.exe mgb2_lcmxo2-256hc_10.jed
write to ram
unable to open ftdi device: -3 (device not found)
[31mJTAG init failed with: std::exception[0m
```
 
Giving me the impression that it at least sees the FTDI chip.

Long story short, can I expect to use the Lattice HW-USBN-2B to program my Lattice devices using openFPGALoader ? 
(My devices are the LCMXO2 family, such as LCMXO2-4000HC)

King regards
Magnus

trabucayre, 2021-06-25:
Hi,
I suspect an access right issue, unfortunately `libftdi` doesn't propagate `libusb`'s error code and this fail may have more than one reason.
I'm only a Linux user so I have no idea about how to check access right in windows, maybe @umarcor has a better knowledge for this configuration.
For lattice probe, you use a custom board or an official one? Are there a schematic?
I have no diamond currently installed in my computer but maybe try with `--ftdi-channel 1` to use FT2232's interface B. I have some eval based on machXO2 and machXO3 and usually jtag interface is a classic FT2232 using interfaceA.
Need to check this probe

umarcor, 2021-06-27:
I have not dealt with it myself, but I've heard that using FTDI devices with libusb on Windows is not straightforward. Tools such as [Zadig](https://zadig.akeo.ie/) are used for specifying which driver is to be used. Otherwise, it assigns the FTDI driver.

metscore, 2021-06-29:
Thanks for your answers! 
It seems as if it is fairly complicated to get this going and since I also use other devices (which as FTDI chips inbuilt), for which I use the FTDI drivers, I'm a bit reluctant to mess with the drivers as such.
I think 'll try to pursue a different path. 

Again, thank you very much for your help !
King regards
Magnus

closed 2021-06-29 by metscore

trabucayre, 2021-06-29:
thanks @umarcor for details.
@metscore I understand. I suppose using libusbk with zadig doesn't break anything, but I'm not sure.
I have created an issue to ask people to add instructions for windows and macOSX. These sections are missing and it's a real problem. I have only computers with Linux OS I'm unable to create these by myself (and, in fact, I have no skill/knowlege about this two OS).
Thanks