trabucayre/openFPGALoader #96: instruction to install and use openFPGALoader with windows and MacOSX

trabucayre, 2021-06-29:
README.md contains requirement, installations instruction and how to use openFPGALoader with GNU/Linux OS. But there is nothing for:
-  windows (manually or using Msys2), zadig, libusbK, ...
- macOSX (homebrew?)

jeanthom, 2021-07-17:
Homebrew for macOS/Linux: https://github.com/Homebrew/homebrew-core/pull/81031

trabucayre, 2021-07-17:
Thanks for this task!
ASAP this PR is applied it's make sense to add a note in the README.

jeanthom, 2021-08-12:
Homebrew & Fedora Copr install instructions added in https://github.com/trabucayre/openFPGALoader/pull/102

trabucayre, 2021-08-14:
Thanks to that!

udif, 2021-12-20:
I just found out that WSL support for USB devices can also be made to work on Windows 10 (You don't need Windows 11).
The trick is to manually install the required 5.10.60.1 kernel.
Please see:
https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/#comment-5654
I just installed it and verified that `lsusb` under WSL shows me my FT2232-based bus-pirate cable, and `openfpgaloader` can detect its onboard XC2C32A CPLD.

trabucayre, 2021-12-21:
Is it possible for you to complete doc page? windows section is empty.
Thanks for informations!

udif, 2021-12-21:
OK, I'll try to prepare a pull request for the install page in the docs for a Windows-WSL section (this is not the same as a native WIndows install).

trabucayre, 2021-12-21:
Thanks. Windows part is a total mystery for me. The last windows I have used is maybe 2k

jrobgh, 2022-12-03:
Here's some notes on building openFPGALoader via Msys on Windows

1) Install msys (e.g. msys2-x86_64-20221028.exe from https://www.msys2.org/ )

2) Run mingw64 (e.g. c:\msys64\mingw64.exe)

3) Get necessary build tools (e.g. x64)
pacman -S mingw64/mingw-w64-x86_64-hidapi mingw64/mingw-w64-x86_64-libusb-win32 mingw64/mingw-w64-x86_64-cmake mingw64/mingw-w64-x86_64-libftdi mingw-w64-x86_64-toolchain

4) In openFPGALoader folder, configure the Makefile

/mingw64/bin/cmake -G "MSYS Makefiles" -D CMAKE_MAKE_PROGRAM:PATH=/mingw64/bin/mingw32-make .

5) In openFPGALoader folder, run the makefile to build the binary (openFPGALoader.exe)

/mingw64/bin/mingw32-make

trabucayre, 2022-12-04:
Thanks for this details! Is it possible to you to open a PR with these informations?
Also as notificed in #245 zadig must be used to replace driver.