trabucayre/openFPGALoader #110: Handling of verbose

UweBonnes, 2021-09-04:
Hello,

handling of verbose seem a little "rough". E.g. Jtag sets it unconditionally to false. So there is no way to control message generation in a probe driver.   I think having a possible fine grain debug system is helpfull.

trabucayre, 2021-09-07:
It's true. In fact jtag verbosity is set to false because this class produce to many "garbage" and it's more a debug mode than a real verbose mode.
I must to cleanup (or simply add a debug mode with a define) to propagate the verbose mode with `--verbose-level`  to have different levels.

trabucayre, 2021-10-03:
I just pushed a --verbose-level: with value= 2, verbose is enabled for lowlevel drivers.

trabucayre, 2021-11-11:
Small update: verbose-level is no more converted to bool for jtag drivers -> this let lowlevel driver to have more choice.

It's good to you?

trabucayre, 2023-01-05:
Can I close this issue? Model is good to you?

UweBonnes, 2023-06-27:
Yes.

UweBonnes, 2023-06-27:
Now used in the BMP  WIP

closed 2023-06-27 by UweBonnes