trabucayre/openFPGALoader #114: Wrong device id - model pair for Lattice ecp5 fpga in part.hpp .

infphyny, 2021-09-21:
I have a ColorLight 5A-75E which have a LFE5U-25 and openFPGALoader detect it with idcode =  0x01111043   and model = LFE5U-12.  Lattice documentation list LFE5U-12 with idcode = 0x21111043  
So device idcode in part.hpp are not the same as [Lattice documentation](https://www.latticesemi.com/view_document?document_id=50462) see Table B.5. p. 58.

https://github.com/trabucayre/openFPGALoader/blob/3698b98976021f80811853fb2205405fb9fbb831/src/part.hpp#L86-L95

Note that openFPGALoader is able to load the bitstream in sram and flash successfully, so maybe just a superficial naming issue.  
Thanks

trabucayre, 2021-09-21:
As mentionned in #107 upper 4 bits are version number. It's in the IEEE 1149.1 specification. I have no idea why 12 and 15 share same idcode (maybe same story as some stm32f1 devices).
And if you take a look in a ECP5 bsm file you can find something like this:

```
--IDCODE and USERCODE register definition
    attribute IDCODE_REGISTER of LAE5UM_45F_XXBG381 : entity is
        "0000" &    --Version number
        "0001000100010010" &    --Device specific number
        "000001000011"; --Company code
```

This why all idcode have upper nibble set to 0 -> avoid failure with device revision. But in the same time it's not funny because it's impossible to differentiate 12 and 25.

Anyway, it's only an issue with display and did not affect capacity to flash/load bitstream.
Maybe the solution is simply to replace this two lines by one with `LFE5U-12/LFE5U-25`

infphyny, 2021-09-21:
Replacing the two lines by one is a good idea since when idcode are the same, only the first match in the table seem to be chosen for the model.
Thank you for the explanation.
Because this issue do not affect openFPGALoader ability to program Lattice ecp5 devices , I will close this issue. 
Thanks.