trabucayre/openFPGALoader #120: ulx3s dump-flash segfault

emard, 2021-10-02:
Dumping 1MB is ok, but something longer like 2MB
segfaults

openFPGALoader -b ulx3s --offset 0x100000 --file-size 0x200000 --dump-flash /tmp/dumpx.bit
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
Enable configuration: DONE
SRAM erase: DONE
dump flash (May take time)
Open dump file DONE
Segmentation fault

openFPGALoader -V
openFPGALoader v0.5.0

trabucayre, 2021-10-02:
Thanks to point this.
I'm able to reproduce. Need to look if it's related to buffer size or pointer issue.

trabucayre, 2021-10-02:
I have added a workaround to limit buffer size. Now no more segfault for 2M dump

emard, 2021-10-02:
Thanx for quick fix, Great! I'll pull and compile

I have also another segfault around that, for example
we are trying to port dfu bootloader for ulx3s,
based on dfu sources from had2019 badge.

But if I specify -b ulx3s --dfu there's also some segfault

trabucayre, 2021-10-03:
Is it : https://github.com/emard/had2019-playground .
Could you provide sequence of operation I can reproduce your issue?

Thanks

emard, 2021-10-03:
Yes that's it!

cd had2019-playground/projects/bootloader
make clean
make multi
fujprog -j flash build-tmp/multiboot.img

I also have binaries here ready for most boards:
https://github.com/emard/ulx3s-bin/tree/master/fpga/dfu

hold BTN1 and plug USB cable to US2,
depending on some conditions (USB, RTC state etc)
board may not power ON immediately from US2.

option 1: additionally plug US1.
option 2: hold BTN1 as "shift" key and press BTN0 to power on

LED D0, D1, D2 should be ON, others OFF.
DFU should enumerate:

[138822.214879] usb 1-9.1: new full-speed USB device number 110 using xhci_hcd
[138822.317921] usb 1-9.1: New USB device found, idVendor=1d50,
idProduct=614b, bcdDevice= 0.05
[138822.317938] usb 1-9.1: New USB device strings: Mfr=2, Product=3,
SerialNumber=1
[138822.317945] usb 1-9.1: Product: ULX3S FPGA (DFU)
[138822.317950] usb 1-9.1: Manufacturer: FER-RADIONA-EMARD
[138822.317954] usb 1-9.1: SerialNumber: 5031473636340015

to program and run bitstream using dfu-util:
(this is what I'd like if the same can be done using openFPGALoader)

dfu-util -a 0 -R -D blink.bit

user bitstream "blink.bit" will be flashed at offset 0x200000.
multiboot feature is used to skip bootloader bitstream that is at flash 0
if BTN1 is not pressed and load user bitstream from flash 0x200000

trabucayre, 2021-10-03:
I have flashed this bitstream this morning and able to see dfu interface.
dfu support has been tested with orangeCrab and fomu. Both boards have only one altsettings.
It's not true for icebreaker-bitsty and ulx3s. I know this limitation and I'm working on fixing that (I espect to push fix today).

emard, 2021-10-03:
basic altmode 0 is very good for now normal use.
Of course, having choice of altmode options is great

currently other altmodes are just for user convenience for some
often used flash starting addresses. It can all be done by
assembling full image and flash at once, DFU is fast.

In the future when riscv firmware is improved and ready,
it could be possible to have more useful altmodes that flash
ESP32 or RTC onboard.

trabucayre, 2021-10-03:
I've pushed my update. But now if board is not known (or if you wish to select non default altsetting) you must use --altsetting

emard, 2021-10-03:
Thanx, you're so fast!! I'll test

trabucayre, 2021-10-03:
It's good ?

emard, 2021-10-03:
I pulled last git,  compiled few minutes ago
maybe I don't know how to use it

./openFPGALoader -b ulx3s --dfu -f blink.bit
write to flash
Open file /tmp/ulx3s_12f_gray_counter.bit DONE
Parse file DONE
Segmentation fault

trabucayre, 2021-10-03:
need to reproduce but you use **blink.bit** and openFPGALoader open **/tmp/ulx3s_12f_gray_counter.bit**
maybe conflict between board (described using jtag and dfu mode)

emard, 2021-10-03:
gray_counter.bit is the blink.bit, it changes LEDs

Iit works from dfu-util but not the openFPGALoader

trabucayre, 2021-10-03:
Could you `git pull` and test with
```
openFPGALoader --dfu --vid 0x1d50 --pid 0x614b --altsetting 0 bink.bit
```

emard, 2021-10-03:
YES, this works and is very fast

trabucayre, 2021-10-03:
Great!
Once your bootloader is stable it more easy to add an entry for ulx3s_dfu

emard, 2021-10-03:
Right, and thanx a lot for for fast activity, we have
DFU mode supported now directly from openFPGALoader
so everything to program the board is one tool

closed 2021-10-03 by emard

trabucayre, 2021-10-04:
Thanks too. Your feedback has allowed me to improve DFU mode.