trabucayre/openFPGALoader #121: ulx3s any image flashing

emard, 2021-10-02:
HI

Is there some option to ignore file content for flashing?
Some files can contain user data, some have different
structure than single bitstream for example multiboot
images are different and currently they won't upload:

openFPGALoader -b ulx3s -f multiboot.img

https://github.com/emard/ulx3s-bin/tree/master/fpga/dfu/85f-v20

trabucayre, 2021-10-03:
Currently no. These informations are provided into file ?

trabucayre, 2021-10-03:
By using `--file-type bin` I'm able to flash img from [dfu bootloader](https://github.com/emard/had2019-playground) (And I see dfu interface).
Maybe I need to be more strict about file format.

emard, 2021-10-03:
thanx, the --file-type bin is the key

emard, 2021-10-03:
this doesn't work for one of my prototype boards I think it's
because of different flash chip type

openFPGALoader -b ulx3s --file-type bin -f multiboot.img
write to flash
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
Enable configuration: DONE
SRAM erase: DONE
Open file DONE
Parse file DONE
Detail:
Jedec ID          : 9d
memory type       : 60
memory capacity   : 18
EDID + CFD length : 9d
EDID              : 1860
CFD               :
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%[A^[[A
Done
Refresh: FAIL
displayReadReg
	Config Target Selection : 0
	Std PreAmble
	ID ERR
	EXEC Error
Error: Failed to program FPGA: std::exception

trabucayre, 2021-10-03:
Are you sure the bitstream is for your ECP5 model? The `ID ERR` is displayed when there is a mismatch.
A check between bitstream idcode and device idcode seems required.

emard, 2021-10-03:
:) true, actually bitstream was for wrong model

trabucayre, 2021-10-03:
Great!
I've added "check idcode" in my todolist

trabucayre, 2021-10-03:
It's good for you?
Can I close this issue?

emard, 2021-10-03:
I think yes, I need to check with various boards but seems ok

closed 2021-10-03 by emard