trabucayre/openFPGALoader #124: flash chip write protection

emard, 2021-10-03:
HI

Is there some support for write protection of flash chips,
usually I need to protect bootloader in first 2MB of flash
from accidental overwriting. My chips of interest are some
ISSI and Winbond 16MB.

trabucayre, 2021-10-04:
Currently, enabling write protection on sectors is not supported.
In fact it's opposite: if a sector is protected openFPGALoader unprotects this one...
But this policy is, maybe, a bad idea.

emard, 2021-10-04:
Yes, I suggest changing that.

I expect it should refuse to write by default,
requrie additional option like "--force" "--unprotect"
or "--protect-override"

I'm trying to protect flash chip with external tools
and check overwriting if protection "works" but
tool already overrides it :)

trabucayre, 2021-10-08:
I answer here for [https://github.com/trabucayre/openFPGALoader/issues/125#issuecomment-937586929].

I'm agree with your point of view. I see how to implement a first draft, no really clean but... I espect to push modifications at end of week-end.
I'm a bit annoyed by breaking current behaviour (inconditional unlock) but it's make sense to add more security and don't do something without reporting to user. So my idea is
- 2 args
  * unprotect (use to unlock sectors before write, lock after if a bitstream is provided, simply unlock when no bitstream)
  * protect to lock a set of sectors (make sense to set tb = 1 to lock low sectors), I'm not sure to have to use start addr since lock always start at 0 (there is another unlock registers but it's not supported by all chips).
- when user ask to write protected sector without unprotect -> fails with a message explaining why and how to unlock
- a first draft for flash recognition based on jedec code: required to check if tb is OTP, and BPx offset (not always same place in status register).

emard, 2021-10-09:
> Personnaly haven't script with autounlock required, and, I prefer too have a
> respect policy by default. Current behaviour is an error. A good message to
> explain why openFPGALoader can't write due to locked sectors make more
> sense.
>
> TB bit seems the most common case, I think, in a first time, limit
> modification to use this one. For a more complex lock use case it's true
> it's maybe overkill for an FPGA purpose where it's not really required. If
> someone ask for this with a good explanation it's possible to modify this in
> future but for now it make sense to use the most common solution.
>
> I have a first proof of concept, need to improve sector lock/unlock
> detection to accept or not write on a (un)locked area.

Yes, I agree that's a good behavior to respect policy.
I also think it is good to have some knowledge of
flash chip protection register and report what range
is protected and what are values of protection registers
with description what is the meaning for their bits and
are they OTP or not.

I know 2 cases from practice. Some of ULX3S
boards have ISSI IS25LP128, some Winbond W25Q128.
For bootloader, first 2MB need to be protected.

To protect first 2MB, ISSI TB must be changed from factory
default 0 to 1, TB cannot be reset back from 1 to 0 but generally
1 is what board needs and it will keep it.

For Winbond, register-2 comes with OTP bits CMP and IRL
both factory default 0. For first 2MB protection this bits and register-2
should not be changed. If it is changed, there's no way back, and
it's good to report to user at least that some changes are already
done and can't be undone because of hard protection.

trabucayre, 2021-10-20:
I've started to push new methods to lock blocks, updated the struct to specify tb location (and added methods to erase 4/32/64 kb).
Next push will be to unprotect block if required and allowed, and to relock after write. But I think, in a first time, I will push in a dedicated branch to check everything without breaking something.

emard, 2021-10-20:
Great, and I'd be happy to test!

trabucayre, 2021-11-06:
I've received my clip (but not currently tested). For de10nano, unfortunately the flash chip has a different package.
Globally, after adding protect/unprotect into some FPGA's driver it appear a most common and generic method must be added to avoid a too big code dupplication.

emard, 2021-11-06:
OK, I will re-ask friend to find IS25LP032 chip which
I think has bug with protection, top/bottom bit is ignored, always
protecting top

emard, 2021-11-08:
HI

Friend found the ISSI 032 chip, normally we don't know will it have same bug
I have seen. We need your shipping address. My mail is ***@***.***
so you can send it to me and I'll coordinate with the friend to send the chip

trabucayre, 2021-11-19:
I've updated openFPGALoader (branch `review_spiflash`) with a first pass of refactoring. Next will be to refactor write, verify and dump, to have, more or less, only one piece of code for all devices.

emard, 2021-11-19:
HI, I compiled "rework_spiflash" branch.
There's small bug in JTAG IDCODE, maybe
only missing a digit.
0x21111043 is 12F
0x41111043 is 25F

12F:
***@***.***:/tmp/openFPGALoader/build$ fujprog -i
ULX2S / ULX3S JTAG programmer v4.8 (git cc3ea93 built May 23 2021 17:03:05)
Copyright (C) Marko Zec, EMARD, gojimmypi, kost and contributors
Using USB cable: ULX3S FPGA 12K v3.0.8
FPGA IDCODE: 21111043
FPGA identified SIZE: 12
***@***.***:/tmp/openFPGALoader/build$ ./openFPGALoader -b ulx3s --detect
write to ram
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
index 0:
	idcode 0x1111043
	manufacturer lattice
	family ECP5
	model  LFE5U-12/25
	irlength 8

25F:
***@***.***:/tmp/openFPGALoader/build$ fujprog -i
ULX2S / ULX3S JTAG programmer v4.8 (git cc3ea93 built May 23 2021 17:03:05)
Copyright (C) Marko Zec, EMARD, gojimmypi, kost and contributors
Using USB cable: ULX3S FPGA 25K v2.1.2
FPGA IDCODE: 41111043
FPGA identified SIZE: 25
***@***.***:/tmp/openFPGALoader/build$ ./openFPGALoader -b ulx3s --detect
write to ram
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
index 0:
	idcode 0x1111043
	manufacturer lattice
	family ECP5
	model  LFE5U-12/25
	irlength 8

trabucayre, 2021-11-19:
It's not really a bug...
4 MSb are version in IEEE 1149. These bits are masked since issue #107 
The funny thing is idcode is common for both 12 and 25. (see #114 ). This has no impact to load/program these devices.

emard, 2021-11-19:
12 and 25 should be internally the same, differing only in this IDCODE.
Bitstream should also match full 32-bit idcode to be accepted by the chip.

There are also serdes chips, this is idcode example with similar difference
bitstream acceptance is also similar, should match full 32-bit).

85F with serdes
***@***.***:/tmp/openFPGALoader/build$ fujprog -i
ULX2S / ULX3S JTAG programmer v4.8 (git cc3ea93 built May 23 2021 17:03:05)
Copyright (C) Marko Zec, EMARD, gojimmypi, kost and contributors
Using USB cable: ULX3S FPGA 85K v3.0.3
FPGA IDCODE: 81113043
FPGA identified SIZE: 0
***@***.***:/tmp/openFPGALoader/build$ ./openFPGALoader -b ulx3s --detect
write to ram
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
index 0:
	idcode 0x1113043
	manufacturer lattice
	family ECP5
	model  LFE5U-85
	irlength 8

85F ordinary
***@***.***:/tmp/openFPGALoader/build$ fujprog -i
ULX2S / ULX3S JTAG programmer v4.8 (git cc3ea93 built May 23 2021 17:03:05)
Copyright (C) Marko Zec, EMARD, gojimmypi, kost and contributors
Using USB cable: ULX3S FPGA 85K v3.0.3
FPGA IDCODE: 41113043
FPGA identified SIZE: 85
***@***.***:/tmp/openFPGALoader/build$ ./openFPGALoader -b ulx3s --detect
write to ram
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
index 0:
	idcode 0x1113043
	manufacturer lattice
	family ECP5
	model  LFE5U-85
	irlength 8

trabucayre, 2021-11-19:
Exactly!
In fact the 24bits idcode is used at enumerate step (when openFPGALoader start). It's a convenient way to avoid to have to fill part.hpp will all possible revision for each FPGA. It's also possible to have access, using jtag instruction `0xE0`, the full 32bits idcode.
A match between full idcode and bitstream's idcode is required (#85) for all device (already done for gowin, TBD for all others). For lattice, I have the piece of code to find this into configuration data (.bit file) it's now only required to integrate this into `lattice.cpp` class.

emard, 2021-11-19:
for ecp5 with some external commandline tools (ecpunpack/eppack)
idcode can be changed to match onboard detected idcode.

If openFPGALoader would like to shell call here's the script that
seamless & user-friendly changes ID to upload blink LED originally
compiled for ordinary 85F into serdes 85F which as different ID
and it will blink as well.

#!/bin/sh
TMPBIT=/tmp/repacking_bitstream.config
ecpunpack \
  --idcode 0x41113043 \
  ${1} ${TMPBIT}
ecppack \
  --idcode 0x81113043 \
  --freq 62.0 \
  ${TMPBIT} ${2}
rm $TMPBIT

trabucayre, 2021-11-19:
It's interesting to see ability to repack bitstream with a different idcode, but I'm not sure if this task has to be done by openFPGALoader or manually (or automatically) previously the call to openFPGALoader.

emard, 2021-11-19:
If openFPGALoader distinguishes devices only by lowest 24-bit
and ID doesn't match openFPGALoader can print message
suggesting some shell commands how to unpack/repack to
get matching ID.

trabucayre, 2021-11-20:
openFPGALoader is able, at device driver level, to obtain the 32bits idcode -> it's possible to know full FPGA model.
But it's true: an exact match must be done before loading/wrting bitstream and a message must be displayed when there is no match (with a link to doc/lattice.mk for instance).
This script example may/must be added to lattice.md to provides a way to adapt bitstream according to the targeted model.

trabucayre, 2021-11-20:
The IDCODE check is now applied in both `master` and `rework_spiflash` branches.

emard, 2021-11-20:
OK, I'm testing:

I have 12F connected and trying to upload 85F bitstream:
Error: Failed to program FPGA: mismatch between target's idcode and
bitstream idcode

This message is not easy to troubleshoot what should be done to succeed.

I'd like to have printed full 32-bit value of expected ID by the device and if
possible to parse the bitstream, also the ID from the file bitstream.

trabucayre, 2021-11-20:
I have updated with a better message (display FPGA and bitstream idcode).
It's okay?

emard, 2021-11-20:
Almost perfect

Error: Failed to program FPGA: mismatch between target's idcode and
bitstream idcode
	expected 0x41113043 got 0x21111043

I'd reword it more to be clearly
hardware requires 0x41113043, bitstream has 0x21111043

emard, 2021-11-20:
Or sorry, it's actually reverse (file is for 85F, board is 12F :)
message should be:
bitstream has 0x41113043 hardware requires 0x21111043

trabucayre, 2021-11-20:
done.
thanks

emard, 2021-11-20:
Yess that's it! Now it's completely clear what should be done

trabucayre, 2021-11-20:
Great!
Now I need to do the same thing for all others devices and file type:-)
... finishing spi flash refactoring and playing with the IS25LP032 ;-)

Thanks

trabucayre, 2021-12-05:
I have more or less finished to refactor code. There is some case where it's not possible due to structure but globally it's seems good.
Could you try again to check if there is no regression? Thanks

trabucayre, 2021-12-13:
flash protection area has been improved to use top/bottom.
A specific WA for IS25LP032 has been added too.

trabucayre, 2021-12-22:
I've integrated everything into master branch.
It's good for you?
Thanks for motivation!

emard, 2021-12-22:
It should be good! I'm going to compile and test master

trabucayre, 2021-12-22:
Thanks!
My worry is your problem with cycloneV

emard, 2021-12-22:
It is a lot of things to test,
quick test of normal programming works.
I'm getting segmentation fault in DFU mode:

openFPGALoader --dfu --altsetting 0 ulx3s_12f_ecp5pll.bit
write to ram
Open file DONE
Parse file DONE
Segmentation fault

emard, 2021-12-22:
CycloneV will be later tested, I must clean the desktop with
some ongoing projects and connect C5 on.

emard, 2021-12-22:
New DFU multiboot.img for 12F 85F and other boards and
its pcb versions are here:

https://github.com/emard/ulx3s-bin/tree/master/fpga/dfu

check your board versions written on bottom like v3.0.8
It's not big deal if wrong versions is flashed,
Most of things will work, something around esp32 will not

trabucayre, 2021-12-23:
Thanks for your feeback!
I'm unable to reproduce (but using a bit file you can't omit --vid --pid);
```
penFPGALoader --dfu --altsetting 0 blink.bit 
write to ram
Open file DONE
Parse file DONE
DFU init failed with: Can't open device vid/pid == 0
```

But a weird thing: after programing through DFU, new bitstream must be loaded, it's not true. I think something is wrong in my DFU class.

trabucayre, 2021-12-23:
Okay. I've updated dfu.cpp. Missing USB reset when download end in DFUIdle state.

emard, 2021-12-23:
Great, when I use --vid --pid then it works and resets usb!

The bitstream can start without usb reset using "exit/execute" option,
it is more recommended way than reset because depending on DIP
switch position, usb reset may not jump to uploaded bitstream but
again to DFU bootloader

example how bitstream is uploaded and started using dfu-util:

dfu-util -a0 -D ulx3s_12f_ecp5pll.bit
dfu-util -a0 -e

However, without --vid -pid I get segfault
ofl --dfu --altsetting=0 ulx3s_12f_ecp5pll.bit
.. segfault like before

If segfault is fixed, it could print message about vid/pid options missing

emard, 2021-12-23:
Apart from segfault, your dfu mode starts correctly uploaded
bitstream from DFU mode. On ULX3S board, DIP SW 1 controls DFU:
Same as DIP SW ON is holding BTN1 (F1) during power on

DIP SW 1 ON (up) board powers up running DFU bitstream
DIP SW 1: OFF (down) board powers on running user (uploaded) bitstream

emard, 2021-12-23:
I can see unpredictable behaviour, sometimes reported ok,
sometimes segfault (valgrinding needed :)?

$ ofl --dfu --altsetting=0 ulx3s_12f_ecp5pll.bit
write to ram
Open file DONE
Parse file DONE
DFU init failed with: No DFU compatible device found

$ ofl --dfu --altsetting=0 ulx3s_12f_ecp5pll.bit
write to ram
Open file DONE
Parse file DONE
Segmentation fault

trabucayre, 2021-12-23:
I've fixed, using valgrind, some memory leaks. But with or without this fix I always have the message and no segfault...
You use master branch?

maybe with gdb to have a backtrack ?
```
ulimit -c unlimited
ofl --dfu --altsetting=0 ulx3s_12f_ecp5pll.bit
gdb ofl core
```

emard, 2021-12-23:
HI

I recompiled latest branch (it should be master)
***@***.***:/tmp/openFPGALoader/build$ git branch
* master

this is what gdb says:
```
***@***.***:~/src/fpga/ulx3s/ulx3s-misc/examples/ecp5pll$ ulimit -c unlimited
***@***.***:~/src/fpga/ulx3s/ulx3s-misc/examples/ecp5pll$ ofl --dfu
--altsetting=0 ulx3s_12f_ecp5pll.bit
write to ram
Open file DONE
Parse file DONE
Segmentacijska greška (**krah**: stanje (dump) memorije je zapisano!)
***@***.***:~/src/fpga/ulx3s/ulx3s-misc/examples/ecp5pll$ gdb ofl core
GNU gdb (Debian 10.1-2) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ofl...
(No debugging symbols found in ofl)

warning: core file may not match specified executable file.
[New LWP 58336]
[New LWP 58337]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `ofl --dfu --altsetting=0 ulx3s_12f_ecp5pll.bit'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f3ac3812101 in libusb_close () from
/lib/x86_64-linux-gnu/libusb-1.0.so.0
[Current thread is 1 (Thread 0x7f3ac323a780 (LWP 58336))]
(gdb)

```

trabucayre, 2021-12-23:
could you use `bt` in gdb prompt ?
And compile using `cmake -DCMAKE_BUILD_TYPE=Debug`

emard, 2021-12-23:
HI

yes, bt gets a more detailed report:

```
***@***.***:~/src/fpga/ulx3s/ulx3s-misc/examples/ecp5pll$ ofl --dfu
--altsetting=0 ulx3s_12f_ecp5pll.bit
write to ram
Open file DONE
Parse file DONE
Segmentacijska greška (**krah**: stanje (dump) memorije je zapisano!)
***@***.***:~/src/fpga/ulx3s/ulx3s-misc/examples/ecp5pll$ ulimit -c unlimited
***@***.***:~/src/fpga/ulx3s/ulx3s-misc/examples/ecp5pll$ gdb ofl core
GNU gdb (Debian 10.1-2) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ofl...

warning: core file may not match specified executable file.
[New LWP 58946]
[New LWP 58947]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `ofl --dfu --altsetting=0 ulx3s_12f_ecp5pll.bit'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f5f8587e101 in libusb_close () from
/lib/x86_64-linux-gnu/libusb-1.0.so.0
[Current thread is 1 (Thread 0x7f5f852a6780 (LWP 58946))]
(gdb) bt
#0  0x00007f5f8587e101 in libusb_close () from
/lib/x86_64-linux-gnu/libusb-1.0.so.0
#1  0x00005613e87b9f6d in DFU::searchDFUDevices (this=0x5613e9882f80)
    at /tmp/openFPGALoader/src/dfu.cpp:309
#2  0x00005613e87b882d in DFU::DFU (this=0x5613e9882f80,
filename="ulx3s_12f_ecp5pll.bit", vid=0,
    pid=0, altsetting=0, verbose_lvl=0) at /tmp/openFPGALoader/src/dfu.cpp:100
#3  0x00005613e881cb8b in main (argc=4, argv=0x7ffd3d7322d8) at
/tmp/openFPGALoader/src/main.cpp:333
(gdb)
``

On 12/23/21, Gwenhael Goavec-Merou ***@***.***> wrote:
> could you use `bt` in gdb prompt ?
> And compile using `cmake -DCMAKE_BUILD_TYPE=Debug`
>
> --
> Reply to this email directly or view it on GitHub:
> https://github.com/trabucayre/openFPGALoader/issues/124#issuecomment-1000212271
> You are receiving this because you authored the thread.
>
> Message ID: ***@***.***>

trabucayre, 2021-12-23:
Yes. This is useful. I have added a check on libusb_open just before libusb_close. Maybe one device on your computer can't be opened and consequently fails at close time.

emard, 2021-12-23:
Yes, recent change fixes the segfault issuse

$ ofl --dfu --altsetting=0 ulx3s_12f_ecp5pll.bit
write to ram
Open file DONE
Parse file DONE
Unable to open device: 1d6b:0003 (bus 4, device  1) Error: LIBUSB_ERROR_ACCESS

DFU init failed with: Devices enumeration failed

This device is the root hub
# lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

trabucayre, 2021-12-24:
Cool!
I have modified a bit: now if libusb_open fails, move directly to next one instead of producing an error.
Thanks!

emard, 2021-12-24:
OK, now I tested, it works if specific vid/pid is given and
if it is not, then it lists probing all devices except the DFU
one and exits. Maybe it is possible to autodetect or if
given -b ulx3s --dfu to assume this vid/pid

$ ofl --dfu --altsetting=0 --vid 0x1d50 --pid 0x614b ulx3s_12f_ecp5pll.bit.gz
write to ram
Cable VID overridden
Cable PID overridden
Open file DONE
Parse file DONE
Open device 1d50:614b DONE
1d50 614b
Loading: [==================================================] 100.00%
Done

$ ofl --dfu --altsetting=0 ulx3s_12f_ecp5pll.bit.gz
write to ram
Open file DONE
Parse file DONE
Unable to open device: 1d6b:0003 (bus 4, device  1) Error:
LIBUSB_ERROR_ACCESS -> skip

Unable to open device: 1d6b:0002 (bus 3, device  1) Error:
LIBUSB_ERROR_ACCESS -> skip

Unable to open device: 0bda:0316 (bus 2, device  2) Error:
LIBUSB_ERROR_ACCESS -> skip

Unable to open device: 1d6b:0003 (bus 2, device  1) Error:
LIBUSB_ERROR_ACCESS -> skip

Unable to open device: 0424:2514 (bus 1, device 35) Error:
LIBUSB_ERROR_ACCESS -> skip

Unable to open device: 8087:0a2a (bus 1, device  4) Error:
LIBUSB_ERROR_ACCESS -> skip

Unable to open device: 4242:e131 (bus 1, device  3) Error:
LIBUSB_ERROR_ACCESS -> skip

Unable to open device: 045e:0039 (bus 1, device  2) Error:
LIBUSB_ERROR_ACCESS -> skip

Unable to open device: 1d6b:0002 (bus 1, device  1) Error:
LIBUSB_ERROR_ACCESS -> skip

DFU init failed with: Can't open device vid/pid == 0

trabucayre, 2021-12-24:
I've added a level of verbosity to avoid this big list...
Adding a board in dfu mode (as orangeCrab, fomu or icebreaker-bitsy) make sense.
I'm not happy by autodetecting approach -> my wifi card has dfu compatible interface -> it's too risky...

emard, 2021-12-24:
OK I compiled, seems fine.

I was also thinking that autodetect can flash wrong device :)

If it's easy code that can iterates over dfu-compatible interfaces
I think it could print --vid= --pid= parameters for each and let
user copy-paste.

Other option is board preset, maybe something like
-b ulx3s --dfu or -b ulx3s-dfu

trabucayre, 2021-12-24:
Added [this](https://github.com/trabucayre/openFPGALoader/blob/master/src/board.hpp#L165)

emard, 2021-12-24:
Hey thanx, I compiled that works great!!!
Wonderful!

sylefeb, 2023-06-25:
Hi everyone, I have an issue that I think is related to this discussion, but otherwise I can open a different issue, let me know.

Using openFPGALoader `openFPGALoader v0.10.0` under MinGW.

On two ULX3S using the W25Q128JV flash chips, after programming the flash with openFPGALoader (`-f` option), the flash is no longer accessible.

The first programming went well with openFPGALoader recognizing the flash as the correct chip. On second programming and since then I get:
```
$ ./openFPGALoader.exe -f -b ulx3s ~/Silice/projects/blinky/BUILD_ulx3s/build.bit
write to flash
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
Open file DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Detail:
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
EDID + CFD length : ff
EDID              : ffff
CFD               :
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash
Enable configuration: DONE
SRAM erase: DONE
Detail:
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
EDID + CFD length : ff
EDID              : ffff
CFD               :
Skip resetting device
Refresh: DONE
Flash is blank
Error: Failed to program FPGA: std::exception
```
The `--unprotect-flash` does not work either (showing only relevant exert):
```
flash chip unknown: use basic protection detection
unlock blocks
timeout: ff ff 1000
ff
wait: Error
```
I also tried with `fujprog` to no avail:
``` 
$ fujprog -j flash ~/Silice/projects/blinky/BUILD_ulx3s/build.bit
ULX2S / ULX3S JTAG programmer v4.8 (git cc3ea93 built May 10 2023 19:51:22)
Copyright (C) Marko Zec, EMARD, gojimmypi, kost and contributors
Using USB cable: ULX3S FPGA 85K v3.0.8
Erasing sectors, please wait.
Programming: 0% - Received and expected data do not match!
TDO: C100 Expected: 4000 mask: C100
Line 32: Operation not permitted

Failed.
```
I usually programmed the flash with `fujprog` and it has always worked on both boards before, but I don't think I had tried with openFPGAloader until now. SRAM programming still works fine.

The flash seems no longer accessible. Problem is, I don't know how to resurrect it. Read through this issue (this is why I tried --unprotect-flash) but could not find a fix for this particular problem. Suggestions welcome - thanks!

trabucayre, 2023-06-25:
My first idea is, indeed an conflict between gateware currently used and ECP5's state. Maybe SPI pins are configured as simple GPIO and, consequenctly, can't be used by internal jtag<->spi controler.
- another idea is a conflict between ESP32 and JTAG
- last question (I need to check datasheet): you use the flash in QSPI mode, is it possible this one is left in this state and don't understand instructions?

I have tested with my ULX3S with a blink in both SRAM and FLASH and can't reproduce your situation.

sylefeb, 2023-06-25:
Thanks for the tests! So very likely something wrong with my gateware preventing reprog through JTAG (and indeed the SPIflash is no longer enumerated properly, if I understand the logs). 

I do use SPI through a USRMCLK for the clock, running 50MHz. Otherwise nothing fancy. I did have similar issues in the past with QSPI, eg. with iceprog due to the flash staying in this mode, so that is a good suspect.

However I checked the code and I believe openFPGAloader SPI reset (with FF) should reset QSPI (which is otherwise enabled by 38h from the gateware). Also the gateware still does read properly (it sends over UART and I get the correct bytes). I also tried to flash over SRAM with a gateware resetting QSPI, and a gateware not using SPIflash, no change so far. I'll try something that powers it down next. 

@emard Is there a way I could prevent the FPGA to configure from the current bitstream in SPIflash on power up?

Unsure for the ESP32, I do not use it at all (always only used the fpga). Could its state have been changed by openFPGAloader?

Thanks!

trabucayre, 2023-06-25:
For icebreaker I have seen this issue, and reset sequence seems not reverting behaviour...
I don't remember exactly but ESP32 must be disabled to avoid this one keep SPI flash?

trabucayre, 2023-06-25:
I'm wrong: ESP32 is connected to SD slot not to the SPI flash

sylefeb, 2023-06-25:
I feel this is due to QPI staying on, and it seems my attempts at disabling it are failing so far (while the simulation model from Winbond recognizes my end QPI commands). I'll keep trying -- also ordered a programming clip just in case. I'll send my results (but if anyone knows a way to prevent the FPGA from reading the flash -- like the CRESET jumper on the icebreaker -- please let me know :) ). Thanks!

trabucayre, 2023-06-26:
Maybe you can try to apply 3.3V (or to shunt) R30 to force CS high -> EPC5 will not be able to receive configuration from SPI flash.
Unfortunately I'm not sure there is something equivalent to CRESET (programming flash for ICE40 doesn't need to pass through FPGA).

Another idea: clearing first sectors (or the full FLASH) by using a gateware with a uC (or something like that).

emard, 2023-06-26:
HI

I think flash SPI is alone depending only on JTAG, no other hardware state
can interfere, so not ESP32 neither user bitstream can affect flash SPI
interface.

ECP5 has factory internal bridge for SPI flash that works over its JTAG and
it
works regardless of what user has loaded. So if JTAG works for other things
like programming FPGA it should work for SPI too.

However flash chip can be in some state, either temporary by
already using it from user bitstream or permanently by writing
something to its flash memory. Probaby in most cases it should
be able to resurrect it, in worst case it's relatively easy to
unsolder 8 pins and place new chip. I think nobody yet bricked
it but Goran will help you even in this case

Of course problem can be that SPI flash chip has locked itself.
I have some tools in ulx3s-bin ecp5wp.py that could be tried
to unlock the flash using esp32, although it's quite cumbersome

this tool probably needs newer passthru from dfu (esp32 needs
a user bitstream to access spi flash, otherwise it couldn't) -
for v3.1.7 and later try
https://github.com/emard/ulx3s-bin/tree/master/fpga/dfu/85f-v317
or 85f-v20 for v3.0.x boards

emard, 2023-06-26:
here is the small esp32 tool that I used to remove write protection
from SPI flash chips on ulx3s boards before openFPGAloader
made support for the chips

https://github.com/emard/esp32ecp5/blob/master/ecp5wp.py

[14 more replies]