trabucayre/openFPGALoader #125: flashrom interoperability

emard, 2021-10-06:
There's utility "flashrom" which already supports a lot
of flash chips and I'm thinking how it could be used for FPGA
https://www.flashrom.org/Supported_hardware#USB_Devices

I'm thinking about idea if openFPGALoader could expose
a dll library or network compatible protocol like "serprog"
so "flashrom" utility could use openFPGALoader as
some virtual programmer bridge to access flash chip.

trabucayre, 2021-10-06:
The idea to split openFPGALoader between a library and a frontend is in my TODO list. The big question is about run time impact. Using FPGA lowlevel code to access flash is a solution to take profit to both code.
Anyway, I've planned ASAP to add features requested by #124 and to improve a bit (lot ?) spiFlash class.
Thanks

emard, 2021-10-07:
> For library it's something I consider interesting/necessary. In fact, to do
> tests or addition usually I create a small app out-of-tree and It's a bit
> boring to have to write a long Makefile with all required direct or indirect
> classes.
> I have started to work to (un)protect sectors but it's required to have
> knowledge about flash chip so I need to integrate list of known devices.
> Bottom to top sectors protection requires to change one bit, on most flash
> it's just non-volatile bit to set but for anothers chips it's an
> OneTimeProgramming bit -> In this situation I'm not happy to be (or
> openFPGALoader) responsable to a unrevertible modification.
>
> I Will check/add flash you point.
>
> WIP

Yes, OTP bit is very annoying, I have already set lock bit to 1
in winbond register and now I can't reset it to 0.

For handling write protection for flash chips found on my boards,
I have made a small esp32 micropython utility - it reports
protection status and sets default protection for first 2MB
that's what I need for bootloader.

https://github.com/emard/esp32ecp5/blob/master/ecp5wp.py

from my user point of view for universal tool, I'd like to enter wanted
protection range  like 0 - 0x1FFFFF and let the tool autodetect chip,
if OTP change is needed, report to user what consequences will be from
setting OTP bits (no more programming or just no more changing top/bottom)
ask for additional confirmation option and set wanted protection range.

trabucayre, 2021-10-08:
My answer in #124