trabucayre/openFPGALoader #129: Programming SRAM fails on USB-Blaster with Cyclone V

hansfbaier, 2021-11-06:
```
$ openFPGALoader -c usb-blaster --write-sram build/qmtech_5cefa2/gateware/qmtech_5cefa2.sof
write to ram
Can't program SPI flash: missing device-package information
Fail to load bridge
```
Also it tries to program SPI flash, when I want to program the SRAM instead.

trabucayre, 2021-11-06:
There is two issue with your commandline:
1. `sof` is not supported. The best way is to convert to `rbf` (without compression). Since `sof` is unknown openFPGALoader considers you want to store a binary data into flash but ;
2. with altera/intel and xilinx SPI flash needs a bridge JTAG<->SPI: this bitstream depends on target FPGA -> you need to provides `--fpga-part` or simply using `-b` with a board name (see `--list-boards`).

All details (but maybe needs to be improved) are [here](https://github.com/trabucayre/openFPGALoader/blob/master/doc/intel.md).

Try:
```
$ quartus_cpf  --option=bitstream_compression=off -c build/qmtech_5cefa2/gateware/qmtech_5cefa2.sof \
                     build/qmtech_5cefa2/gateware/qmtech_5cefa2.rbf
# SRAM:
$ openFPGALoader -c usb-blaster -b qmtechCycloneV build/qmtech_5cefa2/gateware/qmtech_5cefa2.rbf
# flash:
$ openFPGALoader  -f -c usb-blaster -b qmtechCycloneV build/qmtech_5cefa2/gateware/qmtech_5cefa2.rbf
```

trabucayre, 2021-11-06:
It's works now?

trabucayre, 2021-11-19:
Can I close this issue ?
Thanks

hansfbaier, 2021-11-19:
Oh I lost track of this issue. I still would have to test. Thanks for the solution!
Will be helpful to others too.
Will reopen, if I should encounter problems.

closed 2021-11-19 by hansfbaier

trabucayre, 2021-11-19:
Don't worry!
I have to complete specific .md files and the FAQ with most common issues to help people with this type of situation.
Thanks