trabucayre/openFPGALoader #131: QMTECH Cyclone IV EP4CE15 starter kit (works)

inkdot7, 2021-11-13:
I'd just like to report success in programming a QMTECH Cyclone IV EP4CE15 starter kit board.

Although not listed in the board or FPGA lists, it just worked:

`./openFPGALoader -v -v -c usb-blaster myproj.rbf`

giving the following output:
```
write to ram
found 1 devices
index 0:
        idcode 0x20f20dd
        manufacturer altera
        family cyclone III
        model  EP3C16
        irlength 10
File type : rbf
Flash SRAM: [==================================================] 100.00%
Done
```
It should be noted that the device on the board is an EP4CE15F23C8N.  According to https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/hb/cyclone-iv/cyiv-51010.pdf it has ID code 0x020f20dd, the same as EP3C16 which already is listed.

I'm especially happy that it worked, since the quartus_pgm randomly refused to work, reporting failure to scan the JTAG chain (error 87).  Mostly not working.

trabucayre, 2021-11-14:
Thanks for this report! I've updated openFPGALoader to display EP3 and/or EP4 name (this issue about idcode is true for cycloneIV and cyclone 10 too)
With a correct spiOverJtag bitstream it may be possible to write to flash (I must to find a way to store compressed bitstream into repository to avoid increasing repository size).
Thanks again

trabucayre, 2021-11-28:
Hi,
I have added spiOverJtag's bitstream for this board (and I've updated board list), so now you will be able to flash a bitstream into the SPI flash.
It's okay to you ?

inkdot7, 2021-11-28:
Worked for me:

`openFPGALoader -c usb-blaster -b qmtechCycloneIV -f myproj.rbf`

closed 2021-11-28 by inkdot7

trabucayre, 2021-11-29:
\o/ First spiOverJtag bitstream using compression!
Thanks !