trabucayre/openFPGALoader #132: Compression of bitstreams in repository

inkdot7, 2021-11-14:
Just to investigate the compression options for bitstreams in a repository (from comment in #131), I took the `spiOverJtag` and placed in new git repositories.  Either with the files as they are now, or individually compressed using gzip or xz.  It should also be noted that the bitstream files contain large sections of zeros, which on some filesystems does not use any space.  empty is just a empty repository for reference.

| Kind  | `du -Bk` (apparent) | `du -Bk` | `.git` (apparent) | `.git gc` (apparent) |
| --- | --- | --- | --- | --- |
| empty | 5               | 8      | 93   | 79  |
| plain | 30640           | 7892   | 606  | 227 |
| gzip  | 232             | 324    | 368  | 230 |
| xz    | 102             | 188    | 307  | 178 |

The second column is the actual file size of the `spiOverJtag` directory (large chunks of zero contribute).
The third column is the used filesystem size (large chunks of zero not contributing; can be larger than actual size due to filessystem block size).
The fourth column is the .git repository size directly after commit (not so interesting).
The fifth column is the .git repository size after `git gc` which applies the delta compression.

It thus looks like if only the size of the .git repository matters, there is no need to do anything.

If the size of the working directory matters, then the bitstream files could be individually compressed.  Which requires the loader to read them from e.g. `.bit.gz` or `.bit.xz`.  (Such an ability might be generally useful by itself.)

trabucayre, 2021-11-14:
Your analyze is really interesting! In fact there is an existing issue about this topic (#47).
My worry about spiOverJtag directory size is primarly to the download/clone step. Xilinx bitstream, with compress option enabled, have an acceptable size, but intel/altera ones must have compress option disabled, and consequently have a size more important. Most globally, due the potential number of devices this directory may become awfully heavy.

Concerning decompression at install step vs on-the-fly decompression:
- the first is the most simple. No openFPGALoader's internal modifications to do, but requires to have access to the correct tools (gunzip/xz) for all systems (possible for windows and mac ?)
- the second only requires to have correct library (again windows/mac ?), no modification at install step but maybe increase programming time due to decompress step (it's maybe a wrong question, maybe impact is really small compared to spi flash transactions).

inkdot7, 2021-11-15:
Thanks for pointing out #47.  I had missed that.  Closing this and putting further comments there.

closed 2021-11-15 by inkdot7