trabucayre/openFPGALoader #142: `openFPGALoader -V` still reports `0.5.0` on version `0.6.0`

carlocab, 2021-12-01:
Hi, we're trying to update Homebrew's openFPGALoader to 0.6.0 at https://github.com/Homebrew/homebrew-core/pull/90231.

CI fails upon verifying the output of `openFPGALoader -V`. It seems to still be reporting `0.5.0` with the new version.

It probably comes from here:

https://github.com/trabucayre/openFPGALoader/blob/130060619e1a6689db44f9a66e1aec4eb214ee93/CMakeLists.txt#L5

I'm not sure if this needs to be changed elsewhere, though.

trabucayre, 2021-12-02:
Thanks @jeanthom for this fix !

closed 2021-12-02 by trabucayre

jeanthom, 2021-12-07:
Don't worry, things have been fixed for 0.6.0 with a little bit of patching: https://github.com/Homebrew/homebrew-core/blob/628aed8b9e1e0d95a834615a9ee57c20bf4b6e5d/Formula/openfpgaloader.rb#L24-L29

0.6.1 without the patch is currently under review: https://github.com/Homebrew/homebrew-core/pull/90663

trabucayre, 2021-12-08:
Great and thanks again!
0.6.1 is a really small minor release: primarly to fix mismatch between version and tag, with actions improvement (without impact for you) and small addition to be compatible with zlib-ng too (when zlib is not found, but again no impact to you).

PR #144 (2021-12-02): Fix version number in CMakeLists.txt
Current version number is stuck at 0.5.0... Fixes #142