trabucayre/openFPGALoader #146: segfault if not file name

emard, 2021-12-07:
I was trying if maybe pipeline is supported in DFU mode
although I know it should not work, but I got segfault,
this is from rework branch:

zcat blink.bit.gz | /tmp/openFPGALoader/build/openFPGALoader --dfu --vid 0x1d50 --pid 0x614b --altsetting 0 --file-type bit
write to ram
Cable VID overridden
Cable PID overridden
Open device 1d50:614b DONE
1d50 614b
Segmentation fault

/tmp/openFPGALoader/build/openFPGALoader --dfu --file-type bit
write to ram
Segmentation fault

trabucayre, 2021-12-07:
Good catch. A wrong, and useless, check into dfu.cpp.
It's now fixed.

It's okay for you?

Thanks

emard, 2021-12-07:
yes it's fixed now I compiled, tested, no more segfault!

trabucayre, 2021-12-08:
Great!
Thanks for this use case I have forgotten when writing DFU support.

trabucayre, 2022-02-12:
It's okay for  you?
Can I close this issue ?
Thanks

emard, 2022-02-12:
If I remember it was fixed long time ago, let's close the issue

closed 2022-02-12 by emard