trabucayre/openFPGALoader #149: Can't flash Tang Nano 4k (JTAG init failed)

Chandler-Kluser, 2021-12-13:
I have compiled openFPGAloader with pamac (I use Manjaro). When I try to flash my Tang Nano 4k, I get:

```
sudo openFPGALoader -b tangnano4k -f '/home/chandler/GoWin Educational IDE/IDE/bin/hello_world_nand/impl/pnr/hello_world_nand.fs' 
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
JTAG init failed with: Unknown device with IDCODE: 0x100981b
```

I have already tried to do as normal user, with the same error...

trabucayre, 2021-12-13:
Could you try 
```openFPGALoader -V```
It must answer **v0.6.0** or greater
and
```
openFPGALoader --list-fpga | grep GW1NSR-4C
```
the answer must be
```
0x100981b   Gowin         GW1NSR         GW1NSR-4C
```

If it's not true, could you try to build from source, it seems manjaro provides 0.5.0 version and the release with **GW1NSR-4C** is 0.6.0

Thanks

Chandler-Kluser, 2021-12-13:
No GWNSR-AC...

```
IDCode      manufacturer  family         model               
0x81111043  lattice       ECP5           LFE5UM5G-25         
0x81112043  lattice       ECP5           LFE5UM5G-45         
0x81113043  lattice       ECP5           LFE5UM5G-85         
0xe12bb043  lattice       MachXO3LF      LCMX03LF-1300C      
0x100381b   Gowin         GW1N           GW1N-4              
0x10f0043   lattice       CrosslinkNX    LIFCL-17            
0x10f1043   lattice       CrosslinkNX    LIFCL-40-ES         
0x1111043   lattice       ECP5           LFE5UM-25           
0x1112043   lattice       ECP5           LFE5UM-45           
0x1113043   lattice       ECP5           LFE5UM-85           
0x120681b   Gowin         GW1N           GW1N-2              
0x129a043   lattice       XP2            LFXP2-8E            
0x12b5043   lattice       MachXO2        LCMXO2-7000HE       
0x12b9043   lattice       MachXO2        LCMXO2-640HC        
0x12ba043   lattice       MachXO2        LCMXO2-1200HC       
0x12bd043   lattice       MachXO2        LCMXO2-7000HC       
0x20f20dd   altera        cyclone III    EP3C16              
0x20f30dd   altera        cyclone 10 LP  10CL025             
0x2b050dd   altera        cyclone V      5CEBA4              
0x2b150dd   altera        cyclone V      5CEA2               
0x2d010dd   altera        cyclone V Soc  5CSEMA4             
0x2d020dd   altera        cyclone V Soc  5CSEBA6             
0x300181b   Gowin         GW1NS          GW1NS-2C            
0x3620093   xilinx        spartan7       xc7s15ftgb196-1     
0x362c093   xilinx        artix a7 50t   xc7a50t             
0x362d093   xilinx        artix a7 35t   xc7a35              
0x362f093   xilinx        spartan7       xc7s50              
0x37c4093   xilinx        spartan7       xc7s25              
0x900281b   Gowin         GW1N           GW1N-1              
0xa014c35   anlogic       eagle s20      EG4S20BG256         
0x1100581b  Gowin         GW1N           GW1NR-9             
0x110f1043  lattice       CrosslinkNX    LIFCL-40            
0x13631093  xilinx        artix a7 100t  xc7a100             
0x13632093  xilinx        artix a7 75t   xc7a75t             
0x13636093  xilinx        artix a7 200t  xc7a200             
0x21111043  lattice       ECP5           LFE5U-12            
0x212e3043  lattice       MachXO3D       LCMX03D-9400HC      
0x23727093  xilinx        zynq           xc7z020             
0x24001093  xilinx        spartan6       xc6slx9             
0x24002093  xilinx        spartan6       xc6slx16            
0x24004093  xilinx        spartan6       xc6slx25            
0x24011093  xilinx        spartan6       xc6slx100           
0x310f0043  lattice       CertusNX       LFD2NX-17           
0x310f1043  lattice       CertusNX       LFD2NX-40           
0x41111043  lattice       ECP5           LFE5U-25            
0x41112043  lattice       ECP5           LFE5U-45            
0x41113043  lattice       ECP5           LFE5U-85            
0x43651093  xilinx        kintex7        xc7k325t            
0x44008093  xilinx        spartan6       xc6slx45            
0x612b2043  lattice       MachXO3LF      LCMX03LF-1300E      
0x612b3043  lattice       MachXO3LF      LCMX03LF-2100E      
0x612b4043  lattice       MachXO3LF      LCMX03LF-4300E      
0x612b5043  lattice       MachXO3LF      LCMX03LF-6900E      
0x612b6043  lattice       MachXO3LF      LCMX03LF-9400E      
0x612bb043  lattice       MachXO3LF      LCMX03LF-2100C      
0x612bc043  lattice       MachXO3LF      LCMX03LF-4300C      
0x612bd043  lattice       MachXO3LF      LCMX03LF-6900C      
0x612be043  lattice       MachXO3LF      LCMX03LF-9400C      
```

Is the openFPGAloader AUR package out of date?

Chandler-Kluser, 2021-12-13:
trying to compile the latest version from source, but there are many dependencies missing...

```[...]
[ 87%] Building CXX object CMakeFiles/openFPGALoader.dir/src/altera.cpp.o
[ 90%] Building CXX object CMakeFiles/openFPGALoader.dir/src/bitparser.cpp.o
[ 92%] Building CXX object CMakeFiles/openFPGALoader.dir/src/xilinx.cpp.o
[ 95%] Building CXX object CMakeFiles/openFPGALoader.dir/src/xilinxMapParser.cpp.o
[ 97%] Building CXX object CMakeFiles/openFPGALoader.dir/src/cmsisDAP.cpp.o
[100%] Linking CXX executable openFPGALoader
/usr/bin/ld: não foi possível localizar -lusb-1.0
/usr/bin/ld: não foi possível localizar -lftdi1
/usr/bin/ld: não foi possível localizar -lusb-1.0
/usr/bin/ld: não foi possível localizar -ludev
/usr/bin/ld: não foi possível localizar -lhidapi-hidraw
/usr/bin/ld: não foi possível localizar -lftdi1
/usr/bin/ld: não foi possível localizar -ludev
/usr/bin/ld: não foi possível localizar -lhidapi-hidraw
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/openFPGALoader.dir/build.make:705: openFPGALoader] Erro 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/openFPGALoader.dir/all] Erro 2
make: *** [Makefile:136: all] Erro 2
```

trabucayre, 2021-12-13:
I far I know archlinux has version 0.5.0 of openFPGALoader and this FPGA has been added after (you need release 0.6.0)
I'm not really aware of archlinux behaviour (maybe I need to test with a docker image) but it's seems you have to install dev libraries.
Will try ASAP an update doc about this distribution.

pu-cc, 2021-12-13:
@Chandler-Kluser did you specify any defines during `cmake`? I just tested this on a fresh VM and only had to install following packages:

```
sudo pacman -S git cmake make gcc pkgconf libftdi libusb
```

Then just proceed as described in the [install docs](https://github.com/trabucayre/openFPGALoader/blob/master/doc/guide/install.rst) to build the app. I build openFPGALoader myself from sources on manjaro, too.

Chandler-Kluser, 2021-12-13:
This is it! I used a fresh minimal manjaro KDE VM to compile it, here are the instructions:
```
sudo pacman -S git cmake make gcc pkgconf libftdi libusb
git clone "https://github.com/trabucayre/openFPGALoader.git"
cd openFPGALoader
mkdir build
cd build
cmake ../ 
cmake --build .
```

This is what I get now in the release when I do `openFPGALoader --list-fpga | grep GW1NSR-4C`:

[image]

I have not tried the binary yet on my main pc.

Thanks to @pu-cc for the help!

closed 2021-12-13 by Chandler-Kluser

Chandler-Kluser, 2021-12-13:
worked flawlessly on Manjaro KDE building from source, thanks.

trabucayre, 2021-12-14:
Thanks for your feedback!
install.rst page must be updated. Have to add details on how to built from source

trabucayre, 2021-12-14:
install.rst updated!
Thanks to all